start-jsk/rtmros_choreonoid #14: choreonoidが死んでしまう

YuOhara, 2015-08-21:
jaxon_jvrc_choreonoid.launchをあげたら、

[image]

これを押したら
'''
[modelloader-1] process has finished cleanly
log file: /home/leus/.ros/log/21dfd144-47d3-11e5-aef2-cc3d825280be/modelloader-1*.log
respawning...
[modelloader-1] restarting process
process[modelloader-1]: started with pid [21084]
IDL:omg.org/CORBA/TRANSIENT:1.0
cloneMap.setNonNodeCloning(true);
cloneMap.setNonNodeCloning(true);
[modelloader-1] process has finished cleanly
log file: /home/leus/.ros/log/21dfd144-47d3-11e5-aef2-cc3d825280be/modelloader-1*.log
respawning...
[modelloader-1] restarting process
process[modelloader-1]: started with pid [21089]
IDL:omg.org/CORBA/TRANSIENT:1.0
cloneMap.setNonNodeCloning(true);
cloneMap.setNonNodeCloning(true);
Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt. You must
reimplement QApplication::notify() and catch all exceptions there.

terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
[modelloader-1] process has finished cleanly
log file: /home/leus/.ros/log/21dfd144-47d3-11e5-aef2-cc3d825280be/modelloader-1*.log
respawning...
[modelloader-1] restarting process
process[modelloader-1]: started with pid [21095]
IDL:omg.org/CORBA/TRANSIENT:1.0
/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid/launch/run_choreonoid.sh: line 41: 16756 Aborted                 (core dumped) choreonoid $cnoid_proj
[modelloader-1] process has finished cleanly
log file: /home/leus/.ros/log/21dfd144-47d3-11e5-aef2-cc3d825280be/modelloader-1*.log
respawning...
[choreonoid-2] process has died [pid 16748, exit code 134, cmd /home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid/launch/run_choreonoid.sh /home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/jaxon_hg.cnoid -o manager.is_master:YES -o corba.nameservers:localhost:2809 -o naming.formats:%n.rtc -o logger.file_name:/tmp/rtc%p.log -endless -realtime -o exec_cxt.periodic.rate:1000000 -o manager.shutdown_onrtcs:NO -o manager.modules.load_path:/home/leus/ros/hydro_parent/devel/share/hrpsys/lib -o manager.modules.preload:.so -o manager.components.precreate: -o exec_cxt.periodic.type:SynchExtTriggerEC -o example.SequencePlayer.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.ForwardKinematics.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.ImpedanceController.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.AutoBalancer.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.StateHolder.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.TorqueFilter.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.TorqueController.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.ThermoEstimator.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.ThermoLimiter.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.VirtualForceSensor.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.AbsoluteForceSensor.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.RemoveForceSensorLinkOffset.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.KalmanFilter.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.Stabilizer.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.CollisionDetector.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.SoftErrorLimiter.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.HGcontroller.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.PDcontroller.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.EmergencyStopper.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.RobotHardware.config_file:/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf __name:=choreonoid __log:=/home/leus/.ros/log/21dfd144-47d3-11e5-aef2-cc3d825280be/choreonoid-2.log].
log file: /home/leus/.ros/log/21dfd144-47d3-11e5-aef2-cc3d825280be/choreonoid-2*.log

'''
こんな出力でchoreonoidが死んでしまうのですが、どうすればいいでしょうか？

garaemon, 2015-08-21:
とりあえずchoreonoidをgdbしてみましょう

garaemon, 2015-08-21:
launch-prefix="xterm -e gdb --args"かな

YoheiKakiuchi, 2015-08-21:
できれば、それはchoreonoidのissueに投げてあまりこだわらずに行きましょう。

スタートを一回した後の停止や再開はうまく行っていないように思います。

YoheiKakiuchi, 2015-08-21:
```S>``` みたいになっているのがシミュレーションのスタートでそれは大丈夫だと思う。

YuOhara, 2015-08-21:
```
leus@leus-ThinkPad-W541:~/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid/launch$ git diff run_choreonoid.sh
diff --git a/hrpsys_choreonoid/launch/run_choreonoid.sh b/hrpsys_choreonoid/laun
index 6ae178e..938e53f 100755
--- a/hrpsys_choreonoid/launch/run_choreonoid.sh
+++ b/hrpsys_choreonoid/launch/run_choreonoid.sh
@@ -38,4 +38,4 @@ echo "<BEGIN: rtc.conf>"
 cat $rtc_conf 1>&2
 echo "<END: rtc.conf>"
 
-(cd /tmp; choreonoid $cnoid_proj)
+(cd /tmp; xterm -e gdb --args choreonoid $cnoid_proj)
(END)
```
でgdb越しに上がりました

YuOhara, 2015-09-11:
こちら、問題は #17 と同じだったようです。omniNamesを上げ直すという方法で治ります。

s-nakaoka/choreonoid#26 で教えていただいてわかったことは、qtでエラーが出ているように見えるが、実際にそうではないので、qtをソースからコンパイルしてgdbをあげないとダメということです。

解決方法がわかったので、closeしたいと思います。

closed 2015-09-11 by YuOhara

k-okada, 2015-09-12:
omninamesを上げ直すときに
rtm-naming
とするとよいです．
 /etc/init.d/omniorb4-nameserver restart
はいくつか問題があって，特にノートPCとかでつかっていて
学校で作業して，帰りの電車のなかでテザリングで作業して，家で作業する，というよく有るシチュエーションで
毎回IPが変わる度におかしい事が起こったりします（IPが変わるとrestartが必要＋ログファイルを消さないとダメだったりする）

rtm-namingすると，ログファイルもちゃんと消してくれたり，プロセスを立ち上げる時に，うらでプロセスが残っていたら消してくれたり
してくれます．

http://www.openrtm.org/openrtm/ja/content/openrtm-users-01747-%E3%82%BE%E3%83%B3%E3%83%93%E3%82%AA%E3%83%96%E3%82%B8%E3%82%A7%E3%82%AF%E3%83%88%EF%BC%9F

YuOhara, 2015-09-12:
なるほど。ありがとうございます。
そのようにします