start-jsk/rtmros_choreonoid #15: joint_statesが出ない

YuOhara, 2015-08-21:
何かプログラムが止まってるような気がするのですが関係あるでしょうか?
```
[abc] setAutoBalancerParam
[abc]   move_base_gain = 0.8
[abc]   default_zmp_offsets = 0 0 0 0 0 0
[abc]   graspless_manip_mode = 0
[abc]   graspless_manip_arm = arms
[abc]   graspless_manip_p_gain =     [0,  0,  0]
[abc]   graspless_manip_reference_trans_pos =     [0,  0,  0]
[abc]   graspless_manip_reference_trans_rot =     [1, 0, 0]
    [0, 1, 0]
    [0, 0, 1]
[abc]   transition_time = 2[s], zmp_transition_time = 1[s], adjust_footstep_transition_time = 2[s]
[abc]   leg_names [rleg]
[abc]   leg_names [lleg]
[kf] setKalmanFilterParam
[kf]   Q_angle=0.001, Q_rate=0.003, R_angle=1000
[kf]   kf_algorithm=RPYKalmanFilter
[kf]   acc_offset =     [0,  0,  0]
[kf]   sensorRPY_offset =     [0,  -0,  0]
Traceback (most recent call last):
  File "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/scripts/jaxon_jvrc_hrpsys_config.py", line 44, in <module>
    hcf.init(sys.argv[1], sys.argv[2])
  File "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/src/hrpsys_ros_bridge_tutorials/urata_hrpsys_config.py", line 21, in init
    self.setStAbcParameters()
  File "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/src/hrpsys_ros_bridge_tutorials/urata_hrpsys_config.py", line 38, in setStAbcParameters
    self.setStAbcIcParametersJAXON(foot="KAWADA")
  File "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/src/hrpsys_ros_bridge_tutorials/urata_hrpsys_config.py", line 176, in setStAbcIcParametersJAXON
    self.st_svc.setParameter(stp)
  File "/usr/lib/python2.7/dist-packages/StabilizerService_idl.py", line 175, in setParameter
    return _omnipy.invoke(self, "setParameter", _0_OpenHRP.StabilizerService._d_setParameter, args)
omniORB.CORBA.BAD_PARAM: CORBA.BAD_PARAM(omniORB.BAD_PARAM_WrongPythonType, CORBA.COMPLETED_NO)
[hrpsys_py-3] process has died [pid 24667, exit code 1, cmd /home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/scripts/jaxon_jvrc_hrpsys_config.py JAXON_JVRC /home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/jvrc_models/JAXON_JVRC/JAXON_JVRCmain.wrl -ORBInitRef NameService=corbaloc:iiop:localhost:2809/NameService __name:=hrpsys_py __log:=/home/leus/.ros/log/16c92b48-47e2-11e5-b6de-fcaa1417ff45/hrpsys_py-3.log].
log file: /home/leus/.ros/log/16c92b48-47e2-11e5-b6de-fcaa1417ff45/hrpsys_py-3*.log
```

```
[rtmlaunch] [ERROR] Could not Connect ( /localhost:2809/JAXON_JVRC.rtc:q , /localhost:2809/HrpsysSeqStateROSBridge0.rtc:rsangle ):  No such object: /localhost:2809/HrpsysSeqStateROSBridge0.rtc:rsangle 

```

YoheiKakiuchi, 2015-08-21:
とりあえず、一部だけでなくてコンソールを全部はってもらえますか。

あと、aptitude show hrpsys-base とするとなんと返って来ますか

YuOhara, 2015-08-24:
```
leus@leus-GB-BXi5G-760:~/ros/indigo/src/jsk-ros-pkg/jsk_recognition/jsk_perception$ roslaunch hrpsys_ros_bridge_jvrc jaxon_jvrc_choreonoid.launch 
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: WARN: unrecognized 'rtconnect' tag in <node> tag. Node xml is <node args="$(arg openrtm_args)" name="CHESTCAMERA" ns="chest_camera" output="$(arg OUTPUT)" pkg="hrpsys_ros_bridge" type="ImageSensorROSBridge [... cut ...]
  File "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_common/hrpsys_ros_bridge/scripts/sensor_ros_bridge_connect.py", line 60, in <module>
    initSensorRosBridgeConnection(sys.argv[1], sys.argv[2], sys.argv[3], sys.argv[4])
  File "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_common/hrpsys_ros_bridge/scripts/sensor_ros_bridge_connect.py", line 44, in initSensorRosBridgeConnection
    time.sleep(1);
KeyboardInterrupt
[kf] onDeactivated(0)
[abc] onDeactivated(0)
[rmfo] onDeactivated(0)
```
ログはこれです

YuOhara, 2015-08-24:
```
leus@leus-GB-BXi5G-760:~$ aptitude show hrpsys-base 
Package: hrpsys-base                     
New: yes
State: installed
Automatically installed: no
Version: 3.1.5+20150820+1160+9~ubuntu14.04.1
Priority: extra
Section: science
Maintainer: Fumio KANEHIRO <f-kanehiro@aist.go.jp>
Architecture: amd64
Uncompressed Size: 74.9 M
Depends: freeglut3, libboost-python1.54.0, libboost-system1.54.0, libc6 (>=
         2.15), libgcc1 (>= 1:4.1.1), libgl1-mesa-glx | libgl1, libglew1.10 (>=
         1.10.0), libglu1-mesa | libglu1, libomniorb4-1 (>= 4.1.6),
         libomnithread3c2 (>= 4.0.6), libopencv-core2.4, libopencv-highgui2.4,
         libopencv-imgproc2.4, libpython2.7 (>= 2.7), libqhull6, libsdl1.2debian
         (>= 1.2.11), libstdc++6 (>= 4.6), libxml2 (>= 2.7.4), openhrp,
         openrtm-aist, openrtm-aist-dev, ipython, jython, openrtm-aist-python,
         omniidl-python, python-tk
Conflicts: hrpsys-base
Description: Basic RT components and utilities to control robots using OpenRTM
 
Homepage: http://code.google.com/p/hrpsys-base

```

YoheiKakiuchi, 2015-08-24:
ここで現在進行形ですが、
https://github.com/fkanehiro/hrpsys-base/issues/764

いまのところrosのhrpsysを使うのが正解です。
sudo apt-get update; sudo apt-get dist-upgradeして。
sudo apt-get remove hrpsys-base してみてください。
/usr/lib/python2.7/dist-packages/以下に hrpsysディレクトリやxxx_idl.pyファイルなどがないか確認して下さい。

YuOhara, 2015-08-24:
```
leus@leus-GB-BXi5G-760:/usr/lib/python2.7/dist-packages$ ls | grep idl.py
ClockGenerator_idl.py
CollisionDetector_idl.py
Controller_idl.py
CosNaming_idl.py
CosNaming_idl.pyc
DynamicsSimulator_idl.py
ModelLoader_idl.py
OnlineViewer_idl.py
OpenHRPCommon_idl.py
OpenHRPCommon_idl.pyc
PathPlanner_idl.py
ViewSimulator_idl.py
World_idl.py
leus@leus-GB-BXi5G-760:/usr/lib/python2.7/dist-packages$ ls | grep hrpsys
hrpsys
```
dis-upgradeまで手動で消すことになりますでしょうか？

```
leus@leus-GB-BXi5G-760:/usr/lib/python2.7/dist-packages$ aptitude show hrpsys-base 
Package: hrpsys-base                     
New: yes
State: not installed
Version: 3.1.5+20150824+1160+11~ubuntu14.04.1
Priority: extra
Section: science
Maintainer: Fumio KANEHIRO <f-kanehiro@aist.go.jp>
Architecture: amd64
Uncompressed Size: 74.9 M
Depends: freeglut3, libboost-python1.54.0, libboost-system1.54.0, libc6 (>=
         2.15), libgcc1 (>= 1:4.1.1), libgl1-mesa-glx | libgl1, libglew1.10 (>=
         1.10.0), libglu1-mesa | libglu1, libomniorb4-1 (>= 4.1.6),
         libomnithread3c2 (>= 4.0.6), libopencv-core2.4, libopencv-highgui2.4,
         libopencv-imgproc2.4, libpython2.7 (>= 2.7), libqhull6, libsdl1.2debian
         (>= 1.2.11), libstdc++6 (>= 4.6), libxml2 (>= 2.7.4), openhrp,
         openrtm-aist, openrtm-aist-dev, ipython, jython, openrtm-aist-python,
         omniidl-python, python-tk
Conflicts: hrpsys-base
Description: Basic RT components and utilities to control robots using OpenRTM
 ```

YoheiKakiuchi, 2015-08-24:
```sudo apt-get remove openhpr3``` もやってみてください。
その上で、のこっている /usr/lib/python2.7/dist-packages の下の、_idl.pyとhrpsys,OpenHRPディレクトリを手で消しましょう。

YuOhara, 2015-08-24:
こちらTFが出るところまで確認できました。ありがとうございます。

closed 2015-08-24 by YuOhara