start-jsk/rtmros_choreonoid #17: modelloaderがRestartし続ける

YuOhara, 2015-08-24:
```
leus@Ubuntu14:~$ roslaunch hrpsys_ros_bridge_jvrc jaxon_jvrc_choreonoid.launch 
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
... logging to /home/leus/.ros/log/38d65a22-4a4d-11e5-8b16-cc3d825280be/roslaunch-Ubuntu14-20458.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
started roslaunch server http://Ubuntu14:36472/

SUMMARY
========

PARAMETERS
 * /HrpsysSeqStateROSBridge/publish_sensor_transforms: True
 * /chest_camera/CHESTCAMERA/frame_id: CHEST_CAMERA
 * /controller_configuration: [{'group_name': '...
 * /diagnostic_aggregator/analyzers/computers/contains: ['HD Temp', 'CPU ...
 * /diagnostic_aggregator/analyzers/computers/path: Computers
 * /diagnostic_aggregator/analyzers/computers/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/hrpsys/contains: ['hrpEC', 'Emerge...
 * /diagnostic_aggregator/analyzers/hrpsys/path: Hrpsys
 * /diagnostic_aggregator/analyzers/hrpsys/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/mode/contains: ['Operating Mode'...
 * /diagnostic_aggregator/analyzers/mode/path: Mode
 * /diagnostic_aggregator/analyzers/mode/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/motor/contains: ['Motor']
 * /diagnostic_aggregator/analyzers/motor/path: Motor
 * /diagnostic_aggregator/analyzers/motor/type: diagnostic_aggreg...
 * /diagnostic_aggregator/base_path: 
 * /diagnostic_aggregator/pub_rate: 1.0
 * /larm_camera/LARMCAMERA/frame_id: LARM_CAMERA
 * /multisense/left/HEADLEFT/frame_id: HEAD_LEFT_CAMERA
 * /multisense/pointcloud_bridge/frame_id: HEAD_LEFT_CAMERA
 * /multisense/pointcloud_bridge/publish_depth: True
 * /multisense/pointcloud_bridge/transformed_camera_frame: True
 * /multisense/range_bridge/frame_id: HEAD_RANGE
 * /rarm_camera/RARMCAMERA/frame_id: RARM_CAMERA
 * /robot_description: <?xml version="1....
 * /robot_pose_ekf/debug: False
 * /robot_pose_ekf/freq: 30.0
 * /robot_pose_ekf/imu_used: False
 * /robot_pose_ekf/odom_used: True
 * /robot_pose_ekf/output_frame: odom
 * /robot_pose_ekf/self_diagnose: False
 * /robot_pose_ekf/sensor_timeout: 1.0
 * /robot_pose_ekf/vo_used: False
 * /rosdistro: indigo
 * /rosversion: 1.11.13
 * /use_sim_time: True

NODES
  /chest_camera/
    CHESTCAMERA (hrpsys_ros_bridge/ImageSensorROSBridge)
  /multisense/
    pointcloud_bridge (hrpsys_ros_bridge/PointCloudROSBridge)
    range_bridge (hrpsys_ros_bridge/RangeSensorROSBridge)
  /multisense/left/
    HEADLEFT (hrpsys_ros_bridge/ImageSensorROSBridge)
  /larm_camera/
    LARMCAMERA (hrpsys_ros_bridge/ImageSensorROSBridge)
  /
    AutoBalancerServiceROSBridge (hrpsys_ros_bridge/AutoBalancerServiceROSBridgeComp)
    DataLoggerServiceROSBridge (hrpsys_ros_bridge/DataLoggerServiceROSBridgeComp)
    EmergencyStopperServiceROSBridge (hrpsys_ros_bridge/EmergencyStopperServiceROSBridgeComp)
    ForwardKinematicsServiceROSBridge (hrpsys_ros_bridge/ForwardKinematicsServiceROSBridgeComp)
    HardEmergencyStopperServiceROSBridge (hrpsys_ros_bridge/EmergencyStopperServiceROSBridgeComp)
    HrpsysJointTrajectoryBridge (hrpsys_ros_bridge/HrpsysJointTrajectoryBridge)
[... 435 lines cut ...]
[modelloader-2] process has finished cleanly
log file: /home/leus/.ros/log/38d65a22-4a4d-11e5-8b16-cc3d825280be/modelloader-2*.log
[modelloader-2] restarting process
process[modelloader-2]: started with pid [20941]
IDL:omg.org/CORBA/TRANSIENT:1.0
[modelloader-2] process has finished cleanly
log file: /home/leus/.ros/log/38d65a22-4a4d-11e5-8b16-cc3d825280be/modelloader-2*.log
[modelloader-2] restarting process
process[modelloader-2]: started with pid [20944]
IDL:omg.org/CORBA/TRANSIENT:1.0
^C[SequencePlayerServiceROSBridge-13] killing on exit
[diagnostic_aggregator-9] killing on exit
[hrpsys_ros_diagnostics-8] killing on exit
[hrpsys_state_publisher-7] killing on exit
[HrpsysJointTrajectoryBridge-6] killing on exit
[HrpsysSeqStateROSBridge-5] killing on exit
[choreonoid-3] killing on exit
[modelloader-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
cannot add process [DataLoggerServiceROSBridge-14] after process monitor has been shut down
The traceback for the exception was written to the log file

```

YuOhara, 2015-08-24:
これは #15 と違う環境で実行していて、
#14 のようにchoreonoidが死にます

YoheiKakiuchi, 2015-08-24:
nameserviceは走っているかな？

ps aux | grep omni はどうだろうか？
omniNamesプロセスが無いようなら、
sudo ~/ros/indigo_parent/devel/bin/rtm-naming 2809

omniNamesプロセスがある場合でも、立ち上げ直すとどうなるかな？

あと、choreonoidが２つ立ち上がってしまったりしていないかな？

YuOhara, 2015-08-25:
sudo ~/ros/indigo_parent/devel/bin/rtm-naming 2809
で動きました

closed 2015-08-25 by YuOhara

k-okada, 2015-08-25:
2809 はデフォルトポートなので
sudo /etc/init.d/omniorb4-nameserver
でいいはず．ただ，ノートとかでIpが変わるとリブート必至なので
rtm-naming 2809でもよい．
上のようにフルパスは必要ない．必要だったrなんかおかしい

eisoku9618, 2016-01-14:
rtm-naming 2809

*ps aux | grep omni*

eisoku9618, 2016-01-14:
すみません，gmailでメモしていたのを送ってしまっていました．