start-jsk/rtmros_choreonoid #18: lleg_end_coordsなどがない

YuOhara, 2015-08-24:
odom_on_groundなどを出そうとしたのですが、実機であるlleg_end_coordsなどがないためそのままではプログラムが動かないです。
これって簡単に出せたりしますでしょうか？
```
roleus@leus-GB-BXi5G-760:~/ros/indigo/src/jsk-ros-pkg/trans_system/jaxon_ros_brie/launch$ roslaunch jsk_footstep_controller hrp2jsk_footcoords.launch
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
... logging to /home/leus/.ros/log/b675b962-4a59-11e5-a69c-fcaa1417ff45/roslaunch-leus-GB-BXi5G-760-12573.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://leus-GB-BXi5G-760:39322/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.13

NODES
  /
    footcoords (jsk_footstep_controller/footcoords)
    stabilizer_watcher (jsk_footstep_controller/stabilizer_watcher.py)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
process[footcoords-1]: started with pid [12592]
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
process[stabilizer_watcher-2]: started with pid [12706]
/home/leus/ros/indigo/src/jsk-ros-pkg/jsk_control/jsk_footstep_controller/scripts/stabilizer_watcher.py:61: SyntaxWarning: The publisher should be created with an explicit keyword argument 'queue_size'. Please see http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers for more information.
  g_odom_init_trigger_pub = rospy.Publisher("/odom_init_trigger", Empty)
/home/leus/ros/indigo/src/jsk-ros-pkg/jsk_control/jsk_footstep_controller/scripts/stabilizer_watcher.py:62: SyntaxWarning: The publisher should be created with an explicit keyword argument 'queue_size'. Please see http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers for more information.
  g_robotsound_pub = rospy.Publisher("/robotsound", SoundRequest)
[ERROR] [1440419015.538312435, 130.304000000]: [Footcoords::waitForEndEffectorTrasnformation] failed to lookup transform between BODY and lleg_end_coords
[ERROR] [1440419015.538369592, 130.304000000]: [Footcoords::computeMidCoords] Failed to lookup endeffector transformation
[ERROR] [1440419015.539371739, 130.304000000]: [Footcoords::publishContactState] transform error: "lleg_end_coords" passed to lookupTransform argument target_frame does not exist. 

```

garaemon, 2015-08-24:
今rosではなんのurdfを使っていますか

YuOhara, 2015-08-24:
実機のtfがこれで、
[image]
choreonoidで現状上がってるのがこれです。
[image]

YuOhara, 2015-08-24:
これでしょうか
```
leus@leus-GB-BXi5G-760:~/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/launch$ grep -r "urdf" *
jaxon_jvrc_ros_bridge_choreonoid.launch:    <arg name="COLLADA_FILE" value="$(find hrpsys_ros_bridge_jvrc)/models/JAXON_JVRC.urdf" />
```

garaemon, 2015-08-24:
`_JVRC`はなんで必要なんだろうか。
https://github.com/jsk-ros-pkg/jsk_control/blob/master/jsk_calibration/hrp2jsknts_calibration/README.md#appendix

`add_sensors_to_collada.py`でendcoordsとかがくっつきます

YoheiKakiuchi, 2015-08-24:
たしかに、end-coordsは付けてないですね。

YoheiKakiuchi, 2015-08-24:
https://github.com/start-jsk/rtmros_choreonoid/pull/22
これで対応しました。

closed 2015-08-27 by YoheiKakiuchi