start-jsk/rtmros_choreonoid #48: choreonoidが落ちてしまう.

otsubo, 2015-09-08:
```bash
roslaunch hrpsys_ros_bridge_jvrc jaxon_jvrc_choreonoid.launch 
```
を実行すると
```bash
[choreonoid-3] process has died [pid 117705, exit code 139, cmd /home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid/launch/run_choreonoid.sh /home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/jaxon_pd.cnoid --start-simulation -o manager.is_master:YES -o corba.nameservers:localhost:2809 -o naming.formats:%n.rtc -o logger.file_name:/tmp/rtc%p.log -endless -realtime -o manager.shutdown_onrtcs:NO -o manager.modules.load_path:/home/otsubo/ros/indigo_parent/devel/share/hrpsys/lib -o manager.modules.preload:.so -o manager.components.precreate: -o example.SequencePlayer.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.ForwardKinematics.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.ImpedanceController.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.AutoBalancer.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.StateHolder.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.TorqueFilter.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.TorqueController.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.ThermoEstimator.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.ThermoLimiter.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.VirtualForceSensor.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.AbsoluteForceSensor.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.RemoveForceSensorLinkOffset.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.KalmanFilter.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.Stabilizer.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.CollisionDetector.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.SoftErrorLimiter.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.HGcontroller.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.PDcontroller.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.EmergencyStopper.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf -o example.RobotHardware.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf __name:=choreonoid __log:=/home/otsubo/.ros/log/929daff2-563b-11e5-8cc6-a0d3c114f842/choreonoid-3.log].
log file: /home/otsubo/.ros/log/929daff2-563b-11e5-8cc6-a0d3c114f842/choreonoid-3*.log
```
というエラーが出て,choreonoidが落ちてしまいます.
どういったことが考えられるでしょうか.
以下全ログです.
```
otsubo@axis:~/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid$ roslaunch hrpsys_ros_bridge_jvrc jaxon_jvrc_choreonoid.launch 
... logging to /home/otsubo/.ros/log/929daff2-563b-11e5-8cc6-a0d3c114f842/roslaunch-axis-117651.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://axis:59637/

SUMMARY
========

PARAMETERS
 * /HrpsysSeqStateROSBridge/publish_sensor_transforms: True
 * /chest_camera/CHESTCAMERA/camera_param_K: [85.7444, 0, 319....
 * /chest_camera/CHESTCAMERA/camera_param_P: [85.7444, 0, 319....
 * /chest_camera/CHESTCAMERA/frame_id: CHEST_CAMERA
 * /controller_configuration: [{'group_name': '...
 * /diagnostic_aggregator/analyzers/computers/contains: ['HD Temp', 'CPU ...
 * /diagnostic_aggregator/analyzers/computers/path: Computers
 * /diagnostic_aggregator/analyzers/computers/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/hrpsys/contains: ['hrpEC', 'Emerge...
 * /diagnostic_aggregator/analyzers/hrpsys/path: Hrpsys
 * /diagnostic_aggregator/analyzers/hrpsys/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/mode/contains: ['Operating Mode'...
 * /diagnostic_aggregator/analyzers/mode/path: Mode
 * /diagnostic_aggregator/analyzers/mode/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/motor/contains: ['Motor']
 * /diagnostic_aggregator/analyzers/motor/path: Motor
 * /diagnostic_aggregator/analyzers/motor/type: diagnostic_aggreg...
 * /diagnostic_aggregator/base_path: 
 * /diagnostic_aggregator/pub_rate: 1.0
 * /larm_camera/LARMCAMERA/camera_param_K: [417.031, 0, 319....
 * /larm_camera/LARMCAMERA/camera_param_P: [417.031, 0, 319....
 * /larm_camera/LARMCAMERA/frame_id: LARM_CAMERA
 * /multisense/left/HEADLEFT/camera_param_K: [240, 0, 319.5, 0...
 * /multisense/left/HEADLEFT/camera_param_P: [240, 0, 319.5, 0...
 * /multisense/left/HEADLEFT/frame_id: multisense/left_c...
 * /multisense/pointcloud_bridge/frame_id: multisense/left_c...
 * /multisense/pointcloud_bridge/publish_depth: True
 * /multisense/pointcloud_bridge/transformed_camera_frame: True
 * /multisense/range_bridge/frame_id: HEAD_RANGE
 * /multisense/range_bridge/intensity: 1000
 * /rarm_camera/RARMCAMERA/camera_param_K: [879.192, 0, 319....
 * /rarm_camera/RARMCAMERA/camera_param_P: [879.192, 0, 319....
 * /rarm_camera/RARMCAMERA/frame_id: RARM_CAMERA
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.13
 * /rotate_range/rotate_cycle: 4.0
 * /rotate_range/rotate_times: 4
 * /use_sim_time: True

NODES
  /chest_camera/
    CHESTCAMERA (hrpsys_ros_bridge/ImageSensorROSBridge)
  /multisense/
    pointcloud_bridge (hrpsys_ros_bridge/PointCloudROSBridge)
    range_bridge (hrpsys_ros_bridge/RangeSensorROSBridge)
[... 700 lines cut ...]
[sensor_ros_bridge_connect-11] killing on exit
[hrpsys_profile-10] killing on exit
[diagnostic_aggregator-9] killing on exit
[hrpsys_ros_diagnostics-8] killing on exit
[hrpsys_state_publisher-7] killing on exit
Traceback (most recent call last):
  File "/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_common/hrpsys_ros_bridge/scripts/sensor_ros_bridge_connect.py", line 60, in <module>
    initSensorRosBridgeConnection(sys.argv[1], sys.argv[2], sys.argv[3], sys.argv[4])
  File "/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_common/hrpsys_ros_bridge/scripts/sensor_ros_bridge_connect.py", line 41, in initSensorRosBridgeConnection
    hcf.waitForRTCManagerAndRoboHardware(simulator_name, managerhost)
  File "/home/otsubo/ros/indigo_parent/devel/lib/python2.7/dist-packages/hrpsys/hrpsys_config.py", line 900, in waitForRTCManagerAndRoboHardware
[HrpsysJointTrajectoryBridge-6] killing on exit
    self.waitForRobotHardware(robotname)
  File "/home/otsubo/ros/indigo_parent/devel/lib/python2.7/dist-packages/hrpsys/hrpsys_config.py", line 856, in waitForRobotHardware
    time.sleep(1);
KeyboardInterrupt
[HrpsysSeqStateROSBridge-5] killing on exit
[ INFO] [1441724979.304967974]: [HrpsysSeqStateROSBridge] @onFinalize : HrpsysSeqStateROSBridge0
[modelloader-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
cannot add process [HardEmergencyStopperServiceROSBridge-24] after process monitor has been shut down
The traceback for the exception was written to the log file
otsubo@axis:~/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid$ 
```

otsubo, 2015-09-08:
#14 と違って,一度choreonoidが上がるのですが,すぐに落ちてしまいます.

snozawa, 2015-09-08:
見ている箇所はすごく惜しいですね。

`[choreonoid-3] ....`の部分はroslaunchの出力しているもので、起動したノードがしんでますというものです。
ログのその１行上にヒントがあります。
数行分くらい上をみると
```
[PDcontroller0] Gain file [/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.PDgains.dat] opened
process[rtmlaunch_hrpsys_ros_bridge-12]: started with pid [118073]
/home/boots/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid/launch/run_choreonoid.sh: line 45: 117713 Segmentation fault      (core dumped) choreonoid $cnoid_proj $start_sim
```
とありますね。
一番最後の行で`run_choreonoid.sh`の45行目のchoreonoidの箇所でエラーがでてる。

次は、`run_choreonoid.sh`が`rtmros_choreonoid/hrpsys_choreonoid/launch`以下にあるとエラー文にでてるので、`run_choreonoid.sh`をみてみる。
と、
```
(cd /tmp; choreonoid $cnoid_proj $start_sim)
```
とあって、エラーにある行がでてくる。

ということで、例えばgdbを
```
(cd /tmp; gdb -ex run --args choreonoid $cnoid_proj $start_sim)
```
のようにして実行してみるとどうなるかな？

YoheiKakiuchi, 2015-09-08:
不真面目デバッグとしては，

choreonoidのプロセスが残っていて，2重立ち上げになっている．

以下を忘れている．
~~~
sudo mkdir -p /usr/lib/choreonoid-1.5/customizer; sudo mkdir /usr/lib/choreonoid-1.5/rtc # if needed
roscd hrpsys_ros_bridge_jvrc; sudo cp config/SensorReaderRTC.*conf /usr/lib/choreonoid-1.5/rtc
sudo cp ~/ros/indigo_parent/devel/lib/{HG,PD}controller.so /usr/lib/choreonoid-1.5/rtc
~~~

hrpsys_choreonoidのmakeの時にルートパスワードを打っていない

くらいでしょうか．

choreonoid単体で立ち上がるか (choreonoidと打ってみる)

choreonoid.launchのstart_simulatorをfalseにすると，自動的にシミュレーションが実行されないのでchoreonoidのメッセージがよく読める．
https://github.com/start-jsk/rtmros_choreonoid/blob/master/hrpsys_choreonoid/launch/choreonoid.launch

jaxon_jvrc_choreonoid.launch ではなくて， jaxon_jvrc_startup_choreonoid.launch にするとブリッジが立ち上がらないので，
メッセージがへってちょっと分かりやすい．

otsubo, 2015-09-08:
@snozawa さん
ありがとうございます.
```bash
(cd /tmp; gdb -ex run --args choreonoid $cnoid_proj $start_sim)
```
で実行してみると,エラー文はでないのですが,choreonoidの画面が暗くなって
```bash
[ WARN] [1441728882.266859367, 0.002000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is not executed last 5[sec]
```
のようなwarningが流れ続けます.
以下全ログです.
```
otsubo@axis:~$ roslaunch hrpsys_ros_bridge_jvrc jaxon_jvrc_choreonoid.launch 
... logging to /home/otsubo/.ros/log/a6b92e3e-5646-11e5-b485-a0d3c114f842/roslaunch-axis-87260.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://axis:35693/

SUMMARY
========

PARAMETERS
 * /HrpsysSeqStateROSBridge/publish_sensor_transforms: True
 * /chest_camera/CHESTCAMERA/camera_param_K: [85.7444, 0, 319....
 * /chest_camera/CHESTCAMERA/camera_param_P: [85.7444, 0, 319....
 * /chest_camera/CHESTCAMERA/frame_id: CHEST_CAMERA
 * /controller_configuration: [{'group_name': '...
 * /diagnostic_aggregator/analyzers/computers/contains: ['HD Temp', 'CPU ...
 * /diagnostic_aggregator/analyzers/computers/path: Computers
 * /diagnostic_aggregator/analyzers/computers/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/hrpsys/contains: ['hrpEC', 'Emerge...
 * /diagnostic_aggregator/analyzers/hrpsys/path: Hrpsys
 * /diagnostic_aggregator/analyzers/hrpsys/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/mode/contains: ['Operating Mode'...
 * /diagnostic_aggregator/analyzers/mode/path: Mode
 * /diagnostic_aggregator/analyzers/mode/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/motor/contains: ['Motor']
 * /diagnostic_aggregator/analyzers/motor/path: Motor
 * /diagnostic_aggregator/analyzers/motor/type: diagnostic_aggreg...
 * /diagnostic_aggregator/base_path: 
 * /diagnostic_aggregator/pub_rate: 1.0
 * /larm_camera/LARMCAMERA/camera_param_K: [417.031, 0, 319....
 * /larm_camera/LARMCAMERA/camera_param_P: [417.031, 0, 319....
 * /larm_camera/LARMCAMERA/frame_id: LARM_CAMERA
 * /multisense/left/HEADLEFT/camera_param_K: [240, 0, 319.5, 0...
 * /multisense/left/HEADLEFT/camera_param_P: [240, 0, 319.5, 0...
 * /multisense/left/HEADLEFT/frame_id: multisense/left_c...
 * /multisense/pointcloud_bridge/frame_id: multisense/left_c...
 * /multisense/pointcloud_bridge/publish_depth: True
 * /multisense/pointcloud_bridge/transformed_camera_frame: True
 * /multisense/range_bridge/frame_id: HEAD_RANGE
 * /multisense/range_bridge/intensity: 1000
 * /rarm_camera/RARMCAMERA/camera_param_K: [879.192, 0, 319....
 * /rarm_camera/RARMCAMERA/camera_param_P: [879.192, 0, 319....
 * /rarm_camera/RARMCAMERA/frame_id: RARM_CAMERA
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.13
 * /rotate_range/rotate_cycle: 4.0
 * /rotate_range/rotate_times: 4
 * /use_sim_time: True

NODES
  /chest_camera/
    CHESTCAMERA (hrpsys_ros_bridge/ImageSensorROSBridge)
  /multisense/
    pointcloud_bridge (hrpsys_ros_bridge/PointCloudROSBridge)
    range_bridge (hrpsys_ros_bridge/RangeSensorROSBridge)
[... 2416 lines cut ...]
(gdb) [ INFO] [1441729341.491509097, 0.002000000]: [HrpsysJointTrajectoryBridge0] @~jointTrajectoryActionObj (larm
[ INFO] [1441729341.491804001, 0.002000000]: [HrpsysSeqStateROSBridge] @onFinalize : HrpsysSeqStateROSBridge0
[modelloader-2] killing on exit
[ INFO] [1441729341.508904533, 0.002000000]: [HrpsysJointTrajectoryBridge0] @~jointTrajectoryActionObj (rarm
[ INFO] [1441729341.525133287, 0.002000000]: [HrpsysJointTrajectoryBridge0] @~jointTrajectoryActionObj (lleg
[ INFO] [1441729341.543624688, 0.002000000]: [HrpsysJointTrajectoryBridge0] @~jointTrajectoryActionObj (rleg
[ INFO] [1441729341.562668377, 0.002000000]: [HrpsysJointTrajectoryBridge0] @~jointTrajectoryActionObj (torso
[ INFO] [1441729341.576606500, 0.002000000]: [HrpsysJointTrajectoryBridge0] @~jointTrajectoryActionObj (head
[ INFO] [1441729341.588659349, 0.002000000]: [HrpsysJointTrajectoryBridge0] @~jointTrajectoryActionObj (rhand
[ INFO] [1441729341.600312707, 0.002000000]: [HrpsysJointTrajectoryBridge0] @~jointTrajectoryActionObj (lhand
[ INFO] [1441729341.611557085, 0.002000000]: [HrpsysJointTrajectoryBridge0] @~jointTrajectoryActionObj (range
[SequencePlayerServiceROSBridge-13] escalating to SIGTERM
[rtmlaunch_hrpsys_ros_bridge-12] escalating to SIGTERM
[HrpsysSeqStateROSBridge-5] escalating to SIGTERM
[HrpsysJointTrajectoryBridge-6] escalating to SIGTERM
[hrpsys_py-4] escalating to SIGTERM
[choreonoid-3] escalating to SIGTERM
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
otsubo@axis:~/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid/launch$ 
```
choreonoid をstartしたところで死んでしまっているようなのですが,これは#14と同じ現象の可能性があるでじょうか.

YoheiKakiuchi, 2015-09-08:
jaxon_jvrc_startup_choreonoid.launch を使って，
https://github.com/start-jsk/rtmros_choreonoid/blob/master/hrpsys_ros_bridge_jvrc/launch/jaxon_jvrc_startup_choreonoid.launch

以下を消すと，hrpsys が立ち上がらなくなるので原因がはっきりすると思います．
```<arg name="HRPSYS_PY_PKG" default="hrpsys_ros_bridge_jvrc" />```

できれば，choreonoid内のメッセージになにか出ていないか見てみて下さい．

otsubo, 2015-09-09:
```
<arg name="HRPSYS_PY_PKG" default="hrpsys_ros_bridge_jvrc" />

```
を消して,jaxon_jvrc_startup_choreonoid.launchを実行すると,上と同じプロセスが死んでしまいます.
```
otsubo@axis:~/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/launch$ roslaunch hrpsys_ros_bridge_jvrc jaxon_jvrc_startup_choreonoid.launch 
... logging to /home/otsubo/.ros/log/8f6f9e36-56cb-11e5-acfd-a0d3c114f842/roslaunch-axis-5281.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://axis:49233/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.13
 * /use_sim_time: True

NODES
  /
    choreonoid (hrpsys_choreonoid/run_choreonoid.sh)
    hrpsys_py (hrpsys_tools/jaxon_jvrc_hrpsys_config.py)
    modelloader (openhrp3/openhrp-model-loader)

auto-starting new master
process[master]: started with pid [5293]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 8f6f9e36-56cb-11e5-acfd-a0d3c114f842
process[rosout-1]: started with pid [5306]
started core service [/rosout]
process[modelloader-2]: started with pid [5330]
ready
process[choreonoid-3]: started with pid [5334]
$ choreonoid /home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/jaxon_pd.cnoid
with rtc.conf file on /tmp/rtc.conf.choreonoid
<BEGIN: rtc.conf>
manager.is_master:YES
corba.nameservers:localhost:2809
naming.formats:%n.rtc
logger.file_name:/tmp/rtc%p.log
manager.shutdown_onrtcs:NO
manager.modules.load_path:/home/otsubo/ros/indigo_parent/devel/share/hrpsys/lib
manager.modules.preload:.so
manager.components.precreate:
example.SequencePlayer.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf
example.ForwardKinematics.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf
example.ImpedanceController.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf
example.AutoBalancer.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf
example.StateHolder.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf
example.TorqueFilter.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf
example.TorqueController.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf
example.ThermoEstimator.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf
example.ThermoLimiter.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf
example.VirtualForceSensor.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf
example.AbsoluteForceSensor.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf
example.RemoveForceSensorLinkOffset.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf
example.KalmanFilter.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf
example.Stabilizer.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf
example.CollisionDetector.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf
example.SoftErrorLimiter.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf
example.HGcontroller.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf
example.PDcontroller.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf
example.EmergencyStopper.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf
example.RobotHardware.config_file:/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.conf
<END: rtc.conf>
[... 98 lines cut ...]
Loading body customizer "/home/otsubo/ros/indigo_parent/devel/share/OpenHRP-3.1/customizer/libSampleCustomizer.so" for springJoint
create customizer : JAXON_JVRC
Warning : Three or more triangles is defined for a edge.
cloneMap.setNonNodeCloning(false);
cloneMap.setNonNodeCloning(false);
cloneMap.setNonNodeCloning(false);
cloneMap.setNonNodeCloning(false);
cloneMap.setNonNodeCloning(false);
PDcontroller0: on Activated 
[PDcontroller0] Gain file [/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/models/JAXON_JVRC.PDgains.dat] opened
*** Error in `choreonoid': double free or corruption (!prev): 0x000000000282a170 ***
/home/otsubo/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid/launch/run_choreonoid.sh: line 45:  5342 Aborted                 (core dumped) choreonoid $cnoid_proj $start_sim
[... cut ...] 3c114f842/choreonoid-3.log].
log file: /home/otsubo/.ros/log/8f6f9e36-56cb-11e5-acfd-a0d3c114f842/choreonoid-3*.log
^C[modelloader-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```
choreonoid内のメッセージはmodelがloadされたというメッセージが出て特に問題なさそうです.
(貼り付けようとしたのですが,コピーする前に落ちてしまいます.)
jaxon_jvrc_startup.launchを立ち上げてみたのですが,やはり同じプロセスが死んでしまいます.
いったん,他の動いてるPCで作業することにしようと思うのですが,もし他に確認すべきことがあれば教えていただけますでしょうか.

YoheiKakiuchi, 2015-09-09:
choreonoid単体では立ち上がる？

シミュレーションをスタートしなければ立ち上がる？ （おちるのはシミュレーションをスタートした後）

カンでいくと、nvidiaは正しく入っているかな？
glxinfo で何が表示される？ (sudo apt-get install mesa-utilsが必要かも)

otsubo, 2015-09-09:
上から純に確認してみたところ,nvidiaが正しく入っていませんでした.
システム設定から追加のドライバーを入れるとchoreonoidがちゃんと立ち上がりました.ありがとうございました.

closed 2015-09-10 by otsubo