start-jsk/rtmros_choreonoid #50: send *ri* :wait-interpolationが帰ってこない

YuOhara, 2015-09-09:
目標関節角にchoreonoid及びrvizでは送られているように見えますが、だいぶ待たないと帰ってこないです

k-okada, 2015-09-09:
joint trajectory action の goal が帰ってきているか見てみて下さい．

YuOhara, 2015-09-09:
かなり遅いですが
 /fullbody_controller/joint_trajectory_action/result
は帰ってきていて、そのだんかいでwait-interpolationを抜けています。

YuOhara, 2015-09-09:
```
header: 
  seq: 9
  stamp: 
    secs: 1497
    nsecs: 793000000
  frame_id: ''
status: 
  goal_id: 
    stamp: 
      secs: 1482
      nsecs: 136000000
    id: /default_robot_interface_1441798677619176060_4449_fullbody_controller/joint_trajectory_action_9
  status: 3
  text: ''
result: 
```

k-okada, 2015-09-09:
次はhrpsysでみてみよう．こちらはseqの中でprint debug するしかないかな．

snozawa, 2015-09-09:
`rtconf localhost:15005/seq.rtc set debugLevel 1`
はどうでしょうか。

https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/SequencePlayer/SequencePlayer.cpp#L223
のprint文がでてくると思います。
1000ループに１回でprint文がでるというif文なので、
JAXONがdt=0.002 => 1秒に500ループとすると、
２秒に１回のペースでprint文がでてくると思います。
４秒くらいでおくったら２回でるはず、といったかたちで:angle-vectorの時間とprint文の回数があっていそうか確認してみてください。

YuOhara, 2015-09-09:
```
virtual RTC::ReturnCode_t SequencePlayer::onExecute(RTC::UniqueId)(1000)
```
多分これですよね
これってanglevector送る前から常に出ているように見えますが、どうすればよいでしょうか？

snozawa, 2015-09-09:
:angle-vectorを送りつつhrpsys-baseの出力をみて、
"遅れてwait-interpolationがかえってくる"タイミングとonExecuteの出力を見比べて見てください。
４秒でおくったら４秒（2回）になってるか

YoheiKakiuchi, 2015-09-09:
思い当たりました．
rangeセンサを回しているんですが，外から常に回すように groupにjoint_trajectoryを送っています．
fullbodyはすべてのseqがemptyでないとemptyを返さないので，良いタイミングがないとSeqStateROSBridgeでemptyを見てresultを返しているのでactionが終了しないことになります．

作戦としてはfullbody使わない方向かな．
ハンドも同じ問題が起きそうだが，ハンドのwaitに全身が引きずられてもそれほど問題ないか．

k-okada, 2015-09-09:
rangeだけのseqを作るということかな．

YoheiKakiuchi, 2015-09-10:
ハンドとrangeを抜いたgroupを作ろうと思っていました。
ROS以降はあまり問題にならないように思います。
hrpsysでzmpなどのseqを使おうと思うと同じような問題になりそうです。

PDコントローラをいじって、rangeの関節だけ一定速度をつねに保つようにするのが、次点の作戦です。

closed 2015-09-17 by YuOhara