start-jsk/rtmros_choreonoid #52: /imuのheaderのstampが固定になってしまう

YuOhara, 2015-09-13:
/odomと/imuをシンクするコードがjskのpkgにあるのですが、
/odomのstampはちゃんと出ていますが、
/imuは固定されたものが出続けます

YoheiKakiuchi, 2015-09-13:
/imuを出しているノードはどれかな？

YuOhara, 2015-09-13:
```
leus@leus-ThinkPad-W541:~$ rostopic info /odom
Type: nav_msgs/Odometry

Publishers: 
 * /HrpsysSeqStateROSBridge (http://leus-ThinkPad-W541:36825/)

Subscribers: 
 * /footcoords (http://leus-ThinkPad-W541:40296/)

leus@leus-ThinkPad-W541:~$ rostopic info /imu
Type: sensor_msgs/Imu

Publishers: 
 * /HrpsysSeqStateROSBridge (http://leus-ThinkPad-W541:36825/)

Subscribers: 
 * /footcoords (http://leus-ThinkPad-W541:40296/)

```

YuOhara, 2015-09-13:
値は入っています
```
header: 
  seq: 137
  stamp: 
    secs: 1011025424
    nsecs: 23853095
  frame_id: gyrometer
orientation: 
  x: 0.00203978037057
  y: -0.00340075006444
  z: -0.0196239669805
  w: 0.999799566971
orientation_covariance: [2.89e-08, 0.0, 0.0, 0.0, 2.89e-08, 0.0, 0.0, 0.0, 2.89e-08]
angular_velocity: 
  x: -0.0373868733669
  y: -0.0432278635694
  z: -0.0508558777606
angular_velocity_covariance: [0.000144, 0.0, 0.0, 0.0, 0.000144, 0.0, 0.0, 0.0, 0.000144]
linear_acceleration: 
  x: 3.84794854475
  y: -0.531724271094
  z: 11.8750414193
linear_acceleration_covariance: [0.0096, 0.0, 0.0, 0.0, 0.0096, 0.0, 0.0, 0.0, 0.0096]
---
^Cheader: 
  seq: 138
  stamp: 
    secs: 1011025424
    nsecs: 23853095
  frame_id: gyrometer
orientation: 
  x: 0.00209254769965
  y: -0.00262182476552
  z: -0.0196747999404
  w: 0.999800804924
orientation_covariance: [2.89e-08, 0.0, 0.0, 0.0, 2.89e-08, 0.0, 0.0, 0.0, 2.89e-08]
angular_velocity: 
  x: 0.033443046304
  y: -0.42538151545
  z: -0.0127360788369
angular_velocity_covariance: [0.000144, 0.0, 0.0, 0.0, 0.000144, 0.0, 0.0, 0.0, 0.000144]
linear_acceleration: 
  x: 7.13172377133
  y: -0.00245147162261
  z: 10.3683487179
linear_acceleration_covariance: [0.0096, 0.0, 0.0, 0.0, 0.0096, 0.0, 0.0, 0.0, 0.0096]
---

```

YoheiKakiuchi, 2015-09-13:
ちゃんと見てみますが，hrpsysで入っていないんじゃないかと思います．

ここを通って，値が入っていると思います．
https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/src/HrpsysSeqStateROSBridge.cpp#L671

YoheiKakiuchi, 2015-09-13:
kfでタイムスタンプが入っていないようです。以下で修正されます。
https://github.com/fkanehiro/hrpsys-base/pull/807

YuOhara, 2015-09-13:
> kfでタイムスタンプが入っていないようです。以下で修正されます。
> fkanehiro/hrpsys-base#807

ありがとうございます

YuOhara, 2015-09-14:
stampが出て、いい感じに見えます。
jsk-ros-pkgのfootcoordsもちゃんと動いているようです

closed 2015-09-14 by YuOhara