start-jsk/rtmros_choreonoid #53: laserについて

YuOhara, 2015-09-14:
jsk_tilt_laser/launch/multisense_laser_pipeline.launchをあげるにあたって、
以下の点で修正が必要なことがわかりました。
* joint_statesのvelocityでmotor_jointの値が出ていない

YuOhara, 2015-09-14:
加えてですが、
* laserのframe_idをhead_hokuyo_frameにする
* laserのintensityを4くらいにする
ができればうれしいのですが、後者は簡単そうですが、
前者は
https://github.com/start-jsk/rtmros_choreonoid/blob/master/hrpsys_ros_bridge_jvrc/launch/vision_connect.launch#L70
を変えるだけではダメっぽく見えるのですがどうでしょうか？

YoheiKakiuchi, 2015-09-14:
> joint_statesのvelocityでmotor_jointの値が出ていないのを出したい

とりあえず、上の問題については以下で大丈夫なはず。
https://github.com/start-jsk/rtmros_common/pull/821

YoheiKakiuchi, 2015-09-14:
> 変えるだけではダメっぽく見えるのですがどうでしょうか？

どういうことかな？ 
rangeのframe_idは変わるけど、フレームがないということかな？

YuOhara, 2015-09-14:
> rangeのframe_idは変わるけど、フレームがないということかな？
そうです。
でもよくみるとカメラ用にstatic_transform使ってるので、それと同じことをすればいいんですね。

YoheiKakiuchi, 2015-09-14:
> でもよくみるとカメラ用にstatic_transform使ってるので、それと同じことをすればいいんですね。

そうです。
ですが、urdfに加えたほうが問題なさそうなのでカメラも含めてそうしようかと思います。

YuOhara, 2015-09-14:
わかりました。ありがとうございます

YuOhara, 2015-09-14:
すいません。
intensityに関しては１０００のままで大丈夫でした

YoheiKakiuchi, 2015-09-14:
> ですが、urdfに加えたほうが問題なさそうなのでカメラも含めてそうしようかと思います。

と思ったのですが、urdfでセンサのリンクがつながっていない(tfはHrpsysSeqStateROSBridgeが出す)のでstatic_tf 作戦にしました。
https://github.com/start-jsk/rtmros_choreonoid/pull/55

closed 2015-09-16 by YuOhara