start-jsk/rtmros_choreonoid #71: hrpsys_ros_bridge_jvrcのコンパイルが通らない

YutaKojio, 2015-09-18:
catkin build hrpsys_ros_bridge_jvrcが通りません．
なにか解決策はありますでしょうか．
```
Starting ==> hrpsys_ros_bridge_jvrc                                             
[hrpsys_ros_bridge_jvrc] ==> '/home/gordon/ros/indigo/build/hrpsys_ros_bridge_jvrc/build_env.sh /usr/bin/make cmake_check_build_system' in '/home/gordon/ros/indigo/build/hrpsys_ros_bridge_jvrc' 
[hrpsys_ros_bridge_jvrc] <== '/home/gordon/ros/indigo/build/hrpsys_ros_bridge_jvrc/build_env.sh /usr/bin/make cmake_check_build_system' finished with return code '0' 
[hrpsys_ros_bridge_jvrc] ==> '/home/gordon/ros/indigo/build/hrpsys_ros_bridge_jvrc/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' in '/home/gordon/ros/indigo/build/hrpsys_ros_bridge_jvrc' 
[  0%] [  7%] [ 22%] [ 23%] Built target dynamixel_controllers_generate_messages_eus 
Built target jaxon_jvrc_hrpsys_ros_bridge_jvrc_compile_eusif                    
Built target household_objects_database_msgs_generate_messages_eus              
Generating /usr/lib/choreonoid-1.5/rtc/SensorReaderRTC.PD.conf                  
[ 24%] [ 37%] [ 63%] Built target dynamixel_msgs_generate_messages_eus          
Built target jaxon_jvrc_hrpsys_ros_bridge_jvrc_compile_dae                      
[ 88%] [ 93%] Built target manipulation_msgs_generate_messages_eus              
Built target jaxon_jvrc_hrpsys_ros_bridge_jvrc_compile_top_launch               
Built target hrpsys_ros_bridge_jvrc_generate_messages_eus                       
sudo:                                                                           
make[2]: *** [/usr/lib/choreonoid-1.5/rtc/SensorReaderRTC.PD.conf]  1           
make[1]: *** [CMakeFiles/copy_bodyrtc.dir/all]  2                               
make[1]: *** ....                                                               
Scanning dependencies of target jaxon_jvrc_hrpsys_ros_bridge_jvrc_compile_joint_minmax 
Scanning dependencies of target jaxon_jvrc_hrpsys_ros_bridge_jvrc_compile_xml   
Scanning dependencies of target jaxon_jvrc_hrpsys_ros_bridge_jvrc_compile_urdf  
[ 93%] [ 94%] [ 95%] Built target JAXON_JVRC_SENSORS.urdf_generate              
[ 97%] Built target jaxon_jvrc_hrpsys_ros_bridge_jvrc_compile_lisp              
Built target jaxon_jvrc_hrpsys_ros_bridge_jvrc_compile_xml                      
[ 97%] Generating jaxon_jvrc_joint_minmax_done                                  
Built target jaxon_jvrc_hrpsys_ros_bridge_jvrc_compile_urdf                     
Xserver connection failed;; JAXON_JVRC already has min-max table.               
[ 98%] Generating jaxon_jvrc_joint_minmax_conf_done                             
Xserver connection failed[100%] Built target jaxon_jvrc_hrpsys_ros_bridge_jvrc_compile_joint_minmax 
make: *** [all]  2                                                              
[hrpsys_ros_bridge_jvrc] <== '/home/gordon/ros/indigo/build/hrpsys_ros_bridge_jvrc/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' failed with return code '2' 
Failed   <== hrpsys_ros_bridge_jvrc      [ 37.0 seconds ]                       
[build] There were '1' errors:                                                  

Failed to build package 'hrpsys_ros_bridge_jvrc' because the following command:

# Command to reproduce:
cd /home/gordon/ros/indigo/build/hrpsys_ros_bridge_jvrc && /home/gordon/ros/indigo/build/hrpsys_ros_bridge_jvrc/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j; cd -

# Path to log:
cat /home/gordon/ros/indigo/build/build_logs/hrpsys_ros_bridge_jvrc.log

Exited with return code: 2 
                                                                                
[build] Runtime: 56.8 seconds 
gordon@gordon-desktop:~/ros/indigo$ b
```

garaemon, 2015-09-18:
sudoで止まってますね。
https://github.com/start-jsk/rtmros_choreonoid/pull/58

YutaKojio, 2015-09-18:
```
gordon@gordon-desktop:~/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc$ git diff CMakeLists.txt
diff --git a/hrpsys_ros_bridge_jvrc/CMakeLists.txt b/hrpsys_ros_bridge_jvrc/CMak
index 3f7e55e..ffe84fb 100644
--- a/hrpsys_ros_bridge_jvrc/CMakeLists.txt
+++ b/hrpsys_ros_bridge_jvrc/CMakeLists.txt
@@ -149,6 +149,7 @@ install(CODE
 ")
 
 add_custom_command(OUTPUT /usr/lib/choreonoid-1.5/rtc/SensorReaderRTC.PD.conf
+  COMMAND sudo -n mkdir /usr/lib/choreonoid-1.5/rtc/
   COMMAND sudo -n cp ${PROJECT_SOURCE_DIR}/config/SensorReaderRTC.PD.conf /usr/
   )
 add_custom_target(copy_bodyrtc ALL DEPENDS /usr/lib/choreonoid-1.5/rtc/SensorRe
(END)
```
で通りました

YutaKojio, 2015-09-18:
hrpsys_choreonoidが先にコンパイルされるはずなのですが、
https://github.com/start-jsk/rtmros_choreonoid/blob/master/hrpsys_choreonoid/CMakeLists.txt#L22
でifがfalseになっていると考えられるのですが、これは普通な挙動なのでしょうか？

YuOhara, 2015-09-18:
すいません。
libcnoid-dev
を入れてないせいでした

YutaKojio, 2015-09-18:
動きました。ありがとうございます。

closed 2015-09-18 by YutaKojio