start-jsk/rtmros_choreonoid #76: jaxonの足にものが

YuOhara, 2015-09-23:


YuOhara, 2015-09-23:
https://github.com/jsk-ros-pkg/jsk_control/issues/457
の続きです。

* jaxonで不整地に斜めに立つとrealtime-factor が小さくなる。

というのが再現性を持って起きていそうです。
また、

* TASKR4でも、ブロックと足がぶつかったときに同じようにrealtime-factorが落ちる

と @k-kimura さんが言っていました。ちなみにrealtime-factorは10分の1以下に落ちていました。

それで、ここからは完全に想像なのですが、
もしかして、あし裏あるいは足側面のmeshが細かかったりすると、そのような挙動になるのでしょうか？

snozawa, 2015-09-23:
https://github.com/s-nakaoka/choreonoid
にきいてみましょう．
一般論としては，接触点数が増えてきたらシミュレーション演算が重くなるので，遅くなると思います．
> もしかして、あし裏あるいは足側面のmeshが細かかったりすると、そのような挙動になるのでしょうか？

これはありうると思います．

YuOhara, 2015-09-23:
ありがとうございました。

gitでなくなってしまったのですが、わりと急ぎなのでCybozuLive（JVRC情報交換フォーラム）にひとまず投稿しました。
あとでissueを投げる形でchoreonoid-userの方々にも情報は共有したいと思います。

snozawa, 2015-09-23:
はい，困ったら本家で最速で質問するのが重要ですね．

YoheiKakiuchi, 2015-09-23:
https://github.com/start-jsk/rtmros_choreonoid/pull/77
choreonoidで使うモデルをconvex hull メッシュにしてみました。
劇的に早くなっていないように見えます。

元のモデルを使いたいときは、 hrpsys_ros_bridge_jvrc/config以下になる ```.cnoid```  ファイルの
JAXON_JVRCモデルをロードするディレクトリを元に戻して見てください。

あと、realtime-factorはgazebo用語なので定義から書くと分かりやすいかもしれません。

YuOhara, 2015-09-23:
ありがとうございます。

@k-kimuraさん。 @YutaKojio
アップデートして試してみてください。

>あと、realtime-factorはgazebo用語なの
>で定義から書くと分かりやすいかもしれません。
なるほど、ありがとうございます。。

k-kimura, 2015-09-23:
ありがとうございます.
R4タスクでconvex hullのモデルを使ってみました.
ロボットが物体に近づいてもスローになることはなくなりました.

添付のような物体とロボットが近接干渉しているときの
/joint_statesトピックのrateを見てみました.
[image]
```
average rate: 191.231
	min: 0.000s max: 0.034s std dev: 0.00937s window: 9605
average rate: 191.337
	min: 0.000s max: 0.034s std dev: 0.00937s window: 9701
average rate: 191.348
	min: 0.000s max: 0.034s std dev: 0.00937s window: 9803
average rate: 191.139
	min: 0.000s max: 0.034s std dev: 0.00937s window: 9902
average rate: 191.162
	min: 0.000s max: 0.034s std dev: 0.00937s window: 9998
```
以前のロボットモデルの場合,average rateが1000(Max)ほど出ていましたが,
今回のconvex hullよりは大分遅かったので改善されている気がします.

snozawa, 2015-09-23:
あれ？黒地に赤もかっこいい。

YoheiKakiuchi, 2015-09-24:
どれぐらい、改善しているのかがよくわからない。
PCによってスタート地点で立っているだけの時の計算時間がすでに違うから、
PCのスペック、スタート地点で立っている時のrate、遅くなった時のrate 
についてはどうだろうか？

余裕があったら、メッシュがconvex hullの時と詳細の時でどう違うかも見てみてください。

また、joint_statesのhzで計測する場合には、以下のようにwindowを決めよう
~~~
rostopic hz -w 200 /joint_states
~~~

k-kimura, 2015-09-24:
計算に使っているPCは,レインのsilverstone(キューブ型PC)で
グラフィックボードがGeForce GTX 780 Tiで,CPUはIntel Core i7-4790になります.

この条件でスタート時と物体に近づいた時(遅くなった時)の場合の
詳細モデルとconvex hullモデルのrate比較をおこないました.

[convex hullモデル]
* スタート時
rate: 140[Hz]前後
* 物体に近づいた時
rate: 200~250[Hz]

[以前の詳細モデル]
* スタート時
rate: 160[Hz]前後
* 物体に近づいた時
rate: 1000[Hz]

snozawa, 2015-09-24:
これデータ逆？詳細モデルのほうがrateが速い？

garaemon, 2015-09-24:
rostopic hzで測るのであってる?
use_sim_timeがtrueだと図りたいものが履かれていない気がするけど

YoheiKakiuchi, 2015-09-24:
https://github.com/start-jsk/rtmros_choreonoid/pull/80
で、教えてもらったsurfaceノードを導入しました。

collisionの形状は jvrc_models/JAXON_JVRC/collision_directoryのシンボリックリンクを張り替えると変更可能です。デフォルトはconvex_hullになってます。
あとは、これより早くなることはないbounding_box形状もあります。
collision_directoryがjvrc_models/JAXON_JVRCを指すとvisualとcollisionが一致します。

> なお、計算用の形状は簡略化しつつも、描画には精細なモデルを使用されたいという場合には、Surfaceという独自定義のVRMLノードが最近導入されておりますので、こちらをお使いいただくことができます。サンプルは以下のURLを御覧ください。
https://github.com/s-nakaoka/choreonoid/blob/master/share/model/GR001/GR001.wrl#L338

collisionとvisualの確認方法は以下

> collisionの形状とvisualの形状を確認することはできますでしょうか？

メインメニューの "Option" - "Scene View" に
"Show visual shapes", "Show collision shapes" という項目があります。
それぞれチェックをON/OFFすることで表示を切り替えられます。

接触の確認方法は以下

>シミュレーション中の接触点や接触力を視覚で確認する方法はありますでしょうか？ 

接触点は、ワールドアイテムの干渉検出プロパティをTrueにして、ツールバーのシーンバーにある、緑色の星形のアイコンのついたボタンを押して頂くと表示されるようになります。

YoheiKakiuchi, 2015-09-24:
> rostopic hzで測るのであってる?
> use_sim_timeがtrueだと図りたいものが履かれていない気がするけど

体感ではあっていそう。
SeqStateROSBridgeが100Hz(現実時間)で回る（ようにしている）のでChoreonoidがそれ以上の周期で回ると、/joint_statesトピックは100Hz(現実時間)で出てくる。
hzで表示されるのは、シミュレーション時間で計測したHzになっていて、現実時間よりシミュレーション時間が遅れるとHz表示が大きくなっていく。

だけど、choreonoidが100Hzで回らなくなると、choreonoidが回っているシミュレーション時間の周期に張り付いてしまう。それが1000Hzですね。なので、1000Hz未満の値はそこそこ信頼していいのではないかと思います。

YuOhara, 2016-01-07:
* とりあえずモデルを変えたおかげで、jvrc本番で使えるくらいには解決していました。
* シミュレーション時間が遅い現象は他のチームもある程度見られました。

closed 2016-01-07 by YuOhara