start-jsk/rtmros_choreonoid #79: JAXONとcollada2eusによってできたモデルファイルをロードしてinitするjaxonとエラーをはく

terasawa, 2015-09-24:
まだよく現象がわかっていないのですが、環境を最新にしてから
jaxon_red-interface.lとeuscolladaのcollada2eusを使ってできたドリルやスプレッダーなどの.lのファイル(spreader.lやtakenoko-drill.l)をともにロードし、(jaxon_red-init)すると
/home/leus/ros/indigo_parent/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: unbound variable glvertices in (if glvertices (progn (send glvertices :draw vwr)))
というエラーをはきます。

ちなみに、単体でspreader.lやtakenoko-drill.lなどをロードしてモデルを表示することは可能です。
今まではなんの問題もなく使えていたので最近の変更に何か問題がある気がしますがどうでしょうか。

YoheiKakiuchi, 2015-09-24:
https://github.com/jsk-ros-pkg/jsk_model_tools/pull/131
最近このPRをマージしたので、それ以前に作ったモデルと整合性がなくなっているとおもいます。
モデルを作りなおしてみてください。

YuOhara, 2015-09-24:
今作り直しましたので、大丈夫じゃないかと思います。

terasawa, 2015-09-24:
大丈夫そうでした。ありがとうございます。

closed 2015-09-24 by terasawa