start-jsk/rtmros_choreonoid #94: jaxonがヘタってしまう．

otsubo, 2015-10-01:
choreonoid $(rospack find hrpsys_ros_bridge_jvrc)/config/JVRCR4.cnoid
を実行して,シミュレーションをスタートすると,ジャクソンがヘタってしまいます.
以下ログです.
otsubo@axis:~$ choreonoid $(rospack find hrpsys_ros_bridge_jvrc)/config/JVRCR4.cnoid
Loading body customizer "/usr/lib/choreonoid-1.5/customizer/HoseCustomizer.so" for hose
Loading body customizer "/usr/lib/choreonoid-1.5/customizer/JAXONCustomizer.so" for JAXON_JVRC
Loading body customizer "/usr/lib/choreonoid-1.5/customizer/CabinetBoxCustomizer.so" for CABINETBOX
create customizer : JAXON_JVRC
PDcontroller0: onInitialize() 
create customizer : JAXON_JVRC
cloneMap.setNonNodeCloning(false);
cloneMap.setNonNodeCloning(false);
cloneMap.setNonNodeCloning(false);
cloneMap.setNonNodeCloning(false);
cloneMap.setNonNodeCloning(false);
cloneMap.setNonNodeCloning(false);
cloneMap.setNonNodeCloning(false);
PDcontroller0: on Activated 
[PDcontroller0] Gain file [] not opened

[image]

hrpsys_ros_bridge_jvrc/models以下にはJAXON_JVRC.PDgains.datがちゃんとあります.
どう対処すればよいでしょうか．

YoheiKakiuchi, 2015-10-01:
> git pull した後、
roscd jvrc_models; bash setup_jvrc.sh
してください

これやってるかな？

> /home/leus で
choreonoid $(rospack find hrpsys_ros_bridge_jvrc)/config/JVRCO1.cnoid

それと、ホームディレクトリでchoreonoidを走らせる必要があります。

あと、homeに rtc.conf.choreonoid はありますか？

otsubo, 2015-10-01:
> あと、homeに rtc.conf.choreonoid はありますか？
これがありませんでした.
このファイルはどこで生成されるのでしょうか.

YoheiKakiuchi, 2015-10-01:
https://github.com/start-jsk/rtmros_choreonoid/pull/95
修正しました。
本番に向けて、いくつかでユーザー名固定になってしまっています。

setup_jvrc.sh をもう一度してみてください。

あと、
```$(rospack find hrpsys_ros_bridge_jvrc)/config/JVRCO1.cnoid```
cnoidファイルにもleusが入ってしまっているので、ここを自分のユーザー名にしてください。

otsubo, 2015-10-01:
なぜか
```bash
bash setup_jvrc.sh
```
すると,許可がないと言われたので,rtmros_choreonoidごと入れなおして,buildしたら正常に動作しました.

closed 2015-10-01 by otsubo