start-jsk/rtmros_choreonoid #104: go-posが帰ってこない

YuOhara, 2015-10-05:
https://github.com/start-jsk/rtmros_common/issues/839
と関係する話という噂をききましたが、go-posが帰ってこなくなりました。。。

garaemon, 2015-10-05:
帰ってこない間、hrpsysは動いていますか? joint_statesは出ていますか?

YuOhara, 2015-10-05:
/joint_statesは来ています。

garaemon, 2015-10-05:
* 次は`:wait-foot-steps`のサービスコールが呼ばれているかを確認、ですね。(たぶん呼ばれている)
* hrpsysの対応するサービスのところにプリント分をいれてみて、どこで止まっているか確認.

snozawa, 2015-10-05:
```
roscd hrpsys_tools/scripts
python hrpsys_config_tools.py -i --use-unstable-rtc
hcf.waitFootSteps()
# or 
hcf.emergencyStop()
```
でpythonで確認できます。

YutaKojio, 2015-10-07:
コジオです。
先ほどgo-posが返ってこない問題が再発しました。

少し前に黒岩さんにdebug文を入れてもらった状態でchoreonoidをあげていましたが、logを見る感じでは、debugをはさんだwhileループから抜けていないように思われます。

logとdebugをはさんだ箇所のスクショが以下になります。

[image]

eisoku9618, 2015-10-07:
go-posからfoot stepを作る部分の収束条件が満たされずに，foot stepを作り続けているのが原因のようです．

positionの方の収束条件：0.1mm
rotationの方の収束条件：0.5度

なところで，以下のようにpositionの方が0.24mmくらいで収束していませんでした．
0.1mmのしきい値を10mmにして試してみています．

In [14]: dp
Out[14]: array([  2.16751000e-18,   1.65349000e-16,   2.47467000e-04])

In [15]: dr
Out[15]: array([ -6.53854000e-05,   2.15807000e-05,  -3.51280000e-16])

In [19]: numpy.sqrt(numpy.dot(dp, dp.transpose())) * 1e3
Out[19]: 0.24746700000000002

In [17]: numpy.sqrt(numpy.dot(dr, dr.transpose()))
Out[17]: 6.8854753980026676e-05

terasawa, 2015-10-07:
ちなみに（歩幅を変えている，go-posの指令を遠くにしすぎていなど）収束していない理由で考えられることはありますか？

今までの現象を見る限り，最初にgo-posが動けばそれ以降はうまく動いていた場合が多かったのですが，
収束しない条件が何かあるとしたら，１回動けばそれ以降は送れるというのはたまたまで
何回かgo-posできていても起こりうる気がしていますが．．．

h-kamada, 2015-10-07:
* 0.1の部分を10にしてやってみたところ、２回ともgo-posできました。
今から1としてやってみます。

YuOhara, 2015-10-16:
hrpsys-baseで報告していただいたようなのでcloseします．

* 現状はgo-posが返ってこないところに関しては解決している．

closed 2015-10-16 by YuOhara