start-jsk/rtmros_choreonoid #134: apt-get upgradeしたらjaxonがヘタるようになった。

YuOhara, 2015-11-19:
ヘタるようになってしまいました。
apt-get upgradeしたあとだと思うのですが、
何かわかりますでしょうか？
```
^[[A^[[Aleus@leus-HP-Z620-Workstation:~$ roslaunch hrpsys_ros_bridge_jvrc jaxon_reonoid.launch TASK:=VALVE
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
... logging to /home/leus/.ros/log/cd3842ac-8ec2-11e5-8716-a0481cabf72f/roslaunch-leus-HP-Z620-Workstation-36461.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/leus/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://133.11.216.41:34557/

SUMMARY
========

PARAMETERS
 * /HrpsysSeqStateROSBridge/publish_sensor_transforms: True
 * /chest_camera/CHESTCAMERA/camera_param_K: [85.7444, 0, 319....
 * /chest_camera/CHESTCAMERA/camera_param_P: [85.7444, 0, 319....
 * /chest_camera/CHESTCAMERA/frame_id: CHEST_CAMERA
 * /controller_configuration: [{'group_name': '...
 * /diagnostic_aggregator/analyzers/computers/contains: ['HD Temp', 'CPU ...
 * /diagnostic_aggregator/analyzers/computers/path: Computers
 * /diagnostic_aggregator/analyzers/computers/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/hrpsys/contains: ['hrpEC', 'Emerge...
 * /diagnostic_aggregator/analyzers/hrpsys/path: Hrpsys
 * /diagnostic_aggregator/analyzers/hrpsys/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/mode/contains: ['Operating Mode'...
 * /diagnostic_aggregator/analyzers/mode/path: Mode
 * /diagnostic_aggregator/analyzers/mode/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/motor/contains: ['Motor']
 * /diagnostic_aggregator/analyzers/motor/path: Motor
 * /diagnostic_aggregator/analyzers/motor/type: diagnostic_aggreg...
 * /diagnostic_aggregator/base_path: 
 * /diagnostic_aggregator/pub_rate: 1.0
 * /larm_camera/LARMCAMERA/camera_param_K: [417.031, 0, 319....
 * /larm_camera/LARMCAMERA/camera_param_P: [417.031, 0, 319....
 * /larm_camera/LARMCAMERA/frame_id: LARM_CAMERA
 * /larm_camera_n/LARMCAMERAN/camera_param_K: [417.031, 0, 319....
 * /larm_camera_n/LARMCAMERAN/camera_param_P: [417.031, 0, 319....
 * /larm_camera_n/LARMCAMERAN/frame_id: LARM_CAMERA_N
 * /multisense/left/HEADLEFT/camera_param_K: [240, 0, 319.5, 0...
 * /multisense/left/HEADLEFT/camera_param_P: [240, 0, 319.5, 0...
 * /multisense/left/HEADLEFT/frame_id: multisense/left_c...
 * /multisense/pointcloud_bridge/frame_id: multisense/left_c...
 * /multisense/pointcloud_bridge/publish_depth: True
 * /multisense/pointcloud_bridge/transformed_camera_frame: True
 * /multisense/range_bridge/frame_id: head_hokuyo_frame
 * /multisense/range_bridge/intensity: 1000
 * /rarm_camera/RARMCAMERA/camera_param_K: [879.192, 0, 319....
 * /rarm_camera/RARMCAMERA/camera_param_P: [879.192, 0, 319....
 * /rarm_camera/RARMCAMERA/frame_id: RARM_CAMERA
 * /rarm_camera_n/RARMCAMERAN/camera_param_K: [879.192, 0, 319....
 * /rarm_camera_n/RARMCAMERAN/camera_param_P: [879.192, 0, 319....
 * /rarm_camera_n/RARMCAMERAN/frame_id: RARM_CAMERA_N
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.16
 * /use_sim_time: True

NODES
  /chest_camera/
    CHESTCAMERA (hrpsys_ros_bridge/ImageSensorROSBridge)
  /multisense/
    pointcloud_bridge (hrpsys_ros_bridge/PointCloudROSBridge)
    range_bridge (hrpsys_ros_bridge/RangeSensorROSBridge)
  /
    AutoBalancerServiceROSBridge (hrpsys_ros_bridge/AutoBalancerServiceROSBridgeComp)
[... 659 lines cut ...]
KeyboardInterrupt
    time.sleep(1);
KeyboardInterrupt
[hrpsys_ros_diagnostics-8] killing on exit
[hrpsys_state_publisher-7] killing on exit
[HrpsysJointTrajectoryBridge-6] killing on exit
[HrpsysSeqStateROSBridge-5] killing on exit
[ INFO] [1447940318.898408495]: [HrpsysSeqStateROSBridge] @onFinalize : HrpsysSeqStateROSBridge0
[hrpsys_py-4] killing on exit
Traceback (most recent call last):
  File "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/scripts/jaxon_jvrc_hrpsys_config.py", line 50, in <module>
    hcf.init(sys.argv[1], sys.argv[2])
  File "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/src/hrpsys_ros_bridge_tutorials/urata_hrpsys_config.py", line 20, in init
    HrpsysConfigurator.init(self, robotname, url)
  File "/home/leus/ros/indigo_parent/devel/lib/python2.7/dist-packages/hrpsys/hrpsys_config.py", line 2032, in init
    self.waitForRTCManagerAndRoboHardware(robotname)
  File "/home/leus/ros/indigo_parent/devel/lib/python2.7/dist-packages/hrpsys/hrpsys_config.py", line 899, in waitForRTCManagerAndRoboHardware
    self.waitForRTCManager(managerhost)
  File "/home/leus/ros/indigo_parent/devel/lib/python2.7/dist-packages/hrpsys/hrpsys_config.py", line 839, in waitForRTCManager
    time.sleep(1)
KeyboardInterrupt
[choreonoid-3] killing on exit
[modelloader-2] killing on exit

```

YuOhara, 2015-11-19:
他のPCでも再現しました...

参考にならないかもしれませんが、一応
upgradeのlistは
```
^[[Aleus@leus-desktop:~/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_bridge_jvrc$ sudo apt-get upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  libcuda1-346 linux-generic-lts-vivid linux-headers-generic-lts-vivid
  linux-image-generic-lts-vivid nvidia-346 nvidia-opencl-icd-346
  ros-indigo-robot-state-publisher
The following packages will be upgraded:
  apport apport-gtk choreonoid choreonoid-data firefox firefox-locale-en
  flashplugin-installer fonts-opensymbol gir1.2-vte-2.90 grub-common grub-pc
  grub-pc-bin grub2-common indicator-session krb5-locales krb5-multidev
  libcnoid-dev libcnoid1 libgssapi-krb5-2 libgssrpc4 libido3-0.1-0
  libk5crypto3 libkadm5clnt-mit9 libkadm5srv-mit9 libkdb5-7 libkrb5-3
  libkrb5-dev libkrb5support0 libminiupnpc8 libmtp-common libmtp-runtime
  libmtp9 libmysqlclient-dev libmysqlclient18 libnautilus-extension1a
  libnm-gtk-common libnm-gtk0 libnspr4 libnss3 libnss3-nssdb
  liboxideqt-qmlplugin liboxideqtcore0 liboxideqtquick0 libpoppler-glib8
  libpoppler44 libpython3.4 libpython3.4-minimal libpython3.4-stdlib
  libreoffice-avmedia-backend-gstreamer libreoffice-base-core libreoffice-calc
  libreoffice-common libreoffice-core libreoffice-draw libreoffice-gnome
  libreoffice-gtk libreoffice-impress libreoffice-math libreoffice-ogltrans
  libreoffice-pdfimport libreoffice-presentation-minimizer
  libreoffice-style-human libreoffice-writer libunity-core-6.0-9 libvte-2.90-9
  libvte-2.90-common libxml2 libxml2-dev libxml2-utils linux-firmware
  linux-libc-dev mysql-common nautilus nautilus-data network-manager-gnome
  notify-osd-icons ntpdate openjdk-7-jdk openjdk-7-jre openjdk-7-jre-headless
  openrtm-aist oxideqt-codecs poppler-utils python-libxml2 python-rospkg
  python-urllib3 python-urllib3-whl python3-apport python3-problem-report
  python3-software-properties python3-uno python3-urllib3 python3.4
  python3.4-minimal ros-indigo-actionlib ros-indigo-actionlib-msgs
  ros-indigo-actionlib-tutorials ros-indigo-allocators ros-indigo-amcl
  ros-indigo-angles ros-indigo-ar-track-alvar ros-indigo-ar-track-alvar-msgs
  ros-indigo-audio-common-msgs ros-indigo-axis-camera
  ros-indigo-base-local-planner ros-indigo-bfl ros-indigo-bond
  ros-indigo-bondcpp ros-indigo-bondpy ros-indigo-calibration-estimation
  ros-indigo-calibration-launch ros-indigo-calibration-msgs
  ros-indigo-camera-calibration-parsers ros-indigo-camera-info-manager
  ros-indigo-camera-info-manager-py ros-indigo-capabilities ros-indigo-catkin
  ros-indigo-class-loader ros-indigo-clear-costmap-recovery
  ros-indigo-cmake-modules ros-indigo-collada-parser ros-indigo-collada-urdf
  ros-indigo-compressed-image-transport ros-indigo-control-msgs
  ros-indigo-control-toolbox ros-indigo-controller-interface
  ros-indigo-controller-manager ros-indigo-controller-manager-msgs
  ros-indigo-controller-manager-tests ros-indigo-convex-decomposition
  ros-indigo-costmap-2d ros-indigo-cpp-common ros-indigo-create-description
  ros-indigo-create-driver ros-indigo-create-node ros-indigo-cv-bridge
  ros-indigo-depth-image-proc ros-indigo-depthimage-to-laserscan
  ros-indigo-diagnostic-aggregator ros-indigo-diagnostic-common-diagnostics
  ros-indigo-diagnostic-msgs ros-indigo-diagnostic-updater
  ros-indigo-diff-drive-controller ros-indigo-driver-base
  ros-indigo-dwa-local-planner ros-indigo-dynamic-reconfigure
  ros-indigo-dynamixel-controllers ros-indigo-dynamixel-driver
  ros-indigo-dynamixel-msgs ros-indigo-ecl-build ros-indigo-ecl-command-line
  ros-indigo-ecl-concepts ros-indigo-ecl-config ros-indigo-ecl-containers
  ros-indigo-ecl-converters ros-indigo-ecl-devices ros-indigo-ecl-eigen
  ros-indigo-ecl-errors ros-indigo-ecl-exceptions ros-indigo-ecl-formatters
  ros-indigo-ecl-geometry ros-indigo-ecl-license ros-indigo-ecl-linear-algebra
  ros-indigo-ecl-math ros-indigo-ecl-mobile-robot ros-indigo-ecl-mpl
  ros-indigo-ecl-sigslots ros-indigo-ecl-streams ros-indigo-ecl-threads
  ros-indigo-ecl-time ros-indigo-ecl-time-lite ros-indigo-ecl-type-traits
  ros-indigo-ecl-utilities ros-indigo-effort-controllers
  ros-indigo-eigen-conversions ros-indigo-eigen-stl-containers ros-indigo-eml
  ros-indigo-ethercat-hardware ros-indigo-ethercat-trigger-controllers
[... 111 lines cut ...]
  ros-indigo-std-capabilities ros-indigo-std-msgs ros-indigo-std-srvs
  ros-indigo-stereo-image-proc ros-indigo-stereo-msgs ros-indigo-tf
  ros-indigo-tf-conversions ros-indigo-tf2 ros-indigo-tf2-geometry-msgs
  ros-indigo-tf2-msgs ros-indigo-tf2-py ros-indigo-tf2-ros
  ros-indigo-tf2-sensor-msgs ros-indigo-tf2-web-republisher
  ros-indigo-timestamp-tools ros-indigo-topic-tools ros-indigo-trajectory-msgs
  ros-indigo-transmission-interface ros-indigo-turtlebot-bringup
  ros-indigo-turtlebot-capabilities ros-indigo-turtlebot-description
  ros-indigo-turtlebot-teleop ros-indigo-ueye-cam ros-indigo-unique-id
  ros-indigo-urdf ros-indigo-urdf-parser-plugin ros-indigo-urdfdom-py
  ros-indigo-urg-c ros-indigo-urg-node ros-indigo-uuid-msgs
  ros-indigo-uvc-camera ros-indigo-velocity-controllers
  ros-indigo-visualization-msgs ros-indigo-voxel-grid ros-indigo-warehouse-ros
  ros-indigo-wge100-camera ros-indigo-wifi-ddwrt ros-indigo-xacro
  ros-indigo-xmlrpcpp ros-indigo-yocs-cmd-vel-mux ros-indigo-yocs-controllers
  ros-indigo-yocs-velocity-smoother ros-indigo-zbar-ros
  ros-indigo-zeroconf-avahi ros-indigo-zeroconf-msgs shotwell shotwell-common
  software-properties-common software-properties-gtk tzdata tzdata-java unity
  unity-services unity-settings-daemon uno-libs3 unzip ure wpasupplicant
505 upgraded, 0 newly installed, 0 to remove and 7 not upgraded.
Need to get 85.8 MB/354 MB of archives.
After this operation, 21.3 MB of additional disk space will be used.
Do you want to continue? [Y/n] y

```

YoheiKakiuchi, 2015-11-19:
僕の手元は問題ないです。
aptitude show choreonoid で見えるバージョンは以下です。
```
バージョン: 1.5.0+dfsg+20151117+474+16~ubuntu14.04.1
```

ログからはPDcontrollerに関するメッセージが見えないので、コントローラが作られていないのだと思いますが、
これは、choreonoid側で作るので、
choreonoidのメッセージを見ると分かることがあるかもしれません。

YuOhara, 2015-11-20:
choreonoid側の出力:
```
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidCorbaPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidPythonPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidOpenRTMPlugin.so"
Loading the plugin failed.
Cannot load library /usr/lib/choreonoid-1.5/libCnoidOpenRTMPlugin.so: (libRTC-1.1.1.so: cannot open shared object file: No such file or directory)
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidJVRCPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidPythonSimScriptPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidGRobotPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidBodyPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidSimpleControllerPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidPoseSeqPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidBalancerPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidOpenRTMPlugin.so"
Loading the plugin failed.
Cannot load library /usr/lib/choreonoid-1.5/libCnoidOpenRTMPlugin.so: (libRTC-1.1.1.so: cannot open shared object file: No such file or directory)
Body-plugin has been activated.
Corba-plugin has been activated.
GRobot-plugin has been activated.
JVRC-plugin has been activated.
PoseSeq-plugin has been activated.
Python-plugin has been activated.
PythonSimScript-plugin has been activated.
SimpleController-plugin has been activated.
Balancer-plugin has been activated.

Loading project file "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/JVRC_VALVE.cnoid" ...
Task sequencer '' has been deactivated.
Restoring WorldItem "World"
Restoring BodyItem "JAXON_JVRC"
Loading OpenHRP Model File "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/jvrc_models/JAXON_JVRC/JAXON_JVRCmain_hrpsys_bush.wrl"
 -> ok!
Warning: OpenRTMPlugin is not loaded.
Warning: BodyRTCItem of OpenRTMPlugin is not a registered item type.
Warning: "BodyRTC" cannot be restored.
Restoring BodyItem "wall"
Loading OpenHRP Model File "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/jvrc_models/model/tasks/O1/wall.wrl"
 -> ok!
Restoring BodyItem "surface"
Loading OpenHRP Model File "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/jvrc_models/model/tasks/O1/surface.wrl"
 -> ok!
Restoring BodyItem "valve"
Loading OpenHRP Model File "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/model/tasks/VALVE/valve.wrl"
 -> ok!
Restoring BodyItem "drcbox"
Loading OpenHRP Model File "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/model/tasks/VALVE/drcbox.wrl"
 -> ok!
Restoring BodyItem "valve2"
Loading OpenHRP Model File "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/model/tasks/VALVE/valve.wrl"
 -> ok!
Restoring BodyItem "drcbox2"
Loading OpenHRP Model File "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/model/tasks/VALVE/drcbox.wrl"
 -> ok!
Restoring AISTSimulatorItem "AISTSimulator"
Restoring JVRCManagerItem "JVRCManager"
Loading JVRC Info "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/jvrc_models/model/tasks/O1/O1.yaml"
 -> ok!
JVRC Robot "JAXON_JVRC" has been detected.
Restoring GLVisionSimulatorItem "GLVisionSimulator"
Restoring GLVisionSimulatorItem "GLVisionSimulator"
Restoring GLVisionSimulatorItem "GLVisionSimulator"
Restoring GLVisionSimulatorItem "GLVisionSimulator"
Restoring GLVisionSimulatorItem "GLVisionSimulator"
Restoring GLVisionSimulatorItem "GLVisionSimulator"
Restoring GLVisionSimulatorItem "GLVisionSimulator"
17 / 18 item(s) are loaded.
Warning: 1 item(s) are not correctly loaded.
Project "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/JVRC_VALVE.cnoid" has been loaded.
SubSimulatorItem "JVRCManager" has been detected.
Warning: "jvrc-team-name.txt" is not found. Outputting record files is not available.
Remaining time is output to "jvrc-remaining-time.txt".
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "HEAD_LEFT_CAMERA" of JAXON_JVRC as a target.
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "CHEST_CAMERA" of JAXON_JVRC as a target.
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "HEAD_RANGE" of JAXON_JVRC as a target.
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "LARM_CAMERA" of JAXON_JVRC as a target.
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "RARM_CAMERA" of JAXON_JVRC as a target.
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "LARM_CAMERA_N" of JAXON_JVRC as a target.
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "RARM_CAMERA_N" of JAXON_JVRC as a target.
Simulation by AISTSimulator has started.
```

YuOhara, 2015-11-20:
```
leus@leus-HP-Z620-Workstation:/usr/lib/choreonoid-1.5$ locate libRTC-
/home/leus/ros/indigo_parent/devel/lib/libRTC-1.1.0.so
/home/leus/ros/indigo_parent/src/openrtm_aist/src/lib/rtm/.libs/libRTC-1.1.0.so
/home/leus/ros/indigo_parent/src/openrtm_aist/src/lib/rtm/.libs/libRTC-1.1.0.soT
/usr/lib/x86_64-linux-gnu/libRTC-1.1.2.so
```

k-okada, 2015-11-20:
小原君につかっているchoreonoid

Versionは垣内君と同じ?

YuOhara, 2015-11-20:
同じでした。
```
Version: 1.5.0+dfsg+20151117+474+16~ubuntu14.04.1
```

YoheiKakiuchi, 2015-11-20:
まちがえました。僕は自分のビルドを使ってました。
aptだとopenrtmのバージョンがずれているんじゃないかと思います。

openrtm-aist
```
バージョン: 1.1.1+20151119+1709+6~ubuntu14.04.1
```

YoheiKakiuchi, 2015-11-20:
なので、aptで入れると現状動かないが正しいように思います。

YuOhara, 2015-11-20:
最悪sudo cp でlibをcpなりlnなりすれば動くことがわかりました。

k-okada, 2015-11-20:
openrtmのバージョンの動いているものと動いていないものの比較は？
また，openrtmのちがいによってロボットがへたるという時に考えられる原因は？

k-okada, 2015-11-20:
そういうことはやめましょう．ちゃんと使っていくなら，ちゃんとしておくべきで，それが面倒なら使わないべきです．

YuOhara, 2015-11-20:
> openrtmのちがいによってロボットがへたるという時
これは
```
Cannot load library /usr/lib/choreonoid-1.5/libCnoidOpenRTMPlugin.so: (libRTC-1.1.1.so: cannot open shared object file: No such file or director
```
というエラーの通り、違うというよりは、ファイルがそもそもないからです。

もちろんchoreonoid(でいいのかあまり自信ないですが)のgithubページにあとで報告しますが、
テンポラリで使いたくて、
apt-cache show-pkgで
```
Reverse Provides: 
choreonoid:i386 1.5.0+dfsg+20151117+474+16~ubuntu14.04.
```
としか出ないので、apt-get upgradeを戻す方法がわからず、
すぐに使うためにはcpしかないと思いました。

k-okada, 2015-11-20:
なるほど．前はlibrtc-1.1.1.so はあったのになくなった？それとも名前が変わった？
松阪さんはopenrtmのdebはnightliy buildだから，と行っていたので，両方そこから入っているとチョット変ですね．
openrtmはros側のものをつかっていると，おこるかもしれない．
いずれにせよ，松阪さんに報告して見て下さい，nightly buildだったら明日には治るので．

YoheiKakiuchi, 2015-11-20:
どうやら、openrtmのソースでtrunkのバージョンが1.1.1 から 1.1.2へ最近変更されたようです。
choreonoidのdebがリリースされると治りそうです。

k-okada, 2015-11-20:
choreonoidのdebもnightly updateでなかったっけ？であればもう一回apt-get upgradeすれば治りそうです．

YuOhara, 2015-11-20:
```
https://launchpad.net/~hrg/+archive/ubuntu/daily/+packages
``` 
17日が最後の更新のように見えます（今日20日）。

k-okada, 2015-11-20:
choreonidのgithub issueでupdateしてと聞いてみましょう．

YuOhara, 2015-11-20:
最新のdebをリリースしてもらうためには、どこにcontactすべきなのでしょうか？
https://github.com/jsk-ros-pkg/jsk_recognition/issues/1331
これと同じでしょうか

YuOhara, 2015-11-20:
あ、ありがとうございます。

closed 2015-12-18 by YuOhara