start-jsk/rtmros_choreonoid #135: srcを最新にしたら、choreonoidが上がらなくなった？

YuOhara, 2015-12-18:
apt-get update & upgrade + hrpsys, hrpsys_ros_bridge_tutorials等を最新にする
と、choreonoidがjaxon_jvrc_choreonoidで上がらなくなってしまいました。
後者のhrpsys系のupdateのあとに起こったので、おそらくそこが原因だと思うのですが、
一方でsrcを古くしても治らないので（compileが足りないかもしれませんが、)確証はないです。
下記がlogです。choreonoid側はすぐ落ちるので、出力が得られていないです。

```
eonoid.launch TASK:=O1
leus@leus-HP-Z620-Workstation:~$ roslaunch hrpsys_ros_bridge_jvrc jaxon_jvrc_choreonoid.launch TASK:=O1
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
... logging to /home/leus/.ros/log/86e9d0b8-a489-11e5-bf2b-a0481cabf72e/roslaunch-leus-HP-Z620-Workstation-25453.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/leus/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://leus-HP-Z620-Workstation:59767/

SUMMARY
========

PARAMETERS
 * /HrpsysSeqStateROSBridge/publish_sensor_transforms: True
 * /chest_camera/CHESTCAMERA/camera_param_K: [85.7444, 0, 319....
 * /chest_camera/CHESTCAMERA/camera_param_P: [85.7444, 0, 319....
 * /chest_camera/CHESTCAMERA/frame_id: CHEST_CAMERA
 * /controller_configuration: [{'group_name': '...
 * /diagnostic_aggregator/analyzers/computers/contains: ['HD Temp', 'CPU ...
 * /diagnostic_aggregator/analyzers/computers/path: Computers
 * /diagnostic_aggregator/analyzers/computers/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/hrpsys/contains: ['hrpEC', 'Emerge...
 * /diagnostic_aggregator/analyzers/hrpsys/path: Hrpsys
 * /diagnostic_aggregator/analyzers/hrpsys/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/mode/contains: ['Operating Mode'...
 * /diagnostic_aggregator/analyzers/mode/path: Mode
 * /diagnostic_aggregator/analyzers/mode/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/motor/contains: ['Motor']
 * /diagnostic_aggregator/analyzers/motor/path: Motor
 * /diagnostic_aggregator/analyzers/motor/type: diagnostic_aggreg...
 * /diagnostic_aggregator/base_path: 
 * /diagnostic_aggregator/pub_rate: 1.0
 * /larm_camera/LARMCAMERA/camera_param_K: [417.031, 0, 319....
 * /larm_camera/LARMCAMERA/camera_param_P: [417.031, 0, 319....
 * /larm_camera/LARMCAMERA/frame_id: LARM_CAMERA
 * /larm_camera_n/LARMCAMERAN/camera_param_K: [417.031, 0, 319....
 * /larm_camera_n/LARMCAMERAN/camera_param_P: [417.031, 0, 319....
 * /larm_camera_n/LARMCAMERAN/frame_id: LARM_CAMERA_N
 * /multisense/left/HEADLEFT/camera_param_K: [240, 0, 319.5, 0...
 * /multisense/left/HEADLEFT/camera_param_P: [240, 0, 319.5, 0...
 * /multisense/left/HEADLEFT/frame_id: multisense/left_c...
 * /multisense/pointcloud_bridge/frame_id: multisense/left_c...
 * /multisense/pointcloud_bridge/publish_depth: True
 * /multisense/pointcloud_bridge/transformed_camera_frame: True
 * /multisense/range_bridge/frame_id: head_hokuyo_frame
 * /multisense/range_bridge/intensity: 1000
 * /rarm_camera/RARMCAMERA/camera_param_K: [879.192, 0, 319....
 * /rarm_camera/RARMCAMERA/camera_param_P: [879.192, 0, 319....
 * /rarm_camera/RARMCAMERA/frame_id: RARM_CAMERA
 * /rarm_camera_n/RARMCAMERAN/camera_param_K: [879.192, 0, 319....
 * /rarm_camera_n/RARMCAMERAN/camera_param_P: [879.192, 0, 319....
 * /rarm_camera_n/RARMCAMERAN/frame_id: RARM_CAMERA_N
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.16
 * /use_sim_time: True

NODES
  /chest_camera/
    CHESTCAMERA (hrpsys_ros_bridge/ImageSensorROSBridge)
  /multisense/
[... 537 lines cut ...]
duplicated sensor Id is specified(id = 0, name = gsensor)
duplicated sensor Id is specified(id = 0, name = gyrometer)
0 physical force sensor : rfsensor
1 physical force sensor : lfsensor
2 physical force sensor : rhsensor
3 physical force sensor : lhsensor
0 acceleration sensor : gsensor
0 rate sensor : gyrometer
[HrpsysSeqStateROSBridge0] End Effector [rleg]RLEG_JOINT5 WAIST
[HrpsysSeqStateROSBridge0]   target = RLEG_LINK5, sensor_name = rfsensor
[HrpsysSeqStateROSBridge0]   localPos =     [0,  0,  -0.1][m]
[HrpsysSeqStateROSBridge0] End Effector [lleg]LLEG_JOINT5 WAIST
[HrpsysSeqStateROSBridge0]   target = LLEG_LINK5, sensor_name = lfsensor
[HrpsysSeqStateROSBridge0]   localPos =     [0,  0,  -0.1][m]
[HrpsysSeqStateROSBridge0] End Effector [rarm]RARM_JOINT7 CHEST_JOINT2
[HrpsysSeqStateROSBridge0]   target = RARM_LINK7, sensor_name = rhsensor
[HrpsysSeqStateROSBridge0]   localPos =     [0,  0.055,  -0.217][m]
[HrpsysSeqStateROSBridge0] End Effector [larm]LARM_JOINT7 CHEST_JOINT2
[HrpsysSeqStateROSBridge0]   target = LARM_LINK7, sensor_name = lhsensor
[HrpsysSeqStateROSBridge0]   localPos =     [0,  -0.055,  -0.217][m]
[ INFO] [1450421067.946110983]: [HrpsysSeqStateROSBridge] Loaded JAXON_JVRC
[ INFO] [1450421067.946366042]: [HrpsysSeqStateROSBridge] @Initilize name : HrpsysSeqStateROSBridge0 done
[rtmlaunch] Wait for  /localhost:2809/JAXON_JVRC.rtc:CHEST_CAMERA   2 /30
[hrpsys.py] wait for JAXON_JVRC : None 
```

YoheiKakiuchi, 2015-12-18:
以下で、choreonoid単体で立ち上げてみるとどうなるでしょうか？

```rosrun hrpsys_choreonoid run_choreonoid.sh /home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/JVRC_O1.cnoid```

YuOhara, 2015-12-18:
こちらは普通に上がりました。
```
leus@leus-HP-Z620-Workstation:~$ rosrun hrpsys_choreonoid run_choreonoid.sh
/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/JVRC_O1.cnoid
$ choreonoid
/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/JVRC_O1.cnoid
with rtc.conf file on /tmp/rtc.conf.choreonoid
<BEGIN: rtc.conf>
cat: /tmp/rtc.conf.choreonoid: No such file or directory
<END: rtc.conf>
Loading body customizer
"/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid/JAXONCustomizer.so"
for JAXON_JVRC
Loading body customizer
"/usr/lib/choreonoid-1.5/customizer/HoseCustomizer.so" for hose
Loading body customizer
"/usr/lib/choreonoid-1.5/customizer/CabinetBoxCustomizer.so" for CABINETBOX
create customizer : JAXON_JVRC
PDcontroller0: onInitialize()

```

YoheiKakiuchi, 2015-12-18:
手元でも、再現しています。

時間が取れていないですが、choreonoidへのRTMのコネクトができずに落ちているように見えます。

YoheiKakiuchi, 2015-12-21:
原因はまったく分かっていませんが、
```sudo apt-get remove choreonoid-jvrc-plugin```
で落ちないようになるかと思います。

choreonoidをgdb上で走らせるコマンドをコメントアウトして入れてみたので、
https://github.com/start-jsk/rtmros_choreonoid/pull/136

バグ報告を以下にしてもらえると助かります。
https://github.com/jvrc/JVRCPlugin

YuOhara, 2015-12-21:
ありがとうございます。確認しました。
報告もします。

closed 2016-01-27 by YuOhara