start-jsk/rtmros_choreonoid #141: ロボットがガシャンと崩れる

eisoku9618, 2016-01-14:
```bash
pkill -9 choreonoid; roslaunch hrpsys_ros_bridge_jvrc jaxon_jvrc_choreonoid.launch TASK:=FLAT
```

でシミュレーションが上がると思う（上がっていた）のですが，最近

```bash
Make sure the hostname is correct and the Nameserver is running.
```

と出てロボットがガシャンと崩れるようになりました @ 共用PC．

rebootして``ps aux | grep omni``してもomniNamesは上がっていないようでしたので，roslaunchをrtmlaunchに変えて，corbaportも15005に指定したら

```bash
pkill -9 choreonoid; rtmlaunch hrpsys_ros_bridge_jvrc jaxon_jvrc_choreonoid.launch TASK:=FLAT corpbaport:=15005
```

``Make sure the hostname is correct and the Nameserver is running.``のエラーは消えましたが，ロボットがシミュレータが上がるやいなやガシャンと崩れます．

何を確認したら良いか教えていただけますでしょうか？
環境はhydro + 12.04で，``choreonoid-jvrc-plugin``は消しています．

ログは以下になります．

https://gist.githubusercontent.com/eisoku9618/f7290d77283c2f72464d/raw/8b05a6f9f33c94873adc6138d0c338a951c9f73c/choreonoid.log

よろしくお願いします．

YoheiKakiuchi, 2016-01-14:
現状では、choreonoid側のネームサービスのポートを変更できていないので、
2809を使うようになっています。
なので、2809でネームサービスを上げみてください。

eisoku9618, 2016-02-08:
ありがとうございます．
（遅くなってしまって申し訳ありません）

2809でやっても以下のようにダメなようです．
また，12.04/hydro で起こっていて，14.04/indigo では起こっていないです．

rebootしてから，``ps aux | grep omni``で``omniNames``が上がっていないことを確認してから，``rtm-naming 2809``でomniNamesを上げて，

```bash
roslaunch hrpsys_ros_bridge_jvrc jaxon_jvrc_choreonoid.launch TASK:=FLAT
```

をしても，やはりガシャンと崩れます．

====

Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidPythonSimScriptPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidGRobotPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidPythonPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidOpenRTMPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidPoseSeqPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidBalancerPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidBodyPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidSimpleControllerPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidCorbaPlugin.so"
Body-plugin has been activated.
GRobot-plugin has been activated.
ChoreonoidExecutionContext has been registered.
ChoreonoidPeriodicExecutionContext has been registered.
OpenRTM-plugin has been activated.
PoseSeq-plugin has been activated.
Python-plugin has been activated.
PythonSimScript-plugin has been activated.
SimpleController-plugin has been activated.
Balancer-plugin has been activated.
Corba-plugin has been activated.

Loading project file "/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/JVRC_FLAT.cnoid" ...
Task sequencer '' has been deactivated.
Restoring WorldItem "World"
Restoring BodyItem "JAXON_JVRC"
Loading OpenHRP Model File "/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/jvrc_models/JAXON_JVRC/JAXON_JVRCmain_hrpsys_bush.wrl"
-> ok!
Warning: Mismatched size of the stored initial joint positions for JAXON_JVRC
Restoring BodyRTCItem "BodyRTC"
Config File "/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/SensorReaderRTC.PD.conf" has been loaded.
A file of RTC "PDcontroller" does not exist.
RTC "PDcontroller" cannot be created.
RTC "JAXON_JVRC" has been created.
Restoring BodyItem "floor"
Loading OpenHRP Model File "/usr/share/choreonoid-1.5/model/misc/floor.wrl"
-> ok!
Warning: Mismatched size of the stored joint positions for floor
Warning: Mismatched size of the stored initial joint positions for floor
Restoring AISTSimulatorItem "AISTSimulator"
Warning: JVRCPlugin is not loaded.
Warning: JVRCManagerItem of JVRCPlugin is not a registered item type.
Warning: "JVRCManager" cannot be restored.
Restoring GLVisionSimulatorItem "GLVisionSimulator"
Restoring GLVisionSimulatorItem "GLVisionSimulator"
Restoring GLVisionSimulatorItem "GLVisionSimulator"
Restoring GLVisionSimulatorItem "GLVisionSimulator"
Restoring GLVisionSimulatorItem "GLVisionSimulator"
Restoring GLVisionSimulatorItem "GLVisionSimulator"
Restoring GLVisionSimulatorItem "GLVisionSimulator"
12 / 13 item(s) are loaded.
Warning: 1 item(s) are not correctly loaded.
Project "/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/JVRC_FLAT.cnoid" has been loaded.
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "HEAD_LEFT_CAMERA" of JAXON_JVRC as a target.
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "CHEST_CAMERA" of JAXON_JVRC as a target.
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "HEAD_RANGE" of JAXON_JVRC as a target.
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "LARM_CAMERA" of JAXON_JVRC as a target.
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "RARM_CAMERA" of JAXON_JVRC as a target.
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "LARM_CAMERA_N" of JAXON_JVRC as a target.
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "RARM_CAMERA_N" of JAXON_JVRC as a target.
RTC "PDcontroller" is not ready.
BodyRTC for JAXON_JVRC failed to initialize.
Simulation by AISTSimulator has started.

ishiguroJSK, 2016-04-19:
これ僕も新しく構築した環境(indigo 14.04)の方でなるんですが，最近まで別の環境で動いていたので不思議です．
PDControllerが上がらないのは僕が忘れているパッケージがあるせいなんでしょうか？
以下ワークスペース内パッケージ(レポジトリ)
hrpsys
openhrp3
jsk_control
rtmros_common
rtmros_tutorials
rtmros_choreonoid
rtmros_gazebo
rtmros_hrp2

choreonoid側のエラーは上記と同じ，
端末側のエラーは以下の様で，見慣れないWARNING(unrecognized tag以降)が出ている気がします

```
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtconnect
WARNING: unrecognized tag rtactivate
WARNING: unrecognized tag rtactivate
WARNING: WARN: unrecognized 'rtconnect' tag in <node> tag. Node xml is <node args="$(arg openrtm_args)" name="CHESTCAMERA" ns="chest_camera" output="$(arg OUTPUT)" pkg="hrpsys_ros_bridge" type="ImageSensorROSBridge">
    <param name="frame_id" value="CHEST_CAMERA"/>
    <rosparam param="camera_param_K">[85.7444, 0, 319.5,  0, 85.7444, 319.5,  0, 0, 1]</rosparam>
    <rosparam param="camera_param_P">[85.7444, 0, 319.5, 0,  0, 85.7444, 319.5, 0,  0, 0, 1, 0]</rosparam>
    <rtconnect from="JAXON_JVRC.rtc:CHEST_CAMERA" to="CHESTCAMERA.rtc:timedImage"/>
    <rtactivate component="CHESTCAMERA.rtc"/>
  </node>
WARNING: WARN: unrecognized 'rtactivate' tag in <node> tag. Node xml is <node args="$(arg openrtm_args)" name="CHESTCAMERA" ns="chest_camera" output="$(arg OUTPUT)" pkg="hrpsys_ros_bridge" type="ImageSensorROSBridge">
    <param name="frame_id" value="CHEST_CAMERA"/>
    <rosparam param="camera_param_K">[85.7444, 0, 319.5,  0, 85.7444, 319.5,  0, 0, 1]</rosparam>
    <rosparam param="camera_param_P">[85.7444, 0, 319.5, 0,  0, 85.7444, 319.5, 0,  0, 0, 1, 0]</rosparam>
    <rtconnect from="JAXON_JVRC.rtc:CHEST_CAMERA" to="CHESTCAMERA.rtc:timedImage"/>
    <rtactivate component="CHESTCAMERA.rtc"/>
[... 78 lines cut ...]
WARNING: WARN: unrecognized 'rtactivate' tag in <node> tag. Node xml is <node args="$(arg openrtm_args)" name="range_bridge" ns="multisense" output="$(arg OUTPUT)" pkg="hrpsys_ros_bridge" type="RangeSensorROSBridge">
    <param name="frame_id" value="head_hokuyo_frame"/>
    <param name="intensity" value="1000"/>
    <remap from="range" to="lidar_scan"/>
    <rtconnect from="JAXON_JVRC.rtc:HEAD_RANGE" to="RangeSensorROSBridge0.rtc:range"/>
    <rtactivate component="RangeSensorROSBridge0.rtc"/>
  </node>
WARNING: WARN: unrecognized 'rtconnect' tag in <node> tag. Node xml is <node args="$(arg openrtm_args)" name="pointcloud_bridge" ns="multisense" output="$(arg OUTPUT)" pkg="hrpsys_ros_bridge" type="PointCloudROSBridge">
    <param name="frame_id" value="multisense/left_camera_optiocal_frame"/>
    <param name="publish_depth" value="true"/>
    <param name="transformed_camera_frame" value="true"/>
    <remap from="points" to="organized_image_points2_color"/>
    <rtconnect from="JAXON_JVRC.rtc:HEAD_LEFT_DEPTH" to="PointCloudROSBridge0.rtc:points"/>
    <rtactivate component="PointCloudROSBridge0.rtc"/>
  </node>
WARNING: WARN: unrecognized 'rtactivate' tag in <node> tag. Node xml is <node args="$(arg openrtm_args)" name="pointcloud_bridge" ns="multisense" output="$(arg OUTPUT)" pkg="hrpsys_ros_bridge" type="PointCloudROSBridge">
    <param name="frame_id" value="multisense/left_camera_optiocal_frame"/>
    <param name="publish_depth" value="true"/>
    <param name="transformed_camera_frame" value="true"/>
    <remap from="points" to="organized_image_points2_color"/>
    <rtconnect from="JAXON_JVRC.rtc:HEAD_LEFT_DEPTH" to="PointCloudROSBridge0.rtc:points"/>
    <rtactivate component="PointCloudROSBridge0.rtc"/>
  </node>
```
[image]

ishiguroJSK, 2016-04-19:
古い環境のワークスペースをごっそり持ってきてcatkin buildしたらちゃんと立てたので，恐らく自分が最小構成で構成したワークスペースに何かが足りないんだと思います．しばらく考えてみます
ちなみに見慣れないエラーは出てても立てました．

YoheiKakiuchi, 2016-04-20:
それで、必要なものはそろっているはずですが、
hrpsys_choreonoidをcatkin build --thisしてためしてもらえますか？

choreonoidでのエラー報告のばあいは、choreonoidの右下のメッセージ欄(Simulation by AISTSimulator has started.とでているところ)の内容もコピペしてもらえるとありがたいです。

それでもガシャンとなっているようなら、
以下のコマンドでなにか返って来ますか？
```
$ RTCTREE_NAMESERVERS=localhost:2809 rtcat -l localhost:2809/PDcontroller0.rtc
```

つながっている場合は以下のように返って来ます。
```
PDcontroller0.rtc  Active
  Category       example
  Description    PDcontroller component
  Instance name  PDcontroller0
  Type name      PDcontroller
  Vendor         JSK
  Version        0.0.0
  Parent         
  Type           Monolithic
 +Extra properties
-Execution Context 0
  State  Running
  Kind   Periodic
  Rate   1000000.0
  Owner  PDcontroller0
-DataInPort: angle
  dataport.data_type          IDL:RTC/TimedDoubleSeq:1.0
  dataport.dataflow_type      push,pull
  dataport.interface_type     corba_cdr
  dataport.subscription_type  Any
  port.port_type              DataInPort
 +Connected to  /localhost:2809/JAXON_RED.rtc:q
-DataInPort: angleRef
  dataport.data_type          IDL:RTC/TimedDoubleSeq:1.0
  dataport.dataflow_type      push,pull
  dataport.interface_type     corba_cdr
  dataport.subscription_type  Any
  port.port_type              DataInPort
 +Connected to  /localhost:2809/el.rtc:q
-DataOutPort: torque
  dataport.data_type          IDL:RTC/TimedDoubleSeq:1.0
  dataport.dataflow_type      push,pull
  dataport.interface_type     corba_cdr
  dataport.subscription_type  flush,new,periodic
  port.port_type              DataOutPort
 +Connected to  /localhost:2809/JAXON_RED.rtc:tauIn
```

PDcontrollerがないといったようなエラーになったら、以下のコマンドでなにが見えるか教えてください。
```
$ RTCTREE_NAMESERVERS=localhost:2809 rtls localhost:2809/
```

見慣れないエラーは、openrtm_toolsにあるrtmlaunch(roslaunchのしくみを使ってopenrtmのコンポーネントを作る、接続するなどを行ってくれる)のためのtagがroslaunchが理解できないためで問題ないです。
http://wiki.ros.org/openrtm_tools
https://github.com/start-jsk/rtmros_common/blob/master/openrtm_tools/scripts/rtmlaunch

ishiguroJSK, 2016-04-21:
> hrpsys_choreonoidをcatkin build --thisしてためしてもらえますか？

最小構成のワークスペースではやはりダメでした・・・
choreonidのログ欄は以下のようです
```
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidSimpleControllerPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidGRobotPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidPoseSeqPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidPythonPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidCorbaPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidPythonSimScriptPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidBodyPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidBalancerPlugin.so"
Detecting plugin file "/usr/lib/choreonoid-1.5/libCnoidOpenRTMPlugin.so"
Body-plugin has been activated.
GRobot-plugin has been activated.
ChoreonoidExecutionContext has been registered.
ChoreonoidPeriodicExecutionContext has been registered.
OpenRTM-plugin has been activated.
PoseSeq-plugin has been activated.
Python-plugin has been activated.
PythonSimScript-plugin has been activated.
SimpleController-plugin has been activated.
Balancer-plugin has been activated.
Corba-plugin has been activated.

Loading project file "/home/ishiguro/catkin_ws/jaxon/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/JVRC_FLAT.cnoid" ...
Task sequencer '' has been deactivated.
Restoring WorldItem "World"
Restoring BodyItem "JAXON_JVRC"
Loading OpenHRP Model File "/home/ishiguro/catkin_ws/jaxon/src/rtm-ros-robotics/rtmros_choreonoid/jvrc_models/JAXON_JVRC/JAXON_JVRCmain_hrpsys_bush.wrl"
 -> ok!
Warning: Mismatched size of the stored joint positions for JAXON_JVRC
Warning: Mismatched size of the stored initial joint positions for JAXON_JVRC
Restoring BodyRTCItem "BodyRTC"
Config File "/home/ishiguro/catkin_ws/jaxon/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/SensorReaderRTC.PD.conf" has been loaded.
A file of RTC "PDcontroller" does not exist.
RTC "PDcontroller" cannot be created.
RTC "JAXON_JVRC" has been created.
Restoring BodyItem "floor"
Loading OpenHRP Model File "/usr/share/choreonoid-1.5/model/misc/floor.wrl"
 -> ok!
Warning: Mismatched size of the stored joint positions for floor
Warning: Mismatched size of the stored initial joint positions for floor
Restoring AISTSimulatorItem "AISTSimulator"
Warning: JVRCPlugin is not loaded.
Warning: JVRCManagerItem of JVRCPlugin is not a registered item type.
Warning: "JVRCManager" cannot be restored.
Restoring GLVisionSimulatorItem "GLVisionSimulator"
Restoring GLVisionSimulatorItem "GLVisionSimulator"
Restoring GLVisionSimulatorItem "GLVisionSimulator"
Restoring GLVisionSimulatorItem "GLVisionSimulator"
Restoring GLVisionSimulatorItem "GLVisionSimulator"
Restoring GLVisionSimulatorItem "GLVisionSimulator"
Restoring GLVisionSimulatorItem "GLVisionSimulator"
12 / 13 item(s) are loaded.
Warning: 1 item(s) are not correctly loaded.
Project "/home/ishiguro/catkin_ws/jaxon/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/JVRC_FLAT.cnoid" has been loaded.
Executing python script "/home/ishiguro/catkin_ws/jaxon/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid/launch/constraint_enable.py" ...
The script finished.
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "HEAD_LEFT_CAMERA" of JAXON_JVRC as a target.
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "CHEST_CAMERA" of JAXON_JVRC as a target.
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "HEAD_RANGE" of JAXON_JVRC as a target.
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "LARM_CAMERA" of JAXON_JVRC as a target.
SubSimulatorItem "GLVisionSimulator" has been detected.
GLVisionSimulator detected vision sensor "RARM_CAMERA" of JAXON_JVRC as a target.
[... 1047 lines cut ...]
[hrpsys.py] create CompSvc -> Stabilizer Service : <OpenHRP._objref_StabilizerService instance at 0x7fa4cdd72098>
[hrpsys.py]   eval : [self.co, self.co_svc, self.co_version] = self.createComp("CollisionDetector","co")
;;
;; Could not open /dev/console for writing.
;;
open: Permission denied
[co] onInitialize()
cache found for file:///home/ishiguro/catkin_ws/jaxon/src/rtm-ros-robotics/rtmros_choreonoid/jvrc_models/JAXON_JVRC/JAXON_JVRCmain_hrpsys.wrl
^C[jointstates_relay-38] killing on exit
[head_root_frame_id-37] killing on exit
[head_range_frame_id-36] killing on exit
[head_left_frame_id-35] killing on exit
[multisense/pointcloud_bridge-34] killing on exit
[rarm_camera_n/RARMCAMERAN-31] killing on exit
[multisense/left/HEADLEFT-32] killing on exit
[larm_camera_n/LARMCAMERAN-29] killing on exit
[multisense/range_bridge-33] killing on exit
[rarm_camera/RARMCAMERA-30] killing on exit
[rtmlaunch] Wait for  /localhost:2809/sh.rtc:StateHolderService   1 /30
rtmlaunch.py: CORBA.BAD_PARAM(omniORB.BAD_PARAM_IndexOutOfRange, CORBA.COMPLETED_NO)
[larm_camera/LARMCAMERA-28] killing on exit
[chest_camera/CHESTCAMERA-27] killing on exit
[rtmlaunch_vision_connect-26] killing on exit
[rtmlaunch] Catch signal 'SIGINT', exitting...
```

YoheiKakiuchi, 2016-04-21:
https://github.com/start-jsk/rtmros_choreonoid/pull/147
ごめんなさい。上のPRで少し新しくなっていて、以下のコマンドではどうでしょう？

catkin build hrpsys_choreonoid_tutorials
roslaunch hrpsys_choreonoid_tutorials jaxon_red_choreonoid.launch

それでもガシャンとなるようなら、
roscd hrpsys_choreonoid
ls
でそのディレクトリになにがあるか見てもらえるでしょうか？

choreonoidのエラー的には、PDcontrollerが見つかっていません。
```
Config File "/home/ishiguro/catkin_ws/jaxon/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_ros_bridge_jvrc/config/SensorReaderRTC.PD.conf" has been loaded.
A file of RTC "PDcontroller" does not exist.
```

/home/ishiguro/catkin_ws/jaxon/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoidに
PDcontroller.soがないんじゃないかと思います。

あとは、これに関係あるかも。（ないかも）
https://github.com/start-jsk/rtmros_common/issues/936

ishiguroJSK, 2016-04-21:
`ｗｓtool update`
して
`catkin clean`
した後
`catkin build`
したら通りました(cleanしないとビルド通らなかったです)
`roslaunch hrpsys_choreonoid_tutorials jaxon_red_choreonoid.launch`
で無事立った状態で起動出来ました．
なにか変更点の中に原因となるものに心当たりがお有りでしょうか？

YoheiKakiuchi, 2016-04-22:
> なにか変更点の中に原因となるものに心当たりがお有りでしょうか？

hrpsys_choreonoid_tutorialsで依存関係は見なおしたけど、それがよかったのか。。。

closed 2016-04-22 by YoheiKakiuchi