start-jsk/rtmros_choreonoid #149: rtmros_choreonoidno

sshige, 2016-04-28:


YoheiKakiuchi, 2016-04-29:
debでインストールされてるでしょうか？

https://github.com/fkanehiro/openhrp3/pull/63
の関係だとおもっていて、 315.9 がdebになったら治ると期待しているのですが、
現状では、 jsk.rosbuild で --from-sourceでソースから入れて下さい。

iory, 2016-04-29:
debでros-indigo-openrhp3をインストールした場合でも，buildが通るようになりました．
@sshige くんのほうでも治ったかと思います．

k-okada, 2016-05-06:
@sshige 
rospack find openhrp3
してみてください．

yoshimalucky, 2016-07-19:
まったく違うエラーですが、ビルドできないので、ここで質問させてください。
末尾に全エラーログを添付しました。

CnoidBodyPlugin と BodyCustomizerInterface にリンクできていないことが原因のようですが、リンクの方法がわかりません。CMakeLists.txtに何か書くことで回避できるでしょうか。

ちなみに、find でこの2つを探した結果は以下です。

leus@y-tanaka-silver-stone:~$ sudo find / -name "*CnoidBody*"
/usr/share/locale/ja/LC_MESSAGES/CnoidBody-1.6.mo
/usr/share/locale/ja/LC_MESSAGES/CnoidBodyPlugin-1.6.mo
/usr/lib/libCnoidBody.so
/usr/lib/libCnoidBody.so.1.6
/usr/lib/choreonoid-1.6/libCnoidBodyPlugin.so

leus@y-tanaka-silver-stone:~$ sudo find / -name "*BodyCustomizerInterface*"
/home/leus/ros/indigo_parent/build/openhrp3/hrplib/hrpModel/CMakeFiles/hrpModel-3.1.dir/BodyCustomizerInterface.o
/home/leus/ros/indigo_parent/src/openhrp3/hrplib/hrpModel/BodyCustomizerInterface.cpp
/home/leus/ros/indigo_parent/src/openhrp3/hrplib/hrpModel/BodyCustomizerInterface.h
/home/leus/ros/indigo_parent/devel/include/OpenHRP-3.1/hrpModel/BodyCustomizerInterface.h
/usr/include/choreonoid-1.6/cnoid/src/Body/BodyCustomizerInterface.h
/usr/include/choreonoid-1.6/cnoid/BodyCustomizerInterface

以下、全エラーログです。

leus@y-tanaka-silver-stone:~/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid_tutorials$ catkin bt
==> Expanding alias 'bt' from 'catkin bt' to 'catkin b --this'
==> Expanding alias 'b' from 'catkin b --this' to 'catkin build --this'
-------------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /home/leus/ros/indigo_parent/devel:/opt/ros/indigo
Workspace:                   /home/leus/ros/indigo
-------------------------------------------------------------------------------
Source Space:       [exists] /home/leus/ros/indigo/src
Log Space:          [exists] /home/leus/ros/indigo/logs
Build Space:        [exists] /home/leus/ros/indigo/build
Devel Space:        [exists] /home/leus/ros/indigo/devel
Install Space:      [unused] /home/leus/ros/indigo/install
DESTDIR:            [unused] None
-------------------------------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
-------------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------------------------------
[build] Found '370' packages in 0.0 seconds.                                                                                                             
[build] Package table is up to date.                                                                                                                     
Starting  >>> actionlib_tutorials                                                                                                                        
Starting  >>> assimp_devel                                                                                                                               
Starting  >>> hrp2_models                                                                                                                                
Starting  >>> hrpsys_choreonoid                                                                                                                          
Starting  >>> jaxon_description                                                                                                                          
Starting  >>> jsk_models                                                                                                                                 
Starting  >>> jvrc_models                                                                                                                                
Starting  >>> multisense_description                                                                                                                     
Finished  <<< assimp_devel                               [ 0.8 seconds ]                                                                                 
Finished  <<< multisense_description                     [ 0.7 seconds ]                                                                                 
Finished  <<< hrp2_models                                [ 0.6 seconds ]                                                                                 
Finished  <<< jaxon_description                          [ 0.4 seconds ]                                                                                 
Starting  >>> openrtm_tools                                                                                                                              
Starting  >>> rtmbuild                                                                                                                                   
Finished  <<< jvrc_models                                [ 0.5 seconds ]                                                                                 
[... 21 lines cut ...]
[... cut ...]
[build]   Ignored:   352 packages were skipped or are blacklisted.                                                                                       
[build]   Warnings:  None.                                                                                                                               
[build]   Abandoned: 4 packages were abandoned.                                                                                                          
[build]   Failed:    1 packages failed.                                                                                                                  
[build] Runtime: 3.8 seconds total.

YoheiKakiuchi, 2016-07-20:
インストール方法はこれですが、
https://github.com/start-jsk/rtmros_choreonoid/blob/master/README.md

libcnoid-devがインストールされていないのではないかと思います。

yoshimalucky, 2016-07-20:
そのページの手順も既に試していて、JVRC公式に書いてあるいくつかdebのパッケージも入れてあります。

一応、一度消すのを試してからやっても結果変わらず：

sudo apt-get purge libcnoid-dev
sudo apt-get install libcnoid-dev

次に、http://wiki.ros.org/rtmros_common/Tutorials/GettingStart に書いてあるトラブルシューティングと同じように、
hrpsys_choreonoidとhrpsys_choreonoid_tutorialsのdevelとbuild以下のディレクトリを消したら上手くいきました。
そこまでのコマンド履歴が以下です。

rm -rf /home/leus/ros/indigo/devel/lib/hrpsys_choreonoid /home/leus/ros/indigo/build/hrpsys_choreonoid
rm -rf /home/leus/ros/indigo/build/hrpsys_choreonoid_tutorials/
source ~/.bashrc
catkin build hrpsys_choreonoid_tutorials

一応、自己解決することができましたが、コメントありがとうございました。

closed 2018-03-08 by sshige