start-jsk/rtmros_choreonoid #166: jaxon_red_choreonoid.launch起動時にCould not Connect ( /localhost:15005/○○○

ishiguroJSK, 2016-05-26:
一週間ほどぶりに
`rtmlaunch hrpsys_choreonoid_tutorials jaxon_red_choreonoid.launch
`
を実行しようとしたら
`[rtmlaunch] [ERROR] Could not Connect ( /localhost:15005/JAXON_RED.rtc:q , /localhost:15005/HrpsysSeqStateROSBridge0.rtc:rsangle ):  CORBA.COMM_FAILURE(omniORB.COMM_FAILURE_WaitingForReply, CORBA.COMPLETED_MAYBE) 
`
のようなエラーでChoreonoidが一瞬起動してすぐ落ちてしまいます．
他の人の環境でも確認されているようです．
`rtmlaunch hrpsys_ros_bridge_tutorials jaxon_red.launch 
`
は問題なく動作しています．
これはどこから疑うのがいいのでしょうか？

以下端末のフルログ
```
 $ rtmlaunch hrpsys_choreonoid_tutorials jaxon_red_choreonoid.launch 
[rtmlaunch] Start omniNames at port 15005 

Thu May 26 22:31:22 2016:

Starting omniNames for the first time.
Wrote initial log file.
Read log file successfully.
Root context is IOR:010000002b00000049444c3a6f6d672e6f72672f436f734e616d696e672f4e616d696e67436f6e746578744578743a312e300000010000000000000070000000010102000e0000003133332e31312e3231362e3237009d3a0b0000004e616d6553657276696365000300000000000000080000000100000000545441010000001c000000010000000100010001000000010001050901010001000000090101000354544108000000aafa465701003cc6
Checkpointing Phase 1: Prepare.
Checkpointing Phase 2: Commit.
Checkpointing completed.
WARNING: Package name "LR_foot_reader" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
... logging to /home/ishiguro/.ros/log/23984f58-2346-11e6-ae37-001f2905a411/roslaunch-ishiguroZ800-15552.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARNING: Package name "LR_foot_reader" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "LR_foot_reader" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "LR_foot_reader" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "LR_foot_reader" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "LR_foot_reader" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "LR_foot_reader" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "LR_foot_reader" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "LR_foot_reader" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "LR_foot_reader" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "LR_foot_reader" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "LR_foot_reader" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "LR_foot_reader" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "LR_foot_reader" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "LR_foot_reader" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "LR_foot_reader" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
started roslaunch server http://133.11.216.27:60392/

SUMMARY
========

PARAMETERS
[... 513 lines cut ...]
[rtmlaunch] Activate <- Inactive /localhost:15005/RangeSensorROSBridge0.rtc
[choreonoid-3] process has died [pid 15609, exit code 139, cmd /home/ishiguro/catkin_ws/jaxon/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid/launch/run_choreonoid.sh /home/ishiguro/catkin_ws/jaxon/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid_tutorials/config/JAXON_RED_FLAT.cnoid --enable-constraint --start-simulation -o manager.is_master:YES -o corba.nameservers:localhost:15005 -o naming.formats:%n.rtc -o logger.file_name:/tmp/rtc%p.log -endless -realtime -o manager.shutdown_onrtcs:NO -o manager.modules.load_path:/home/ishiguro/catkin_ws/jaxon/devel/share/hrpsys/lib -o manager.modules.preload:.so -o manager.components.precreate: -o example.SequencePlayer.config_file:/home/ishiguro/catkin_ws/jaxon/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid_tutorials/models/JAXON_JVRC.conf -o example.ForwardKinematics.config_file:/home/ishiguro/catkin_ws/jaxon/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid_tutorials/models/JAXON_JVRC.conf -o example.ImpedanceController.config_file:/home/ishiguro/catkin_ws/jaxon/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid_tutorials/models/JAXON_JVRC.conf -o example.AutoBalancer.config_f [... cut ...]
^C[multisense_local/jstate_bridge-38] killing on exit
[head_root_frame_id-37] killing on exit
[head_range_frame_id-36] killing on exit
[head_left_frame_id-35] killing on exit
[multisense_image_points2_color_relay-34] killing on exit
[multisense_organized_image_points2_relay-33] killing on exit
[multisense_local/pointcloud_bridge-32] killing on exit
[multisense/range_bridge-31] killing on exit
[multisense_right_image_rect_relay-30] killing on exit
[multisense_local/right/HEADRIGHT-29] killing on exit
The model was successfully loaded ! 

```

YoheiKakiuchi, 2016-05-26:
今日にapt-get dist-upgradeしたでしょうか？
aptitude show choreonoid して、20160526だと落ちるようになっています。
詳細なレポートできていませんが、OpenRTMが最新のためだと思います。
/var/cache/apt/archivesに最新ではないdebが残っていたら、それを使うと問題ないかと思います。
ソースからコンパイルしても問題ないです。

ishiguroJSK, 2016-05-26:
今日しました・・・
5/12あたりのchoreonoid関連の.debを4つほどdpkg -i して無事起動出来ました．
ありがどうございます．

snozawa, 2016-05-26:
このopenrtmのdebて、aist版でしょうか？
それともJSK版（tork-a release版）のdebでしょうか？

YoheiKakiuchi, 2016-05-26:
aistのdebですね。choreonoidのdebはaistのopenrtmのdebに依存していて、jskのとバージョンが違うので入れざるを得ません。
今回の件は、たぶんですが問題があるのはヘッダなので、choreonoidのdebだけ入れ替えれば動いています。

ishiguroJSK, 2016-05-30:
今日不覚にも`wstool update`してしまい，気づいたら再び動かなくなってしまったのですが，
その後，  
choreonoidのdebを以前のものに戻す， 
choreonoidをソースからインストールする， 
openrtmのdebを前に戻す(openrtm-aist-dev，openrtm-aistは05/25のものまでしかありませんでしたが)， 
など試しましたが依然エラーが出てしまいます．
正確にはソース版のchoreonoidなら落ちずに起動しますが，途中でネームサーバが落ちる(？)ようです．
実際にRT-system editor などで確認してもchoreonoid起動中にネームサーバが落ちており，ロボットは崩れ落ちています． 
hrpsysのアップデートによるものかと思ってgit checkoutで以前のものに戻してコンパイル後実行などしてみましたが効果なしでした． 
どこが悪さしてるか心当たりありますでしょうか・・・？
```
$ wstool info
workspace: /home/ishiguro/catkin_ws/jaxon/src

 Localname                          S  SCM Version (Spec)  UID  (Spec)  URI  (Spec) [http(s)://...]
 ---------                          -  --- --------------  -----------  ---------------------------
 trans_system                          git master  (-)     3029efa78972 github.com/jsk-ros-pkg/trans_system.git
 rtm-ros-robotics/rtmros_tutorials     git master  (-)     9651a62ceb5f github.com/start-jsk/rtmros_tutorials.git
 rtm-ros-robotics/rtmros_hrp2          git master  (-)     0308a2697271 github.com/start-jsk/rtmros_hrp2.git
 rtm-ros-robotics/rtmros_gazebo        git master  (-)     c16011ae09b8 github.com/start-jsk/rtmros_gazebo.git
 rtm-ros-robotics/rtmros_common     MC git master  (-)     c193f9c9175e github.com/start-jsk/rtmros_common.git
 rtm-ros-robotics/rtmros_choreonoid    git master  (-)     26a301002cdf github.com/start-jsk/rtmros_choreonoid.git
 openhrp3                              git master  (-)     4234293e18ac github.com/fkanehiro/openhrp3.git
 jsk-ros-pkg/jsk_openni_kinect         git master  (-)     2ee59a39b853 github.com/jsk-ros-pkg/jsk_openni_kinect.git
 jsk-ros-pkg/jsk_control               git master  (-)     cde35869ddb1 github.com/jsk-ros-pkg/jsk_control.git
 hrpsys                                git <detached>  (-) 36a39ac7d9cf github.com/fkanehiro/hrpsys-base.git

```

```
(略)
process[HrpsysJointTrajectoryBridge-6]: started with pid [30916]
[modelloader-2] process has finished cleanly
log file: /home/ishiguro/.ros/log/9c046f94-263e-11e6-bdc7-001f2905a411/modelloader-2*.log
process[hrpsys_state_publisher-7]: started with pid [30917]
process[hrpsys_ros_diagnostics-8]: started with pid [30965]
process[diagnostic_aggregator-9]: started with pid [30983]
process[hrpsys_profile-10]: started with pid [31007]
process[sensor_ros_bridge_connect-11]: started with pid [31029]
process[rtmlaunch_hrpsys_ros_bridge-12]: started with pid [31037]
process[SequencePlayerServiceROSBridge-13]: started with pid [31066]
process[DataLoggerServiceROSBridge-14]: started with pid [31085]
process[ForwardKinematicsServiceROSBridge-15]: started with pid [31111]
configuration ORB with localhost:15005
[ERROR] Connection Failed with the Nameserver (hostname=localhost port=15005).
Make sure the hostname is correct and the Nameserver is running.
CORBA.TRANSIENT(omniORB.TRANSIENT_ConnectFailed, CORBA.COMPLETED_NO)
process[StateHolderServiceROSBridge-16]: started with pid [31171]
process[AutoBalancerServiceROSBridge-17]: started with pid [31179]
process[StabilizerServiceROSBridge-18]: started with pid [31205]
[ERROR] Connection Failed with the Nameserver (hostname=localhost port=15005).
Make sure the hostname is correct and the Nameserver is running.
CORBA.TRANSIENT(omniORB.TRANSIENT_ConnectFailed, CORBA.COMPLETED_NO)
process[KalmanFilterServiceROSBridge-19]: started with pid [31229]
[sensor_ros_bridge_connect.py]  start
configuration ORB with localhost:15005
[ERROR] Connection Failed with the Nameserver (hostname=localhost port=15005).
Make sure the hostname is correct and the Nameserver is running.
CORBA.TRANSIENT(omniORB.TRANSIENT_ConnectFailed, CORBA.COMPLETED_NO)
process[ImpedanceControllerServiceROSBridge-20]: started with pid [31250]
[hrpsys_py-4] process has died [pid 30911, exit code 1, cmd /home/ishiguro/catkin_ws/jaxon/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid_tutorials/scripts/jaxon_red_setup.py JAXON_RED /home/ishiguro/catkin_ws/jaxon/src/rtm-ros-robotics/rtmros_choreonoid/jvrc_models/JAXON_JVRC/JAXON_JVRCmain_hrpsys.wrl -ORBInitRef NameService=corbaloc:iiop:localhost:15005/NameService __name:=hrpsys_py __log:=/home/ishiguro/.ros/log/9c046f94-263e-11e6-bdc7-001f2905a411/hrpsys_py-4.log].
log file: /home/ishiguro/.ros/log/9c046f94-263e-11e6-bdc7-001f2905a411/hrpsys_py-4*.log
[sensor_ros_bridge_connect-11] process has died [pid 31029, exit code 1, cmd /home/ishiguro/catkin_ws/jaxon/src/rtm-ros-robotics/rtmros_common/hrpsys_ros_bridge/scripts/sensor_ros_bridge_connect.py /home/ishiguro/catkin_ws/jaxon/src/rtm-ros-robotics/rtmros_choreonoid/jvrc_models/JAXON_JVRC/JAXON_JVRCmain_hrpsys.wrl JAXON_RED HrpsysSeqStateROSBridge0 localhost new all 50.0 -ORBInitRef NameService=corbaloc:iiop:localhost:15005/NameService __name:=sensor_ros_bridge_connect __log:=/home/ishiguro/.ros/log/9c046f94-263e-11e6-bdc7-001f2905a411/sensor_ros_bridge_connect-11.log].
log file: /home/ishiguro/.ros/log/9c046f94-263e-11e6-bdc7-001f2905a411/sensor_ros_bridge_connect-11*.log
process[RemoveForceSensorLinkOffsetServiceROSBridge-21]: started with pid [31292]
process[EmergencyStopperServiceROSBridge-22]: started with pid [31318]
process[HardEmergencyStopperServiceROSBridge-23]: started with pid [31376]
process[footcoords-24]: started with pid [31397]
process[stabilizer_watcher-25]: started with pid [31408]
process[rtmlaunch_vision_connect-26]: started with pid [31422]
[... 29 lines cut ...]
[ INFO] [1464596106.594505613]: [HrpsysSeqStateROSBridge] @Initilize name : HrpsysSeqStateROSBridge0
[... cut ...]
[ INFO] [1464596106.787058956]: [ImageSensorROSBridge] @Initilize name : HEADLEFT
[ INFO] [1464596106.805318031]: [ImageSensorROSBridge] @Initilize name : HEADRIGHT
Loading body customizer "/home/ishiguro/catkin_ws/jaxon/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid/JAXONCustomizer.so" for JAXON_JVRC
create customizer : JAXON_JVRC
create customizer : JAXON_JVRC
cloneMap.setNonNodeCloning(false);
cloneMap.setNonNodeCloning(false);
cloneMap.setNonNodeCloning(false);
^C[multisense_local/jstate_bridge-38] killing on exit

```

YoheiKakiuchi, 2016-05-30:
debで入れなおしたらいいのは、libcnoid1, libcnoid-devだろうと思います。choreonoid本体と、openrtmのリンクはないので。

ソースからインストールする場合の注意点としては、enable_corba, build_corbaplugin, build_openrtmpluginをチェックすることと、ros/indigo_parent/devel をopenrtm directoryとして指定することです。

ishiguroJSK, 2016-05-30:
libcnoid1, libcnoid-devを前回同様以前のものに戻す方法を試した場合にも今回はダメでした．
違う種類のエラーという可能性は無いのでしょうか？

YoheiKakiuchi, 2016-05-30:
だとすると、なにが悪いのだろう？
以下はどうなるでしょうか？

chireonoid単体で立ち上がるか
$ choreonoid .....
みたいな行がログのどこかにあるので、そのコマンドを /tmp ディレクトリで実行してみる。
rtm-naming 15005をほかのターミナルで実行しておく必要がある。

hrpsysだけにするとどうなるか
jaxon_red_choreonoid.launch で ros_bridge_choreonoid.launch をコメントアウトする。

ishiguroJSK, 2016-05-30:
この理不尽な感じは・・・と思ってPC再起動したら治ってしまいました・・・ 
何とも煮え切らない感じですが，裏で何か残ってたんですかね・・・
現状choreonoidがソース版で各種debとワークスペースのブランチは全部最新の状態です．  
ちなみに`sudo make install`したchoreonoidをアンインストールしたい場合は/usr/local/include,share,lib,binのそれらしいものを消せばいいのでしょうか？ 
(一回試した時catkin buildが通らなくなってすぐ戻してしまったのですが)

ishiguroJSK, 2016-05-30:
すいません自己解決しました．
> /usr/local/include,share,lib,binのそれらしいものを消せばいい 

で合ってました．再起動してから全部うまく行っています．

closed 2016-05-30 by ishiguroJSK

YoheiKakiuchi, 2016-06-02:
20160602にビルドされたdebだと落ちないようになっているかと思います。
https://github.com/s-nakaoka/choreonoid/issues/104

ちなみに、インストールディレクトリを/usr/local/choreonoid などにしておくとアンインストールが楽です。
（パスを通す必要はありますが。。。）