start-jsk/rtmros_choreonoid #189: 2016/06/29あたりからjaxon_red_choreonoid.launchに対してhcfが繋がらない

ishiguroJSK, 2016-07-01:
2016/06/29時点でのChoreonoidのdebにおいて
 `rtmlaunch hrpsys_choreonoid_tutorials jaxon_red_choreonoid.launch `
に対してhrpsys_choreonoid_tutorials/scriptの
`ipython -i jaxon_red_setup.py "JAXON_RED(Robot)0"`
が接続できなくなっているようです．
`rtmlaunch hrpsys_choreonoid_tutorials chidori_choreonoid.launch `
`ipython -i chidori_rh_setup.py "CHIDORI(Robot)0"`
では出来ています．
`rtls`の結果がCHIDORIの場合正常な
localhost:15005
となっているところがJAXON_REDだと
CORBA.MARSHAL(omniORB.MARSHAL_MessageSizeExceedLimit, CORBA.COMPLETED_YES)
という怪しげな文字列になっています．

現状choreonoidを以前のdebに戻すことで解決されていますが，
CHIDORIでできているということはJAXONのscriptをアップデートすることで治ったりしないでしょうか？

YoheiKakiuchi, 2016-07-01:
直感的な返しですが、以下で環境変数を設定するとどうなるでしょう？

```export ORBgiopMaxMsgSize=2097152000```

ishiguroJSK, 2016-07-01:
一瞬で治ってしまいました・・・
この方法は何をしてくれるものなんですか？

YoheiKakiuchi, 2016-07-01:
CORBAのメッセージの最大サイズを変更するものです。
```omniORB.MARSHAL_MessageSizeExceedLimit``` を読んでのカンです。
が、あるあるな話で、model-loaderなどは、設定なしでは使い物になりません。

なにが変わって必要になったかわからないですが、どこかで通信しているパケットサイズを大きくする修正
（ポートを増やすとか、メッセージが大きくなるとか、）があったのだろうと思います。

ishiguroJSK, 2016-07-01:
なるほど・・・
確かにCHIDORIは上半身もビジョンも無かったりしますからね・・・
ありがとうございました．

closed 2016-07-01 by ishiguroJSK