start-jsk/rtmros_choreonoid #190: 歩くと左に傾く

YutaKojio, 2016-07-01:
https://github.com/s-nakaoka/choreonoid/issues/111 の解決により，
今までは両足立ちでstを入れると右に傾いていたのが，傾かなくなりました．
しかし，この状態でgoposすると左に傾き気味になり，たまに転けるようになりました．

https://github.com/start-jsk/rtmros_choreonoid/blob/master/jvrc_models/JAXON_JVRC/JAXON_JVRCmain_hrpsys_bush.wrl#L240-L245 を```rotation 1 0 0 0```にして，
あえて右に傾かせると今まで通りにきれいに歩けます．

snozawa, 2016-07-01:
どこかに
s-nakaoka/choreonoid#111
対策のhotfixコードが残ってたりしないかな？

YoheiKakiuchi, 2016-07-01:
https://drive.google.com/open?id=0B5IZpi1aPoszNU9VT09sN1ZoN2M

僕の手元の最新と思う環境です。(ソースの修正なし、wrlの修正も入っていない）
(choreonoidはdebで、20160629のバージョン)

止まっている時の JAXON_RED/WAISTのRollは0.0degで、このときkf.rtcの出力は
wrlで傾けている分だけ、傾いていました。

歩いているときは、+0.4 ~ -0.6 deg ぐらいRollがふらつきますが、問題ない範囲かと思います。
これとは明らかに違いますか？

YutaKojio, 2016-07-01:
歩けている人と歩けない人がいる理由はわかりました．
st-algorithmがEEFMだと歩けてEEFMQPだとおかしくなります．
https://drive.google.com/open?id=0B9Cu2eewhmiwRnQ3SHJMT256b0E

hrpsysをビルドするときにDUSE_QPOASES=ONにしていいない人はEEFMなので歩けているようです．
EEFMQPのバグだとしても右に傾けるとなぜうまくいくのかはわからないです．．．

snozawa, 2016-07-01:
あれこれ実機でも一時期おきてませんでしたっけ？
２月くらいに。
それで、取り急ぎでEEFMQPCOPにすると動くようになってた気がします。

YutaKojio, 2016-07-01:
EEFMQPCOPにすると歩けました．
choreonoidでimuが傾いていたバージョンでずっと歩けていたのはたまたまで，
最近転けるようになったのはchoreonoidの問題ではなく，hrpsys側の問題ということでよさそうです．

snozawa, 2016-07-01:
そうするとJAXONで実機もsimulatorもEEFMQPだと歩けないような状況のきがしますね。
取り急ぎ問題ないのであれば、EEFMQPCOPがいいような気がします。
（本当は調べておきたいですが）

YoheiKakiuchi, 2016-07-01:
> hrpsysをビルドするときにDUSE_QPOASES=ONにしていいない人はEEFMなので歩けているようです．

これは、hrpsysを(indigo_parentなどのソースで)ただcatkin buildするだけではなくて、
オプションをつけてビルドする必要があるということでしょうか？
それで、体内はそうなっていますか？

ロボットがそこそこ可動実績あるなら、ロボットに入っているdiffはとりあえずコミット
してもらってロボット内にdiffがある状況をなくしていきましょう。

snozawa, 2016-07-01:
>  それで、体内はそうなっていますか？

手元でも体内でもそうしています。
ただ、このQPOASESが必要なSTのモード（EEFMQP）自体最近あまり使ってないと思います。
このSTモードを変えるdiffは、体内に残ってる気がしてきたので、ここ最近動いてきてることを
考えるとコミットでよいかもしれません。

YoheiKakiuchi, 2016-09-05:
https://github.com/start-jsk/rtmros_tutorials/pull/489
ここで変更しようとしているかと思います。

> hrpsysをビルドするときにDUSE_QPOASES=ONにしていいない人はEEFMなので

EEFMQPCOP は 同じく DUSE_QPOASES=ON が必要でしょうか？

YutaKojio, 2016-09-05:
> EEFMQPCOP は 同じく DUSE_QPOASES=ON が必要でしょうか？

いらないです．

YutaKojio, 2017-11-14:
https://github.com/start-jsk/rtmros_tutorials/pull/489 でとりあえず歩くようにはなりました．

closed 2017-11-14 by YutaKojio