start-jsk/rtmros_choreonoid #198: 最新の状態(2016/07~?)でchidori_choreonoid.launchのSTが効かない

ishiguroJSK, 2016-08-02:
急ぎではないのですが，
何度かワークスペースを最新に更新したときに気のせいかとも思っていたのですが，
やはり最新の状態で以下のコマンドでCHIDORIだけSTが入らない気がします(JAXONは入ることを確認しました)．
`rtmlaunch hrpsys_choreonoid_tutorials chidori_choreonoid.launch`
比較的初期のワークスペース(2016/06~ hrpsys_choreonoid_tutorialsだと以下のあたり)
```
$ git log
commit cef6d0be04dfa1b1f245ca654619c185ed1800d5
Merge: 5d9aff0 4654526
Author: Yohei Kakiuchi <youhei@jsk.imi.i.u-tokyo.ac.jp>
Date:   Wed Jun 15 23:55:24 2016 +0900

    Merge pull request #184 from YoheiKakiuchi/fix_customizer
    
    [CHIDORI] change to use bush model
```
で機能していたので色々比較していたのですが，原因がわかりません．
端末のログのdiffを見ているとABCやST周りの初期設定(?)がごっそり抜け落ちてる感じです．
(といってもgoPosはできているようです)

[image]

そこ以降，
`rtconf localhost:15005/st.rtc set debugLevel 1`
などでSTの状態を見ようとしても意味のある情報が返ってこない感じです．

[image]

ちなみに最新といってもchoreonoid-1.5⇒choreonoid-1.6にはしていません．
また，あまり影響なさそうですが
/rtmros_tutorials/hrpsys_ros_bridge_tutorials/CMakeLists.txtの
「--conf-file-option "pdgains.file_name: ${PROJECT_SOURCE_DIR}/models/CHIDORI_PDgains.sav"」
は
「--conf-file-option "pdgains.file_name: ${PROJECT_SOURCE_DIR}/models/CHIDORI.PDgains.sav"」
な気がします．(これを直しても挙動は変わらなかったのですが…)

[chidori_choreonoid_terminal_log_new.txt](https://github.com/start-jsk/rtmros_choreonoid/files/398330/chidori_choreonoid_terminal_log_new.txt)
[chidori_choreonoid_terminal_log_old.txt](https://github.com/start-jsk/rtmros_choreonoid/files/398331/chidori_choreonoid_terminal_log_old.txt)

snozawa, 2016-08-03:
起動時にoldのほうで呼ばれていて、newのほうでよばれてないのは、
hrpsysにpythonで初期設定をするスクリプトのようです。

chidori_choreonoid.launchの中でchidori_setup.pyというのが呼ばれて入れば
OKそうなんだけど、たどって確認できるかな？

ishiguroJSK, 2016-08-03:
chidori_rh_setup.pyの中の
「hcf = CHIDORI_HrpsysConfigurator("CHIDORI(Robot)0")」
↓
「hcf = CHIDORI_HrpsysConfigurator("CHIDORI")」
が正解っぽいです．
(chidori_setup.pyは最新版では廃止されているようです)

ちなみに○○_setup.pyが正しく実行されているか否かはログメッセージなどで分かるのですか？
それとも[abc]や[st]のパラメータがセットされたか否かから類推したのですか？

snozawa, 2016-08-03:
> 「hcf = CHIDORI_HrpsysConfigurator("CHIDORI(Robot)0")」
> ↓
> 「hcf = CHIDORI_HrpsysConfigurator("CHIDORI")」

あれ、垣内さん、こちらRobot名でなくて、ロボット名(Robot)0に統一しませんでしたっけ？
確か、どちらでも良いということで、openhrp3, hrpsys-simulatorの命名足と
互換性を保つように、(Robot)0をつけるようにしたと思います。
なので、(Robot)0で動くようにcnoidファイルかな？設定を変える必要がありそうです。

> ちなみに○○_setup.pyが正しく実行されているか否かはログメッセージなどで分かるのですか？
> それとも[abc]や[st]のパラメータがセットされたか否かから類推したのですか？

これは、[abc], [st]の設定パラメータがセットされてる箇所が
ログのなかになかったので、そうだと判断しました。
（内部むけページですが）
https://github.com/jsk-ros-pkg/trans_system/tree/master/jaxon_ros_bridge#upstream%E3%81%AE%E3%83%95%E3%82%A1%E3%82%A4%E3%83%AB%E3%81%AEinclude%E3%81%AA%E3%81%A9%E3%81%AE%E9%96%A2%E4%BF%82
に実機の例があります。
実機やシミュレータで、共通部分継承しつつ初期設定ファイル（今回の場合はhrpsysのxx.py）が呼ばれるようになってます。
なかでは、set st paramやset auto balancer param的なことをおこなっており、
それがhrpsysのログにでてくるはず、というのを佐賀しまいS他。

ishiguroJSK, 2016-08-03:
なるほど．今後のデバッグの参考になります．ありがとうございます．

> こちらRobot名でなくて、ロボット名(Robot)0に統一しませんでしたっけ？

chidori_rh_setup.py内のCHIDORI_HrpsysConfigurator("○○")は辿った感じ，最終的にurata_hrpsys_config.pyの
機体ハードウェアに依存した設定にたどり着いているので，(Robot)0を付けなくていいのではないでしょうか？
(JAXONや他の機体もそのようですし)
```
    def setStAbcParameters (self):
        if self.ROBOT_NAME == "STARO":
            self.setStAbcParametersSTARO()
        elif self.ROBOT_NAME == "JAXON":
            self.setStAbcIcParametersJAXON(foot="LEPTRINO")
        elif self.ROBOT_NAME == "JAXON_RED":
            self.setStAbcIcParametersJAXON(foot="KAWADA")
        elif self.ROBOT_NAME == "URATALEG":
            self.setStAbcParametersURATALEG()
        elif self.ROBOT_NAME == "YSTLEG":
            self.setStAbcParametersYSTLEG()
        elif self.ROBOT_NAME == "CHIDORI":
            self.setStAbcParametersCHIDORI()
        elif self.ROBOT_NAME == "TQLEG0":
            self.setStAbcParametersTQLEG0()
```

一方でhrpsys-simulator系のlaunchにはchidori_choreonoid.launch内からロボット名(Robot)0を渡しているっぽいので，
懸念されていることはこちらで対応されているのではないでしょうか？(あくまで推測ですが)

```
  <arg name="SIMULATOR_NAME" default="CHIDORI(Robot)0" />
  <arg if="$(arg USE_ROBOTHARDWARE)"
       name="BRIDGE_SIMULATOR_NAME" default="RobotHardware_choreonoid0" />
(略)
  <!-- hrpsys -->
  <include file="$(find hrpsys_choreonoid)/launch/startup_choreonoid.launch" >
    <!-- robot dependant settings -->
    <arg name="SIMULATOR_NAME" value="$(arg SIMULATOR_NAME)" />
    <arg name="PROJECT_FILE"   value="$(arg PROJECT_FILE)" />
    <arg name="MODEL_FILE"     value="$(arg MODEL_FILE)" />
    <arg name="CONF_FILE"      value="$(arg CONF_FILE)"/>
    <arg name="HRPSYS_PY_PKG"  value="$(arg HRPSYS_PY_PKG)"/>
    <arg name="HRPSYS_PY_NAME" value="$(arg HRPSYS_PY_NAME)"/>
    <!-- -->
    <arg name="KILL_SERVERS" value="$(arg KILL_SERVERS)" />
    <arg name="NOSIM" value="$(arg NOSIM)" />
    <arg name="REALTIME" value="$(arg REALTIME)" />
    <arg name="GUI" value="$(arg GUI)" />
    <arg name="hrpsys_precreate_rtc" value="$(arg hrpsys_precreate_rtc)" />
    <arg name="CONNECT_CONSTRAINT_FORCE_LOGGER_PORTS" default="$(arg CONNECT_CONSTRAINT_FORCE_LOGGER_PORTS)"/>
  </include>
  <!-- ros_bridge -->
  <rosparam command="load" file="$(arg CONTROLLER_CONFIG)" />
  <include file="$(find hrpsys_choreonoid)/launch/ros_bridge_choreonoid.launch" >
    <!-- robot dependant settings -->
    <arg name="SIMULATOR_NAME" value="$(arg BRIDGE_SIMULATOR_NAME)" />
    <arg name="MODEL_FILE"     value="$(arg MODEL_FILE)" />
    <arg name="COLLADA_FILE"   value="$(arg COLLADA_FILE)"/>
    <arg name="CONF_FILE"      value="$(arg CONF_FILE)" />
  </include>
```

YoheiKakiuchi, 2016-08-03:
> あれ、垣内さん、こちらRobot名でなくて、ロボット名(Robot)0に統一しませんでしたっけ？

それは、BodyRTCのインスタンス名ですね。
ここは、実機のhrpsysを使う関係上、robotnameである必要があります。

そもそも、実機用のURATAHrpsysConfiguratorで、
コンストラクタのときと、initに渡すロボット名の意味が違っています。
コンストラクタに渡すのは、ロボットの名前でinitに渡すのはインスタンス名かと思います。
https://github.com/start-jsk/rtmros_tutorials/blob/master/hrpsys_ros_bridge_tutorials/src/hrpsys_ros_bridge_tutorials/urata_hrpsys_config.py#L12

なので、
> 機体ハードウェアに依存した設定にたどり着いているので，(Robot)0を付けなくていいのではないでしょうか？
> (JAXONや他の機体もそのようですし)

これが正解です。

snozawa, 2016-08-03:
お、そうっぽいですね。
今回のものは、インスタンス名でなくてロボット名を師弟すべきところで、
インスタンス名が師弟されてしまってたということですね。

snozawa, 2016-08-03:
k

ishiguroJSK, 2016-08-04:
修正がマージされたようなのでcloseします．
お二方ともありがとうございました．

closed 2016-08-04 by ishiguroJSK