start-jsk/rtmros_choreonoid #205: [hrpsys_choreonoid_tutorials] config/flat.yaml.inがないと言われてbuildに失敗する

mttamtam, 2016-08-07:
ノートPCでもデスクトップでも、

roscd hrpsys_choreonoid_tutorials
catkin bt
すると、

CMake Error: File /home/tamura/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid_tutorials/config/flat.yaml.in does not exist.
CMake Error at CMakeLists.txt:174 (configure_file):
  configure_file Problem configuring file

というふうに言われてしまうようになったので、とりあえず
https://github.com/start-jsk/rtmros_choreonoid/blob/master/hrpsys_choreonoid_tutorials/CMakeLists.txt#L174
この行をコメントアウトしてビルドしたら成功しました。
((CMakeLists.txtの読み方を知らないのですが、これはflat.yaml.inからflat.yamlを作ろうとしているのでしょうか？)
(ほかの.inファイルがあるcnoidファイルはgit statusで見てみるとUntracked filesとして出てくるので）)

#203 で変更が加わっている箇所なので、これでファイルが入ってなかったのが原因のように思ったのですが、どうでしょうか？

YoheiKakiuchi, 2016-08-07:
すいません、コミット忘れでした。

mttamtam, 2016-08-07:
#206 でビルドできるようになりました．
ありがとうございます．

closed 2016-08-08 by mttamtam