start-jsk/rtmros_choreonoid #211: choreonoidが頻繁に落ちる

mttamtam, 2016-09-28:
二ヶ月ぐらい前から、シミュレーションを立ち上げて割とすぐにchoreonoidが死んでしまうことが多いのですが、
何か原因と対策方法をご存じの方はいますか？？
安定しているときは何時間でも動き続けるのですが。
choreonoidはデスクトップPCではaptで入れてあって、ソースから入れたものをノートPCで使っていますが、どちらでも起きてしまいます。

YoheiKakiuchi, 2016-09-28:
見てみます。

YoheiKakiuchi, 2016-09-30:
現状分かったところの報告

choreonoid + hrpsys は落ちない
choreonoid + hrpsys + vision系() は落ちない(試行50回程度)
choreonoid + hrpsys + ROSBridge は落ちない(試行30回程度)

choreonoid + hrpsys + vision系 + ROSBridge 1,2割の確率で落ちそう
13回落ちた/100回の試行 （１つの試行は開始から3分間choreonoidの実行
を見る、落ちていなかったら次の試行）

ここに貼ったが、gdbの結果からはあまり分かることがない。
https://github.com/s-nakaoka/choreonoid/issues/127
openrtmでメッセージを送ろうとしているようなところに見える。

https://github.com/start-jsk/rtmros_choreonoid/pull/212
において、

- constraintのポートを作らなくする
- rtmlaunch.pyでのチェックを止める

をしたが効果があるような気もするしないような気もする。
（落ちるが頻度が減っていないか？？）

mttamtam, 2016-09-30:
調べていただきありがとうございます。
choreonoidにつなぐものを減らすと落ちなくなるのですね。
毎回起きる問題というわけではないので（しかも起きる時と起きない時とまとまっている気がしていますが、気のせいでしょうか）、しばらく使ってどうなったか改めて報告します。

hikun5296, 2016-10-19:
これと関連があるのか分からないですが、今日シミュレータを立ち上げようとしたところ(`rtmlaunch hrpsys_choreonoid_tutorials chidori_choreonoid.launch`)
エラーでchoreonoidが落ちるようになってしまいました。

環境: choreonoid(deb) + hrpsys等(source)
思い当たる節は月曜くらいにchoreonoidのアップグレード(`apt-get upgrade`)をしたくらいだと思います

何か解決方法等分かる方はいますでしょうか？

```
*** Error in `/usr/bin/choreonoid': malloc(): memory corruption: 0x0000000001638cd0 ***

Program received signal SIGABRT, Aborted.
0x00007ffff5f10c37 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56	../nptl/sysdeps/unix/sysv/linux/raise.c: そのようなファイルやディレクトリはありません.

(gdb) bt
#0  0x00007ffff5f10c37 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007ffff5f14028 in __GI_abort () at abort.c:89
#2  0x00007ffff5f4d2a4 in __libc_message (do_abort=1, fmt=fmt@entry=0x7ffff605b6b0 "*** Error in `%s': %s: 0x%s ***\n") at ../sysdeps/posix/libc_fatal.c:175
#3  0x00007ffff5f5ae26 in malloc_printerr (ptr=0x1638cd0, str=0x7ffff6057882 "malloc(): memory corruption", action=<optimized out>) at malloc.c:4996
#4  _int_malloc (av=0x7ffff6298760 <main_arena>, bytes=41) at malloc.c:3447
#5  0x00007ffff5f5c6c0 in __GI___libc_malloc (bytes=41) at malloc.c:2891
#6  0x00007ffff6513dad in operator new(unsigned long) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff656f249 in std::string::_Rep::_S_create(unsigned long, unsigned long, std::allocator<char> const&) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#8  0x00007ffff6570971 in char* std::string::_S_construct<char const*>(char const*, char const*, std::allocator<char> const&, std::forward_iterator_tag) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#9  0x00007ffff6570d88 in std::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(char const*, std::allocator<char> const&) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#10 0x00007fffc5284248 in RTC::Logger::Logger(char const*) () from /usr/lib/x86_64-linux-gnu/libRTC-1.1.2.so
#11 0x00007fffc52ff8db in RTC::PortBase::PortBase(char const*) () from /usr/lib/x86_64-linux-gnu/libRTC-1.1.2.so
#12 0x00007fffc52e61de in RTC::InPortBase::InPortBase(char const*, char const*) () from /usr/lib/x86_64-linux-gnu/libRTC-1.1.2.so
#13 0x00007fffc5956c1d in RTC::InPort<RTC::TimedDoubleSeq>::InPort(char const*, RTC::TimedDoubleSeq&, int, bool, bool, int, int) ()
   from /usr/lib/choreonoid-1.6/libCnoidOpenRTMPlugin.so
#14 0x00007fffbb4e1eb5 in open_iob () from /home/suzuki/ros/indigo/devel/.private/hrpsys_choreonoid/lib/libhrpIo_choreonoid.so
#15 0x00007fffbb76f477 in robot::init() () from /home/suzuki/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid/lib/RobotHardware_choreonoid.so
#16 0x00007fffbb782e3c in RobotHardware::onInitialize() ()
   from /home/suzuki/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid/lib/RobotHardware_choreonoid.so
#17 0x00007fffbb77d7ab in RobotHardware_choreonoid::onInitialize() ()
   from /home/suzuki/ros/indigo/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid/lib/RobotHardware_choreonoid.so
#18 0x00007fffc529bc29 in RTC::RTObject_impl::on_initialize() () from /usr/lib/x86_64-linux-gnu/libRTC-1.1.2.so
#19 0x00007fffc52a3539 in RTC::RTObject_impl::initialize() () from /usr/lib/x86_64-linux-gnu/libRTC-1.1.2.so
#20 0x00007fffc52598f2 in RTC::Manager::createComponent(char const*) () from /usr/lib/x86_64-linux-gnu/libRTC-1.1.2.so
#21 0x00007fffc58f2723 in cnoid::createManagedRTC(char const*) () from /usr/lib/choreonoid-1.6/libCnoidOpenRTMPlugin.so
#22 0x00007fffc5907c30 in ?? () from /usr/lib/choreonoid-1.6/libCnoidOpenRTMPlugin.so
#23 0x00007fffc59086bd in ?? () from /usr/lib/choreonoid-1.6/libCnoidOpenRTMPlugin.so
#24 0x00007fffc5909078 in ?? () from /usr/lib/choreonoid-1.6/libCnoidOpenRTMPlugin.so
#25 0x00007fffc5937801 in cnoid::BodyRTCItem::createRTC(cnoid::ref_ptr<cnoid::Body>) () from /usr/lib/choreonoid-1.6/libCnoidOpenRTMPlugin.so
#26 0x00007fffc593c324 in cnoid::BodyRTCItem::onPositionChanged() () from /usr/lib/choreonoid-1.6/libCnoidOpenRTMPlugin.so
#27 0x00007ffff7a5bb0f in cnoid::Item::callSlotsOnPositionChanged() () from /usr/bin/../lib/libCnoidBase.so.1.6
#28 0x00007ffff7a5cb34 in cnoid::Item::doInsertChildItem(cnoid::Item*, cnoid::Item*, bool) () from /usr/bin/../lib/libCnoidBase.so.1.6
#29 0x00007ffff7abd79d in ?? () from /usr/bin/../lib/libCnoidBase.so.1.6
#30 0x00007ffff7abca79 in ?? () from /usr/bin/../lib/libCnoidBase.so.1.6
#31 0x00007ffff7abca79 in ?? () from /usr/bin/../lib/libCnoidBase.so.1.6
#32 0x00007ffff7abca79 in ?? () from /usr/bin/../lib/libCnoidBase.so.1.6
#33 0x00007ffff7abe53f in cnoid::ItemTreeArchiver::restore(cnoid::Archive*, cnoid::Item*, std::set<std::string, std::less<std::string>, std::allocator<std::string> > const&) () from /usr/bin/../lib/libCnoidBase.so.1.6
#34 0x00007ffff7a0b048 in ?? () from /usr/bin/../lib/libCnoidBase.so.1.6
#35 0x00007ffff7a0b934 in ?? () from /usr/bin/../lib/libCnoidBase.so.1.6
#36 0x00007ffff7aa9417 in cnoid::OptionManager::parseCommandLine(int, char**) () from /usr/bin/../lib/libCnoidBase.so.1.6
#37 0x00007ffff7a05e3b in cnoid::App::exec() () from /usr/bin/../lib/libCnoidBase.so.1.6
#38 0x000000000040102b in main ()
```

YoheiKakiuchi, 2016-10-19:
詳細まで突っ込んで見れていませんが、choreonoidをソースから入れると大丈夫なようです。
（この場合ソースからいれるとは、ros/xxx_parent/src/openrtm-aist とリンクするということ）
choreonoidのdebはopenrtm-aistのdebとリンクされていて、そのバージョンがros-xxx-openrtm-aistと違うことでおきているのではないかと推測されます。

YoheiKakiuchi, 2016-10-19:
Readmeにchoreonoidのソースからのインストール方法を書きました。
https://github.com/start-jsk/rtmros_choreonoid/blob/master/README.md

hikun5296, 2016-10-19:
choreonoid(source) + hrpsys等(source)で
問題なく動きました
ありがとうございます

mttamtam, 2016-12-06:
@terasawa さんと @YutaKojio さんによると、roscoreを立ち上げておくと落ちないみたいです。
（しかも、ロボットがこけてchoreonoidを立ち上げなおしても、roseusのほうは立ち上げ直さずとも*ri*をそのまま使ってangle-vectorを送れる。)

YoheiKakiuchi, 2016-12-07:
> @terasawa さんと @YutaKojio さんによると、roscoreを立ち上げておくと落ちないみたいです。

理由はわかりませんが、ちょっとマシなようですね。

前回は以下の通りでしたが、

> 13回落ちた/100回の試行 （１つの試行は開始から3分間choreonoidの実行を見る、落ちていなかったら次の試行）

今回は、同じ方法で 4回落ち  / 100回試行 でした。