start-jsk/rtmros_choreonoid #224: chidoriだけ落ちる

hikun5296, 2017-04-25:
hrpsys+rtmros_common+rtmros_tutorials+rtmros_choreonoidとchoreonoid本体はsrcで
それ以外はdebで入れている環境なのですが、
chidori_choreonoid.launchを立ち上げたときだけ落ちます
jaxon_red_choreonoid.launchでは問題なく立ち上がります

```
Traceback (most recent call last):
  File "/home/suzuki/ros/default/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid_tutorials/scripts/chidori_rh_setup.py", line 45, in <module>
    hcf.init(sys.argv[1], sys.argv[2], connect_constraint_force_logger_ports=connect_constraint_force_logger_ports)
  File "/home/suzuki/ros/default/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid_tutorials/src/hrpsys_choreonoid_tutorials/choreonoid_hrpsys_config.py", line 41, in init
    URATAHrpsysConfigurator.init(self, robotname, url)
  File "/home/suzuki/ros/default/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/src/hrpsys_ros_bridge_tutorials/urata_hrpsys_config.py", line 22, in init
    self.setStAbcParameters()
  File "/home/suzuki/ros/default/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/src/hrpsys_ros_bridge_tutorials/urata_hrpsys_config.py", line 63, in setStAbcParameters
    self.setStAbcParametersCHIDORI()
  File "/home/suzuki/ros/default/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/src/hrpsys_ros_bridge_tutorials/urata_hrpsys_config.py", line 339, in setStAbcParametersCHIDORI
    stp=self.st_svc.getParameter()
  File "/home/suzuki/ros/default/devel/lib/python2.7/dist-packages/hrpsys/StabilizerService_idl.py", line 255, in getParameter
    return _omnipy.invoke(self, "getParameter", _0_OpenHRP.StabilizerService._d_getParameter, args)
omniORB.CORBA.COMM_FAILURE: CORBA.COMM_FAILURE(omniORB.COMM_FAILURE_WaitingForReply, CORBA.COMPLETED_MAYBE)
/home/suzuki/ros/default/src/rtm-ros-robotics/rtmros_choreonoid/hrpsys_choreonoid/launch/run_choreonoid.sh: 76 行: 31424 Segmentation fault      (コアダンプ) $choreonoid_exe $enable_const $add_objects $cnoid_proj $start_sim
```
心当たりや解決策等分かる方いませんでしょうか？

ishiguroJSK, 2017-04-25:
確かに落ちますね・・・
しかも
```
rtmlaunch hrpsys_ros_bridge_tutorials chidori.launch
hrpsyspy
```
からだと`hcf.st_svc.getParameter()`は普通に呼べるという・・・

kyawawa, 2017-04-25:
@hikun5296 hrpsysのdebが入っていると落ちるみたいですが，どうでしょうか？
今日思い出して，調査が進んでおらず，debだけだと落ちないのか，なぜchidoriだけ落ちるのかは分かっていません…

ishiguroJSK, 2017-04-25:
これは確かにros-indigo-hrpsysの有無のやつっぽいですね・・・
消したら上がるのを確認しました．

hikun5296, 2017-04-25:
同じく解決しました
ありがとうございました

kyawawa, 2017-04-26:
良くない現象だとは思うので，openにしておいてもらえますか？

kyawawa, 2017-04-26:
hrpsysのsrcとdebが共存している状態で，USE_ROBOTHARDWAREがtrueだとchidori, jaxon共に落ちるみたいです

YoheiKakiuchi, 2017-04-26:
> hrpsysのsrcとdebが共存している状態で，USE_ROBOTHARDWAREがtrueだとchidori, jaxon共に落ちるみたいです

なんとなく思い出しましたが、そうだった気がします。

kyawawa, 2017-04-26:
https://github.com/start-jsk/rtmros_choreonoid/blob/master/hrpsys_choreonoid/iob/CMakeLists.txt#L18
lddでRobotHardware_choreonoid.soを見たところ，ここのhrpIo_choreonoidを除くライブラリ群がソースの有無に関わらずdebの方を指していることが問題でした
(特に今回のセグフォはlibhrpsysBaseStub.soが直接の原因みたいです)

良く分からなかったのですが，link_directoriesの優先順位ってどうなっているんでしょうか…？
find_libraryで明示的にパスを指定したところ正しく動作したので，この方法で良ければPR送ることができます(ちなみに公式でもlink_directoriesはめったに使わないと書かれていました)

YoheiKakiuchi, 2017-04-27:
そういうことか。
僕の手元の変更が大きくない感じだと以下の直し方。
https://github.com/start-jsk/rtmros_choreonoid/pull/225
（debが入っている時に動くかは未確認）

石川くんの方法もPRしてもらえるかな。

jsk-ros-pkg/rtm-ros-roboticsのcmakeではlink_directoriesを使っている例の方が多い。
pkg_check_modulesで絶対パスが返って来ていないようで、そうすると、
link_directoriesかfind_libraryを使わざるを得ない気がする。
find_libraryを使っている例はjsk関係だと、これくらいか。
https://github.com/jsk-ros-pkg/jsk_control/blob/master/eus_nlopt/CMakeLists.txt#L12-L36

確認方法はrpathで/opt/ros/indigo/libがhrpsysのディレクトリより先に来ていないこと。
(rm -rf ~/ros/indigo/build/hrpsys_choreonoid をする)
catkin build --this --start-with-this -iv してログを眺める
objdump -x RobotHardware_choreonoid0.so | grep RPATH して出来たバイナリのrpathを確認する

ishiguroJSK, 2017-04-27:
RPATHは意図して指定しない限りバイナリに埋め込まれないという理解なのですが，
CMakeのどのコマンドがRPATHを指定したことになっているのでしょうか?

kyawawa, 2017-04-27:
@YoheiKakiuchi さん
find_libraryを使った方法を #226 に送りました

参考&勉強になります
確信が持てていないのですが，link_directories や include_directoriesの左にあるほう(リストの先頭?)が優先的に読み込まれるのでしょうか
それと，link_directoriesを2回呼んでいますが，この場合上書きと追加どちらなのでしょう？

YoheiKakiuchi, 2017-04-27:
> RPATHは意図して指定しない限りバイナリに埋め込まれないという理解なのですが，
> CMakeのどのコマンドがRPATHを指定したことになっているのでしょうか?

仕様などをあたったわけでないですが、link_directoriesの指定とtarget_link_librariesの絶対パスでのsoの指定はrpthに書かれるように見えますね。

catkinでroscppをrequireするとrpathに/opt/ros/indigo/libが入っていて、それより先に正しいhrpsysのrpathを入れられればいいようです。

kyawawa, 2017-05-29:
ちょくちょくこちらの問題が起こっている人がいるみたいなので， #226 のreviewとマージをお願いできますか．
それと，僕の手元(hrpsysのdeb, src共にあり)では #225 では/opt/ros/indigo/lib がRPATHの先頭に来ているみたいでした

PR #226 (2017-05-30): [RobotHardware_choreonoid] fix rpath of hrp libraries
fix #224 
少し長くなりましたが，find_libraryで明示的にライブラリの場所を指定することでdebとsrcの混同を避けました
closed 2017-05-30 by YoheiKakiuchi