start-jsk/rtmros_choreonoid #250: position limit over がずっと出る

mttamtam, 2017-07-04:
jaxon_redについて、シミュレータを立ち上げた姿勢のままでいると、
```
[el] [20.578000] position limit over CHEST_JOINT1(13), qRef=-0.000765713, llimit =0, ulimit =0.575959, servo_state = ON, prev_angle = 0
```
が延々と出るのが前から気になっていて、ロボットモデルファイルを見てもCHEST_JOINT1の llimitは約-2degになっているはずなので、このリミットがおかしいということになりそうですが、どこを治すといいんでしょうか？
(hrpsysから出てくるログがどんどん流れていってしまうのでできれば阻止したいです）

YoheiKakiuchi, 2017-07-04:
モデルファイルは-2degなのですが、min-max tableによって 0deg 制限になっています。

hrpsys_ros_bridge_tutorials/models/JAXON.conf の joint_limit_table: がそれ。

起こる原因は、simulatorで厳密に関節角度が0でない状態でgoActualされて、それが残っている。
実機でも起こっていいはずだが、ほとんどwith-reset-poseしているのでその時に消えていると思われる。

対処方法としては、
~~~
send *ri* :angle-vector (send *ri* :reset-pose)
~~~
する

or

cnoidファイルの相当する部位のinitial-angleを0.1degぐらいにしておく

くらいだろうか。

mttamtam, 2017-07-05:
なるほど、また別のところで制限がかかっていたのですね。

> 実機でも起こっていいはずだが、ほとんどwith-reset-poseしているのでその時に消えていると思われる。
というのが、前にも別件で実機では動くけどシミュレータではangle-vectorを送ってからじゃないといけない、というのがあったので、
```
send *ri* :angle-vector (send *robot* :reset-pose)
```
的なものを起動時に送るようにしたいですが、どこに書くとよろしいでしょうか？

YoheiKakiuchi, 2017-07-05:
https://github.com/start-jsk/rtmros_choreonoid/blob/master/hrpsys_choreonoid_tutorials/scripts/jaxon_red_setup.py#L43-L54

ここの self.startAutoBalancer()のあとあたりに、resetposeになるように書くのがいいように思います。

mttamtam, 2017-07-05:
書いたらなくなりました。
ありがとうございます。

closed 2017-07-05 by mttamtam