start-jsk/rtmros_choreonoid #264: hrpsys_choreonoid_tutorialsnojaxon_red_choreonoid.launch 起動時にjaxonが崩れる

shintarokkk, 2017-10-25:
leus@leus-ThinkPad-T440p:~$ rtmlaunch hrpsys_choreonoid_tutorials jaxon_red_choreonoid.launch 
[rtmlaunch] Start omniNames at port 15005 

Wed Oct 25 15:02:41 2017:

Starting omniNames for the first time.
Wrote initial log file.
Read log file successfully.
Root context is IOR:010000002b00000049444c3a6f6d672e6f72672f436f734e616d696e672f4e616d696e67436f6e746578744578743a312e300000010000000000000074000000010102000f00000031302e3231332e3139362e31373300009d3a00000b0000004e616d6553657276696365000300000000000000080000000100000000545441010000001c0000000100000001000100010000000100010509010100010000000901010003545441080000000129f05901001431
Checkpointing Phase 1: Prepare.
Checkpointing Phase 2: Commit.
Checkpointing completed.
... logging to /home/leus/.ros/log/1cf0955c-b94a-11e7-947a-e8b1fce1a569/roslaunch-leus-ThinkPad-T440p-5164.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://leus-ThinkPad-T440p:36190/

SUMMARY
========

PARAMETERS
 * /HrpsysSeqStateROSBridge/publish_sensor_transforms: True
 * /controller_configuration: [{'group_name': '...
 * /diagnostic_aggregator/analyzers/computers/contains: ['HD Temp', 'CPU ...
 * /diagnostic_aggregator/analyzers/computers/path: Computers
 * /diagnostic_aggregator/analyzers/computers/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/hrpsys/contains: ['hrpEC', 'Emerge...
 * /diagnostic_aggregator/analyzers/hrpsys/path: Hrpsys
 * /diagnostic_aggregator/analyzers/hrpsys/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/mode/contains: ['Operating Mode'...
 * /diagnostic_aggregator/analyzers/mode/path: Mode
 * /diagnostic_aggregator/analyzers/mode/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/motor/contains: ['Motor']
 * /diagnostic_aggregator/analyzers/motor/path: Motor
 * /diagnostic_aggregator/analyzers/motor/type: diagnostic_aggreg...
 * /diagnostic_aggregator/base_path: 
 * /diagnostic_aggregator/pub_rate: 1.0
 * /ground_truth_bridge/initial_relative: False
 * /ground_truth_bridge/invert_tf: True
 * /ground_truth_bridge/publish_odom: True
 * /ground_truth_bridge/publish_tf: True
 * /ground_truth_bridge/rate: 100.0
 * /ground_truth_bridge/tf_frame: BODY
 * /ground_truth_bridge/tf_parent_frame: choreonoid_origin
 * /ground_truth_bridge/use_ros_name: True
 * /multisense/range_bridge/frame_id: head_hokuyo_frame
 * /multisense/range_bridge/intensity: 1000
 * /multisense_local/jstate_bridge/names: ['motor_joint']
 * /multisense_local/jstate_bridge/rate: 100.0
 * /multisense_local/left/HEADLEFT/camera_param_K: [240, 0, 319.5, 0...
 * /multisense_local/left/HEADLEFT/camera_param_P: [240, 0, 319.5, 0...
 * /multisense_local/left/HEADLEFT/frame_id: left_camera_optic...
 * /multisense_local/pointcloud_bridge/frame_id: left_camera_optic...
 * /multisense_local/pointcloud_bridge/publish_depth: True
 * /multisense_local/pointcloud_bridge/transformed_camera_frame: True
 * /multisense_local/right/HEADRIGHT/camera_param_K: [240, 0, 319.5, 0...
 * /multisense_local/right/HEADRIGHT/camera_param_P: [240, 0, 319.5, -...
 * /multisense_local/right/HEADRIGHT/frame_id: left_camera_optic...
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /use_sim_time: True

NODES
  /multisense_local/left/
    HEADLEFT (hrpsys_ros_bridge/ImageSensorROSBridge)
  /multisense_local/right/
    HEADRIGHT (hrpsys_ros_bridge/ImageSensorROSBridge)
[... 651 lines cut ...]
[hrpsys_py-4] killing on exit
[ INFO] [1508911368.988734808]: [HrpsysSeqStateROSBridge] @onFinalize : HrpsysSeqStateROSBridge0
Traceback (most recent call last):
  File "/home/leus/ros/indigo/src/rtmros_choreonoid/hrpsys_choreonoid_tutorials/scripts/jaxon_red_setup.py", line 63, in <module>
    hcf.init(sys.argv[1], sys.argv[2], connect_constraint_force_logger_ports=connect_constraint_force_logger_ports)
  File "/home/leus/ros/indigo/src/rtmros_choreonoid/hrpsys_choreonoid_tutorials/src/hrpsys_choreonoid_tutorials/choreonoid_hrpsys_config.py", line 41, in init
    URATAHrpsysConfigurator.init(self, robotname, url)
  File "/home/leus/ros/indigo/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/src/hrpsys_ros_bridge_tutorials/urata_hrpsys_config.py", line 21, in init
    HrpsysConfigurator.init(self, robotname, url)
  File "/home/leus/ros/indigo_parent/devel/lib/python2.7/dist-packages/hrpsys/hrpsys_config.py", line 2288, in init
    self.waitForRTCManagerAndRobotHardware(robotname)
  File "/home/leus/ros/indigo_parent/devel/lib/python2.7/dist-packages/hrpsys/hrpsys_config.py", line 1015, in waitForRTCManagerAndRobotHardware
    self.waitForRTCManager(managerhost)
  File "/home/leus/ros/indigo_parent/devel/lib/python2.7/dist-packages/hrpsys/hrpsys_config.py", line 950, in waitForRTCManager
    time.sleep(1)
KeyboardInterrupt
[choreonoid-3] killing on exit
[modelloader-2] killing on exit
terminate called after throwing an instance of 'CORBA::COMM_FAILURE'
^C^C^C^C^C^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
[rtmlaunch] terminate omniNames at port 15005

ishiguroJSK, 2017-10-25:
ロボットが崩れ落ちるのはBodyRTCがうんぬんで，環境構築時などにたまに見る現象です
@YoheiKakiuchi さんが詳しいですが，
https://github.com/start-jsk/rtmros_choreonoid#compile-bodyrtc-etc
の部分を実行したか or OpenRTMはaptかsourceか等を試行すると治るかもしれません．

shintarokkk, 2017-10-26:
@ishiguroJSK さんにさらにオフラインで教えていただき解決しました。ありがとうございます。
hrpsys_choreonoidのビルドがうまくできていなかったのが原因だったようです。
https://github.com/start-jsk/rtmros_choreonoid/のREADMEの
$ export CNOID_INSTALL_DIR=/usr/local/choreonoid
$ export PKG_CONFIG_PATH=${CNOID_INSTALL_DIR}/lib/pkgconfig:$PKG_CONFIG_PATH
$ roscd hrpsys_choreonoid
までやってから(上記のexport~は~/.bashrcに書き込んだ)、一度
$ catkin clean hrpsys_choreonoid
して
$catkin bt
でビルドし直すと正常にchoreonoidを立ち上げることができました。

closed 2017-10-26 by shintarokkk