start-jsk/rtmros_choreonoid #267: hoge.inのようなファイルを自動生成する方法はありますか

mttamtam, 2017-11-10:
hrpsys_choreonoid_tutorials にある.cnoidファイルがすべて.inになっていて、この利点は絶対パスを指定する必要がなくビルドするときにパッケージを探してきて補完してくれるというものだとおもうのですが、これは自動でできるようにしていますか？

cnoidファイルで実験環境を作るときに、外部のVRMLファイルを用いて障害物としていて、これも同じようにしたいです。

ishiguroJSK, 2017-11-10:
.in自体は自分で思うようにデザインして書くものです．

> これは自動でできるようにしていますか？

.cnoid.in -> .cnoidの変換はcmakeによる自動変換ですが，.cnoid.inは誰かが手書き(といってもcnoidの場合はchoreonoidで保存された.cnoidの一部を手で編集)で書いています

> cnoidファイルで実験環境を作るときに、外部のVRMLファイルを用いて障害物としていて、これも同じようにしたいです。

cmakeが取得できるパスであれば(find_packageできるやつとかROSパッケージとか環境変数とか)以下のように変数にsetしてconfigure_fileすることで自動書き換えしてくれています

```
set(JVRC_RTC_DIRECTORY ${hrpsys_choreonoid_SOURCE_PREFIX})
set(JVRC_CONF_DIRECTORY ${PROJECT_SOURCE_DIR}/config)

###                                                                                                                                                         
#JAXON_RED conid                                                                                                                                            
###                                                                                                                                                         
configure_file(${PROJECT_SOURCE_DIR}/config/JAXON_RED_FLAT.cnoid.in ${PROJECT_SOURCE_DIR}/config/JAXON_RED_FLAT.cnoid @ONLY)
configure_file(${PROJECT_SOURCE_DIR}/config/JAXON_RED_VALVE.cnoid.in ${PROJECT_SOURCE_DIR}/config/JAXON_RED_VALVE.cnoid @ONLY)
configure_file(${PROJECT_SOURCE_DIR}/config/JAXON_RED_DRCBOX.cnoid.in ${PROJECT_SOURCE_DIR}/config/JAXON_RED_DRCBOX.cnoid @ONLY)
configure_file(${PROJECT_SOURCE_DIR}/config/JAXON_RED_DOOR.cnoid.in ${PROJECT_SOURCE_DIR}/config/JAXON_RED_DOOR.cnoid @ONLY)
configure_file(${PROJECT_SOURCE_DIR}/config/JAXON_RED_RH_FLAT.cnoid.in ${PROJECT_SOURCE_DIR}/config/JAXON_RED_RH_FLAT.cnoid @ONLY)
configure_file(${PROJECT_SOURCE_DIR}/config/JAXON_RED_LOAD_OBJ.cnoid.in ${PROJECT_SOURCE_DIR}/config/JAXON_RED_LOAD_OBJ.cnoid @ONLY)

configure_file(${PROJECT_SOURCE_DIR}/config/JAXON_RED_TUTORIALS.cnoid.in ${PROJECT_SOURCE_DIR}/config/JAXON_RED_TUTORIALS.cnoid @ONLY)

###                                                                                                                                                         
#CHIDORI conid                                                                                                                                              
###                                                                                                                                                         
if (${jsk_models_FOUND})
configure_file(${PROJECT_SOURCE_DIR}/config/CHIDORI_RH_FLAT.cnoid.in ${PROJECT_SOURCE_DIR}/config/CHIDORI_RH_FLAT.cnoid @ONLY)
endif()

###                                                                                                                                                         
# scene.yaml                                                                                                                                                
###                                                                                                                                                         
configure_file(${PROJECT_SOURCE_DIR}/config/flat.yaml.in ${PROJECT_SOURCE_DIR}/config/flat.yaml @ONLY)
configure_file(${PROJECT_SOURCE_DIR}/config/soccer.yaml.in ${PROJECT_SOURCE_DIR}/config/soccer.yaml @ONLY)
configure_file(${PROJECT_SOURCE_DIR}/config/footsal.yaml.in ${PROJECT_SOURCE_DIR}/config/footsal.yaml @ONLY)
```

YoheiKakiuchi, 2017-11-10:
外部のvrmlとはなにかな？
公開してもんだいないものなら、jvrc_modelsに置きましょう。

ちなみに環境だけなら、yamlファイルを書いて、.launchの ```LOAD_OBJECTS:=true```にするとcnoidを書かずに環境が変更できます。
yamlはこんな感じのものです。
https://github.com/start-jsk/rtmros_choreonoid/blob/master/hrpsys_choreonoid_tutorials/config/footsal.yaml.in

mttamtam, 2017-11-10:
お二人ともありがとうございます。
cnoidファイル-> cnoid.inファイルは手書きで変換してたんですね。
ご指摘のようにCMakeLists.txtの中身などを見て同じ雰囲気で新しくファイルを追加すれば行けるのではないかと思ってました。
（ちなみになぜDRCBOXとVALVEの2つがあるんでしたっけ）

外部のvrmlというのはopenhrp3のサンプルにあるものを使っています。
ものを置いたり沈めるだけなので、LOAD_OBJECTのほうがよさそうです。
footsal.yamlだとサッカーボールも指定していますが、これ以外は静的モデルであるのに対し、ボールだけ動的モデルになっているのはVRMLの中身によるんでしょうか。