start-jsk/rtmros_choreonoid #275: 16.04で動かすためのTIPS

ishiguroJSK, 2018-04-06:
choreonoid本体のcmake時に
`export CNOID_RTM_DIR=${HOME}/ros/indigo_parent/devel`
を
`export CNOID_RTM_DIR=/opt/ros/kinetic`
に読み替えて，
（この設定もしないとrtlsとかでエラーが出る気がする
```
$ sudo vi /etc/omniORB.cfg
giopMaxMsgSize = 209715200     ＃ 200MB に設定
```
）
以下の.rosinstallで最小構成のワークスペースを作って，
```
- git:
    local-name: hrpsys
    uri: https://github.com/fkanehiro/hrpsys-base.git
- git:
    local-name: openhrp3
    uri: https://github.com/fkanehiro/openhrp3.git
- git:
    local-name: rtm-ros-robotics/rtmros_choreonoid
    uri: https://github.com/start-jsk/rtmros_choreonoid.git
- git:
    local-name: rtm-ros-robotics/rtmros_common
    uri: https://github.com/start-jsk/rtmros_common.git
- git:
    local-name: rtm-ros-robotics/rtmros_hrp2
    uri: https://github.com/start-jsk/rtmros_hrp2.git
- git:
    local-name: rtm-ros-robotics/rtmros_tutorials
    uri: https://github.com/start-jsk/rtmros_tutorials.git
- git:
    local-name: trans_system
    uri: https://github.com/jsk-ros-pkg/trans_system.git
- hg:
    local-name: multisense
    uri: https://bitbucket.org/crl/multisense_ros
    version: 3.4.9
```
○○ws/以下で`rosdep install -v -r --from-paths src --ignore-src -y`して

`catkin b -c`でビルドする(gazebo関係がエラー出るけどそれはそれとして，)
最近hrpsys_choreonoidのビルドで`TimedRobotState2`周りでエラーが出る気もするので
(古いaptのhrpsysのidl見に行ってる気がする．ros-kinetic-hrpsys消すとそれはそれでhrpsys/idl/Img.hhがないとかエラーが出る)，
```
sudo mv /opt/ros/kinetic/include/hrpsys/idl/RobotHardwareService.hh /opt/ros/kinetic/include/hrpsys/idl/RobotHardwareService.hh.org
sudo cp ~/○○ws/devel/include/hrpsys/idl/RobotHardwareService.hh /opt/ros/kinetic/include/hrpsys/idl/
```
とかしちゃうと一応ビルドが通って動く．
ちゃんと直したほうがいい気もする．

YoheiKakiuchi, 2018-04-06:
16.04(kinetic)に特有なことは無いと思っているんだけどどうだろうか？

これは、リンクされるrtmのヘッダーがあるところを指定する必要があると思っている。
なので、ソースで入っていると、上記で、aptで入っていると後者。
~~~
export CNOID_RTM_DIR=${HOME}/ros/indigo_parent/devel
を
export CNOID_RTM_DIR=/opt/ros/kinetic
~~~

こちらは、
~~~
$ sudo vi /etc/omniORB.cfg
giopMaxMsgSize = 209715200     ＃ 200MB に設定
~~~

envで設定されていて、必要ないはず。あってもとくに悪さはしないが。
https://github.com/start-jsk/rtmros_choreonoid/blob/master/hrpsys_choreonoid/launch/choreonoid.launch#L21

> 最近hrpsys_choreonoidのビルドでTimedRobotState2周りでエラーが出る気もするので
> (古いaptのhrpsysのidl見に行ってる気がする．ros-kinetic-hrpsys消すとそれはそれでhrpsys/idl/Img.hhがないとかエラーが出る)，

これもよく見る気がするけど、ソースとaptが混ざっているとうまく行かない。
全部ソースならうまく行く気がしている。全部aptの場合はどうだろう？

ishiguroJSK, 2018-04-06:
確かに16.04特有ではないですね．
(久しぶりに一から環境構築してみたというだけでした)
CNOID_RTM_DIRはhrpsysがある方を指せばいいんですね，間違ってました．
`giopMaxMsgSize = 209715200`はやっぱり書かないとrtmlaunchそのままではいろいろ不具合出たのですが，
僕だけですかね・・・VM上だからとかいろいろ地雷踏んでる気もしますが・・・
hrpsysとopenhrp3が両方aptの場合もビルドと実行共に問題なく動いているように見えます．

YoheiKakiuchi, 2018-04-06:
> giopMaxMsgSize = 209715200はやっぱり書かないとrtmlaunchそのままではいろいろ不具合出たのですが，
僕だけですかね・・・VM上だからとかいろいろ地雷踏んでる気もしますが・・・

これは、indigoではなかったのでkinetic特有かも。
maxsizeが小さいと、model-loaderでloadが失敗するので、choreonoidだけでなくて、hrpsys-simulatorもうまく行かないかもしれないのだけど、あっちはテストしてるから問題ないはずだしなあ。
モデルのサイズが大きい（JAXON）と問題あるかもしれない。

ishiguroJSK, 2018-04-06:
やっぱり/etc/omniORB.cfg戻すとrtlsでエラー出たりしますね．
あくまでVM上の16.04での報告ですが・・・
CNOID_RTM_DIRはhrpsysのRobotHardwareのidlに変更があったらそのhrpsys側を指してChoreonoidビルドし直しということになるんでしょうか？
(つまり別バージョンのRobotHardwareを含むhrpsysに対して同じChoreonoidを使うのは良くないということに？)

ishiguroJSK, 2018-04-06:
openrtm関連もhrpsysもCNOID_RTM_DIRも全てソースにして，aptのopenrtm,hrpsysを消したらビルドも実行も出来き，/etc/omniORB.cfgの設定なしでもうまく動きました．
今までhrpsysはソースでopenrtmとCNOID_RTM_DIRはapt側で動かしていましたが，たまたま動いてた感ありますね・・・

ishiguroJSK, 2018-04-06:
いろいろやってて気づいたのですが
```
    local-name: rtm-ros-robotics/rtmros_○○
```
とするか
```
    local-name: rtmros_○○
```
とするかでhrp2wのビルドができるorできないが分岐しますね・・・

ishiguroJSK, 2020-04-09:
18.04に移行しつつあるのでclose

closed 2020-04-09 by ishiguroJSK