start-jsk/rtmros_choreonoid #276: choreonoid1-7でrtmros_choreonoidビルド失敗した

epsilonkei, 2018-04-09:
環境構築した手順
```
mkdir -p catkin_ws/jaxon_tutorial/src
cd  catkin_ws/jaxon_tutorial
wstool init src
curl --user <your-github-username> -L https://raw.githubusercontent.com/jsk-ros-pkg/trans_system/master/jaxon_tutorial.rosinstall | wstool merge -t src -
wstool update -t src
source /opt/ros/indigo/setup.bash
rosdep install -y -r --from-paths src --ignore-src
catkin build
source devel/setup.bash
```
* gazebo関連のpackage以外全部ビルドできました。
* .bashrcで以下に追加して
```
export CNOID_INSTALL_DIR=/usr/local/choreonoid
export CNOID_RTM_DIR=/opt/ros/indigo/
export PKG_CONFIG_PATH=$PKG_CONFIG_PATH:${CNOID_INSTALL_DIR}/lib/pkgconfig
export PATH=$PATH:${CNOID_INSTALL_DIR}/bin
```
* choreonoidをsourceから入れて
```
sudo apt-get install libyaml-dev
export CNOID_INSTALL_DIR=/usr/local/choreonoid
export CNOID_RTM_DIR=/opt/ros/indigo/devel
git clone https://github.com/s-nakaoka/choreonoid.git
mkdir -p choreonoid/build
cd choreonoid/build
## not use python3, and not use pybind11 for backward compatibility
$ cmake .. -DCMAKE_INSTALL_PREFIX=${CNOID_INSTALL_DIR} -DOPENRTM_DIR=${CNOID_RTM_DIR} -DENABLE_INSTALL_RPATH=ON -DENABLE_CORBA=ON -DBUILD_CORBA_PLUGIN=ON -DBUILD_OPENRTM_PLUGIN=ON -DBUILD_HELLO_WORLD_SAMPLE=ON -DBUILD_SPRING_MODEL_SAMPLE=ON -DUSE_PYTHON3=OFF -DUSE_PYBIND11=OFF
$ make -j8
$ sudo make install
```

```
catkin clean hrpsys_choreonoid
catkin b -c
```
すると、以下のエラーで、ビルド止まった

```
Errors     << hrpsys_choreonoid:make /home/kangaroo/catkin_ws/jaxon_tutorial/logs/hrpsys_choreonoid/build.make.024.log
In file included from /usr/local/choreonoid/include/choreonoid-1.7/cnoid/BodyCustomizerInterface:1:0,
                 from /home/kangaroo/catkin_ws/jaxon_tutorial/src/rtmros_choreonoid/hrpsys_choreonoid/src/DOORCustomizer.cpp:9:
/usr/local/choreonoid/include/choreonoid-1.7/cnoid/src/Body/BodyCustomizerInterface.h: In constructor ‘cnoid::BodyCustomizerInterface::BodyCustomizerInterface()’:
/usr/local/choreonoid/include/choreonoid-1.7/cnoid/src/Body/BodyCustomizerInterface.h:67:29: error: ‘nullptr’ was not declared in this scope
         getTargetModelNames(nullptr),
                             ^
make[2]: *** [CMakeFiles/DOORCustomizer.dir/src/DOORCustomizer.cpp.o] エラー 1
make[1]: *** [CMakeFiles/DOORCustomizer.dir/all] エラー 2
make[1]: *** 未完了のジョブを待っています....
In file included from /usr/local/choreonoid/include/choreonoid-1.7/cnoid/BodyCustomizerInterface:1:0,
                 from /home/kangaroo/catkin_ws/jaxon_tutorial/src/rtmros_choreonoid/hrpsys_choreonoid/src/JAXONCustomizer.cpp:9:
/usr/local/choreonoid/include/choreonoid-1.7/cnoid/src/Body/BodyCustomizerInterface.h: In constructor ‘cnoid::BodyCustomizerInterface::BodyCustomizerInterface()’:
/usr/local/choreonoid/include/choreonoid-1.7/cnoid/src/Body/BodyCustomizerInterface.h:67:29: error: ‘nullptr’ was not declared in this scope
         getTargetModelNames(nullptr),
                             ^
make[2]: *** [CMakeFiles/JAXONCustomizer.dir/src/JAXONCustomizer.cpp.o] エラー 1
make[1]: *** [CMakeFiles/JAXONCustomizer.dir/all] エラー 2
make: *** [all] エラー 2
cd /home/kangaroo/catkin_ws/jaxon_tutorial/build/hrpsys_choreonoid; catkin build --get-env hrpsys_choreonoid | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
```

一応、16.04 + kineticと14.04 + indigoで両方、このエラーで止まった。
これを解決するのはどうしたらいいのか教えてもらえませんか

ishiguroJSK, 2018-04-09:
とりあえずhrpsys_choreonoid/CMakeLists.txtの最初の方に
`set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")`
と書くと解決します．
しばらく様子見て大丈夫そうならPR送るといいです．

ishiguroJSK, 2020-04-09:
解決済みだと思うのでcloseお願いします

closed 2020-04-09 by epsilonkei