start-jsk/rtmros_choreonoid #284: rtmlaunch hrpsys_choreonoid_tutorials jaxon_red_choreonoid.launchで出るjaxon_redではRANGE_LINKto

sshige, 2018-06-15:


ishiguroJSK, 2018-06-15:
昔どうだったかは知らないのですが，確かに僕の環境でもRANGE_LINKのTFは繋がっていないようです．
とりあえず`rosrun tf static_transform_publisher 0 0 0 0 0 0 /head_root /RANGE_LINK 100`とかやると，
つながって何かが見えますが，
恐らく/RANGE_LINKは無限回転してるHokuyoのTFに繋がるべきものな気がしていて，
現在Rviz側に回転しているTFが無いので，HokuyoのJoint周りの何かを整備しないといけない気がします．
ChoreonoidでのHokuyoの扱いは少し特殊ですし．

YoheiKakiuchi, 2018-06-17:
choreonoid上ではrangeが回っていて、multisense/lider_scanからデータは出ているという状況でしょうか？

rosのrange関係は以下でlaunchされます。
https://github.com/start-jsk/rtmros_choreonoid/blob/master/hrpsys_choreonoid_tutorials/launch/jaxon_red_vision_connect.launch#L96-L106

1. jaxonの本体からheadqが出ていることを確認
rtls localhost:15005/'JAXON_RED(Robot)0.rtc:headq'

2. JointStateROSBridge0.rtc と接続されていることを確認
rtcat -v localhost:15005/JointStateROSBridge0.rtc

3. /joint_states (もしくは、multisense/joint_statesかも） で motor_joint が含まれているものが出ていることを確認

4. robot_state_publisher に /joint_statesが渡っているか確認

ishiguroJSK, 2018-06-17:
> choreonoid上ではrangeが回っていて、multisense/lider_scanからデータは出ているという状況でしょうか？

そうですね．

hokuyo以外の/joint_stateと，hokuyoの/multisense_local/joint_statesは出ていますが，
```
$ rostopic echo /joint_states 
header: 
  seq: 100790
  stamp: 
    secs: 641
    nsecs: 927000000
  frame_id: ''
name: ['RLEG_JOINT0', 'RLEG_JOINT1', 'RLEG_JOINT2', 'RLEG_JOINT3', 'RLEG_JOINT4', 'RLEG_JOINT5', 'LLEG_JOINT0', 'LLEG_JOINT1', 'LLEG_JOINT2', 'LLEG_JOINT3', 'LLEG_JOINT4', 'LLEG_JOINT5', 'CHEST_JOINT0', 'CHEST_JOINT1', 'CHEST_JOINT2', 'HEAD_JOINT0', 'HEAD_JOINT1', 'RARM_JOINT0', 'RARM_JOINT1', 'RARM_JOINT2', 'RARM_JOINT3', 'RARM_JOINT4', 'RARM_JOINT5', 'RARM_JOINT6', 'RARM_JOINT7', 'LARM_JOINT0', 'LARM_JOINT1', 'LARM_JOINT2', 'LARM_JOINT3', 'LARM_JOINT4', 'LARM_JOINT5', 'LARM_JOINT6', 'LARM_JOINT7', 'LARM_F_JOINT0', 'LARM_F_JOINT1', 'RARM_F_JOINT0', 'RARM_F_JOINT1']
position: [0.00013606518280792677, 0.0004595302725378445, -0.23965881416120877, 0.6779311764428471, -0.4310003217495804, -0.00048017347060990903, -1.9679615365269516e-05, 0.0003044225797279202, -0.23963812364420772, 0.6776844469067491, -0.43189738217748525, 9.299052435370321e-05, -6.384228323914684e-06, -0.0008276135789665081, 1.967418868533579e-08, 1.2078328922693002e-07, 0.00020756317094223817, -8.627482053286123e-06, 0.6979800370825542, -0.3486553151886269, -0.08700599034817857, -1.3957354908607327, 1.5820065225021382e-05, 6.480226182616179e-05, -0.348943811163171, 8.633409013913311e-06, 0.6979800381976624, 0.34865529119754896, 0.08700597866559066, -1.3957354979892809, -1.582119980908442e-05, -6.480449955100325e-05, -0.3489438126009499, 0.00012131898498891149, 0.0002713981163537486, 0.00011941849203322088, 0.00026530537359883917]
velocity: [8.356536118334237e-06, 2.2357788587549793e-05, -2.2454518939027572e-05, 3.844034524654821e-05, -5.312919552945719e-05, -5.6625969652935684e-05, -7.009866671951767e-06, 1.555881204447235e-05, 1.4907123990841138e-05, -4.480879630719785e-05, 3.359440881147976e-05, -2.3125017965456305e-05, 1.6124292029258133e-05, -2.555496492015017e-06, 7.221143599634566e-06, 2.2925313274936666e-07, -7.748617504447665e-08, 1.360752437030126e-07, 3.8514674122993457e-07, -2.7514098956040756e-06, -2.0661539642664915e-06, -9.218113166536252e-07, -1.216255892928462e-07, -5.393401512728484e-07, -1.9473764466883494e-07, -1.8364578826948134e-07, -8.303983425399041e-07, -2.4465018242924835e-06, -1.8208296530837109e-06, 2.9993083749565735e-07, -1.0014121169822826e-07, -4.663870633956942e-07, 6.526658700609264e-08, -5.121680673295611e-08, -8.395447130586519e-08, 7.54706441817591e-09, 9.311100405625922e-09]
effort: [-1.3602692285514872, -2.798222296924626, -46.18185945491065, -97.73324656590188, -0.25080457784598664, -0.007157099044113283, 1.2000483681183491, 2.378446823491195, -44.53349787274907, -96.63676053140158, 0.12258815922322475, -0.10661798107854856, 0.5268750267246677, 68.6932388740404, -0.0034204502022236903, -0.001239833131387863, -2.075627118248862, 0.1726595013198746, 3.0389153136622937, -8.212726711482876, -5.1994597450352735, -10.48994531612113, -0.3163529869642137, -1.295855599298603, -2.4437190773629602, -0.1725448707378951, 3.0395767514357352, 8.215079454968466, 5.2011109760807805, -10.490166448478, 0.31646957533035697, 1.2962674275598678, -2.443778795627738, -0.026290574171765342, -0.052563051919998584, -0.026287917105328365, -0.052558250489610395]
---
```

```
$ rostopic echo /multisense_local/joint_states 
header: 
  seq: 39507
  stamp: 
    secs: 435
    nsecs: 785000000
  frame_id: ''
name: ['motor_joint']
position: [435.77474760002315]
velocity: [0.9999999991331757]
effort: []
---
```
robot_state_publisherには/joint_stateしか渡っていないように思います．
そもそもrobor_state_publisherは別々のjoint_stateを入力するような使いかたできるものなのでしょうか？

```
$ rosnode info /hrpsys_state_publisher
--------------------------------------------------------------------------------
Node [/hrpsys_state_publisher]
Publications: 
 * /tf [tf2_msgs/TFMessage]
 * /tf_static [tf2_msgs/TFMessage]
 * /rosout [rosgraph_msgs/Log]

Subscriptions: 
 * /joint_states [sensor_msgs/JointState]
 * /clock [rosgraph_msgs/Clock]

Services: 
 * /hrpsys_state_publisher/get_loggers
 * /hrpsys_state_publisher/set_logger_level

contacting node http://192.168.96.127:38127/ ...
Pid: 113363
Connections:
 * topic: /rosout
    * to: /rosout
    * direction: outbound
    * transport: TCPROS
 * topic: /tf
    * to: /rviz_1529212386927537577
    * direction: outbound
    * transport: TCPROS
 * topic: /tf
    * to: /tf2_buffer_server
    * direction: outbound
    * transport: TCPROS
 * topic: /tf
    * to: /footcoords
    * direction: outbound
    * transport: TCPROS
 * topic: /tf_static
    * to: /rviz_1529212386927537577
    * direction: outbound
    * transport: TCPROS
 * topic: /tf_static
    * to: /tf2_buffer_server
    * direction: outbound
    * transport: TCPROS
 * topic: /tf_static
    * to: /footcoords
    * direction: outbound
    * transport: TCPROS
 * topic: /clock
    * to: /HrpsysSeqStateROSBridge (http://192.168.96.127:33356/)
    * direction: inbound
    * transport: TCPROS
 * topic: /joint_states
    * to: /HrpsysSeqStateROSBridge (http://192.168.96.127:33356/)
    * direction: inbound
    * transport: TCPROS
```

YoheiKakiuchi, 2018-06-17:
そこまで出てたら choreonoidの問題ではないね。
実機と上げているlaunchにdiffがあるんだろうか。

最終的にtfを出す robot_state_publisher にマージしたjoint_statesを渡しているようなことをしているはずです。

YoheiKakiuchi, 2018-06-17:
これを一緒にあげてみて
roslaunch hrpsys_choreonoid_tutorials jaxon_multisense_local.launch

sshige, 2018-06-18:
ishiguroJSKさんに見てもらい、解決しました。

自分の問題は、以前にmultisenseのlidarの回転を止めるようにjaxon_jvrcのモデルファイルを書き換えた時のファイルが残っていたことでした。choreonoidの方ではlidarが回っていたのでjaxon_jvrcのbuildしなおしで直っていると思っていたのですが、それでは十分ではなく、git clean -xfdなどでしっかり自動生成ファイルも消して、buildしなおさなくてはならないというところが問題だったようです。

ご対応いただき、ありがとうございます。

closed 2018-06-18 by sshige

k-okada, 2018-06-19:
> aptのjsk_robot, jsk_tilit_laser, jsk_toolsにlaunchディレクトリがなく，
> そのままではそれらに依存しているjaxon_multisense_local.launchが動きませんでしたが，
> それらをソースで持ってこれば，回転TFもlaser_scanも表示できました．
> ありがとうございます．

> aptのjsk_robot, jsk_tilit_laser, jsk_toolsにlaunchディレクトリがなく，

具体的にどのディレクトリがないか教えてください．
プラス，可能であれば修正のPRを作ってください．

ishiguroJSK, 2018-06-19:
```
 ~/catkin_ws/***/src/jsk_common/jsk_tilt_laser $ ls
CHANGELOG.rst  CMakeLists.txt  README.md  cfg  config  launch  package.xml  scripts  src
```
```
/opt/ros/indigo/share/jsk_tilt_laser $ ls
cmake  package.xml
```

このPRで正しいとすると，同様の修正をすべきパッケージが他にも多く見受けられるようなのですが，
本当にこれでいいのでしょうか・・・
https://github.com/jsk-ros-pkg/jsk_robot/pull/942
https://github.com/jsk-ros-pkg/jsk_common/pull/1583