start-jsk/rtmros_choreonoid #289: JAXON_JVRCで物体を把持する際に全身が振動する

shintarokkk, 2018-09-14:
一応自己解決した問題ですが、解決したというより回避策を出しただけなのでissueに上げておきます。

現象：choreonoid上のJAXON_JVRC（モデルファイルは[こちら](https://github.com/start-jsk/rtmros_choreonoid/blob/master/jvrc_models/JAXON_JVRC/JAXON_JVRCmain_hrpsys_bush.wrl)、ゲインは[こちら](https://github.com/start-jsk/rtmros_choreonoid/blob/master/hrpsys_choreonoid_tutorials/models/JAXON_JVRC.PDgains_sim.dat)）のハンドで適当な物体を把持させると全身が振動する。（物体をゆるく握らせた状態から(send \*ri\* :start-grasp)して握りこむと振動し始める）
[image]

解決法；しっかり握りこまずに、ゆるい把持のまま動作させた

おそらくですが、手先リンク、指リンク0と指リンク1で物体を挟み込んで押すような動作が振動を起こしているようです。リンク同士で物体を押し付け合うような状態を作らなければ一応振動は回避できます。

ただ、握りこまなくても以下の画像のように手の中と指先で物体を押し付け合うような状況で振動が起こります。（この画像の状態から右手を下に下げていくと振動し始めます。）
[image]
これを回避するために、以下の画像のように把持部分の形状を単純化しました。
[image]

同様の問題が発生した際には参考にしてください。また、今回はしっかりと把持しなくても良いような動作だったのでこの方法で大丈夫でしたが、もっと良い解決法を知っている方は教えていただけると幸いです。

kindsenior, 2018-09-14:
僕は残念ながら同じ現象になったことはないんですけど，
どういう動作・状況か(st,autobalancerは入っているか)等がわかるように確実に状況を再現できるeusのスクリプトがあるといいと思います
あと，hrpsysのログファイルもあるといいと思います

shintarokkk, 2018-09-14:
きれいな環境で再現しようとしてみたところ、普通に把持できてしまいました...
把持の際の振動が起きたのはhrpsysやros_bridge等少しいじってあった環境だったのですが、その変更が振動の原因だとは思っていませんでした。どの部分が原因だったのか探してみたいと思います。

closed 2018-09-14 by shintarokkk