start-jsk/rtmros_choreonoid #291: [hrpsys_choreonoid]tilt_laser_jointの可動範囲をrosから変更できるようにしたい

yuki-shark, 2018-10-23:
[hrpsys_choreonoid/src/JAXONCustomizer.cpp]で、tilt_laser_jointの可動範囲が、

``` cpp
#define TILT_UPPER_BOUND  0.7
#define TILT_POSITIVE_SPEED 1.0
#define TILT_LOWER_BOUND -0.7
#define TILT_NEGATIVE_SPEED -1.0
```

のように直接定義されているのですが、これをlaunchなど外側からいじれるようにするにはどのようにするのが良いでしょうか。
[hrpsys_choreonoid_tutorials/launch/chidori_choreonoid.launch]などから、変更できるようにしたいと考えています。

ishiguroJSK, 2018-10-23:
Customizerは完全にChoreonoid側の機能なので，rosbridge経由で変更するのは結構大変な気がする．
将来的にeusとかrosのコマンドでどうしてもインタラクティブに操作したいということなら，ros_bridgeにポートとかトピックとかサービスとか追加して，hrpsys_choreonoidのRobotHardwareにポートかサービス追加して，Customizerに届くようになんかしなきゃいけなくなるけど，
起動毎に簡単に切り替えたいというだけなら適当にConfigファイルとか読むようにするだけでダメだろうか？

yuki-shark, 2018-10-24:
コメントありがとうございます．結構大変なんですね．
現状起動時に変更ができればいいので，適当なconfigファイルを読むようにしてみます.