start-jsk/rtmros_choreonoid #305: 爽やかな床のテクスチャ重くてGLVisionSimlator有効の時に遅くなる

ishiguroJSK, 2019-03-19:
一部のロボットのlaunchが2倍近く遅いなと思っていたら，違いは床のようです．
まずGLVisionSimlatorの有無で変化が認められて，次に床をすり替えるだけで変わったので．

floor_surface.pngを2676x2676から512x512にしたらおよそ実時間に戻ります．
とくにこだわりが無ければ縮小したものをPR送るべきなんでしょうか？

YoheiKakiuchi, 2019-03-19:
テクスチャは2のべき乗が良いとのことだが、2676x2676 を 2048x2048にするとどうなるかな？

ishiguroJSK, 2019-03-19:
それだけでもかなり効果ありますね．すごい．

デフォルト
```
rostopic hz /joint_states -w 200
average rate: 198.801
	min: 0.000s max: 0.044s std dev: 0.01088s window: 200
```
`cp floor_surface.png floor_surface.png.bak`
↓
```
convert -geometry 2048x floor_surface.png.bak floor_surface.png
rostopic hz /joint_states -w 200
average rate: 115.029
	min: 0.000s max: 0.055s std dev: 0.01527s window: 200
```

```
convert -geometry 512x floor_surface.png.bak floor_surface.png
rostopic hz /joint_states -w 200
average rate: 101.582
	min: 0.000s max: 0.052s std dev: 0.01737s window: 200
```

```
convert -geometry 16x floor_surface.png.bak floor_surface.png
rostopic hz /joint_states -w 200
average rate: 102.736
	min: 0.000s max: 0.057s std dev: 0.01740s window: 200
```

ishiguroJSK, 2019-03-20:
こちらどうしましょうか？
- 問題のlaunchやyamlからは真っ青な床を呼ぶように変更
- 512x512のpngと床を追加してそちらを呼ぶよう変更
ができますが

YoheiKakiuchi, 2019-03-20:
> 問題のlaunchやyamlからは真っ青な床を呼ぶように変更

こちらでお願いします。

closed 2019-03-20 by ishiguroJSK