start-jsk/rtmros_choreonoid #312: choreonoid上でJAXONがシャベルを把持できない

takayuki5168, 2019-05-20:
5/20(月)の研究会でM1室岡が報告したバグを共有させていただきます．

choreonoid上でJAXONにシャベルを持たせたいのですが，シャベルが手からすり抜けてしまいます．

https://github.com/takayuki5168/rtmros_choreonoid/tree/t-murooka-shovel をbuildすると `JAXON_RED_DIG.cnoid` が生成されるので，
```sh
rtmlaunch hrpsys_choreonoid_tutorials jaxon_red_choreonoid.launch TASK:=DIG
```
で実行できます．

ishiguroJSK, 2019-05-20:
ちょっと試した感じ，ショベルのモデルのマスパラがおかしかったので，変えると治ります．
元のショベルモデルのままだと，床に置くだけでも暴れていました．
```
diff --git a/shovel_models/shovel.body b/shovel_models/shovel.body
index e9db894..577b6fa 100644
--- a/shovel_models/shovel.body
+++ b/shovel_models/shovel.body
@@ -8,12 +8,12 @@ links:
     translation: [0, 0, 0]
     jointType: free
     material: { viscosity: 0.01 }
-    centorOfMass: [0, 0.4, 0]
+    centerOfMass: [0.15, -1.5, 0]
     mass: 2.0
     inertia: [
-      0.0000283075, 0, 0,
-      0, 0.0000283075, 0,
-      0, 0, 0.0000283075 ]
+      0.283075, 0, 0,
+      0, 0.283075, 0,
+      0, 0, 0.283075 ]
     elements:
       Transform:
         translation: [0, 0, 0]
```
あとJAXON_RED_DIG.cnoid.inに絶対パスが掛かれているので，相対パスかcmakeのconfigure_fileで置換されるようにしておくとよいです

takayuki5168, 2019-05-20:
ありがとうございます！

土掘りの動作までできるか検証を続けてみます．

cc @YoheiKakiuchi

closed 2019-05-20 by takayuki5168