start-jsk/rtmros_choreonoid #321: シミュレーション中にオブジェクトを追加したい

ishiguroJSK, 2020-03-10:
ros_service_server.pyが含まれているcnoidプロジェクトで

```
>>> addObject("aaa", "/usr/local/choreonoid/share/choreonoid-1.6/model/misc/box1.body")
'(:success)'
```
としても非干渉なモデルが描画されるだけになる
今回は踏み台のようなstaticなモデルですが，ボールのようなオブジェクトだとアイテムツリー内でボール以下にAISTSimulatorがぶら下がる気がしますが，だれかこれでうまく行った実績ありますか？

[image]

YoheiKakiuchi, 2020-03-10:
前にできると言ったんですが、ダイナミックにできていなかった気がしてきました。
一回シミュレーションを止めて、再度実行すると認識される気がします。
(スタート・ポーズはjskのhrpsysを使うパターンだと上手く行っていないきがする。)

動的物体になるか、静的物体になるかは、モデルファイルの書き方にも依存しているかもしれない。
wrlだと、worldとのfreeジョイントがないと動的物体にならないように動いていたように思う。
.bodyはどうだか分からない。

ishiguroJSK, 2020-03-10:
なるほど，ありがとうございます．
階段の高さのパラチュンが出来ればいいので，yamlに書いて起動時に読ませる方針で行こうと思います．

closed 2020-03-10 by ishiguroJSK