start-jsk/rtmros_choreonoid #323: 18.04

ishiguroJSK, 2020-03-19:
試してみたら半分くらいしか機能していない．環境は，
```
- git:
    local-name: hrpsys
    uri: https://github.com/fkanehiro/hrpsys-base.git
- git:
    local-name: openhrp3
    uri: https://github.com/fkanehiro/openhrp3.git
- git:
    local-name: rtmros_choreonoid
    uri: https://github.com/start-jsk/rtmros_choreonoid.git
- git:
    local-name: rtmros_common
    uri: https://github.com/start-jsk/rtmros_common.git
- git:
    local-name: rtmros_hrp2
    uri: git@github.com:start-jsk/rtmros_hrp2.git
- git:
    local-name: rtmros_tutorials
    uri: https://github.com/start-jsk/rtmros_tutorials.git
- git:
    local-name: trans_system
    uri: git@github.com:jsk-ros-pkg/trans_system.git
- hg:
    local-name: multisense_ros
    uri: https://bitbucket.org/crl/multisense_ros
    version: 3.4.9
```
choreonoidはrelease-1.7で./misc/script/install-requisites-ubuntu-18.04.sh を走らせて，.bashrcは以下
```
export CNOID_INSTALL_DIR=/usr/local/choreonoid
export CNOID_RTM_DIR=/opt/ros/${ROS_DISTRO}
export PKG_CONFIG_PATH=${CNOID_INSTALL_DIR}/lib/pkgconfig:$PKG_CONFIG_PATH
export PATH=${CNOID_INSTALL_DIR}/bin:$PATH
```
で
`cmake .. -DCMAKE_INSTALL_PREFIX=${CNOID_INSTALL_DIR} -DOPENRTM_DIR=${CNOID_RTM_DIR} -DENABLE_INSTALL_RPATH=ON -DENABLE_CORBA=ON -DBUILD_CORBA_PLUGIN=ON -DBUILD_OPENRTM_PLUGIN=ON -DBUILD_HELLO_WORLD_SAMPLE=ON -DBUILD_SPRING_MODEL_SAMPLE=ON -DUSE_PYTHON3=OFF -DUSE_PYBIND11=OFF -DUSE_BUILTIN_CAMERA_IMAGE_IDL=ON`
でmake

jsk-footstep-controllerが18.04だとリリースされていないのか，launch内コメントアウトしつつ
 `reset; pkill -9 choreonoid; rtmlaunch hrpsys_choreonoid_tutorials jaxon_red_choreonoid.launch USE_VISION:=False`とかで起動するとRTMのポート繋げません的なエラーが出ている(全ログ後述)

> [rtmlaunch] [ERROR] Could not Connect ( /localhost:15005/JAXON_RED(Robot)0.rtc:q , /localhost:15005/HrpsysSeqStateROSBridge0.rtc:rsangle ):  CORBA.MARSHAL(omniORB.MARSHAL_InvalidEnumValue, CORBA.COMPLETED_YES) 
> [rtmlaunch] Could not Activate ( localhost:15005/HrpsysSeqStateROSBridge0.rtc ) :  CORBA.MARSHAL(omniORB.MARSHAL_InvalidEnumValue, CORBA.COMPLETED_YES) 
> The model was successfully loaded !

ROSBridgeが落ちているが，hcfからgoPosとかいじれる．STがバグっている気もするが．

18.04ではRTMは1.1.2 (16.04までは1.1.0)
```
 $ dpkg -l *rtm*
要望=(U)不明/(I)インストール/(R)削除/(P)完全削除/(H)保持
| 状態=(N)無/(I)インストール済/(C)設定/(U)展開/(F)設定失敗/(H)半インストール/(W)トリガ待ち/(T)トリガ保留
|/ エラー?=(空欄)無/(R)要再インストール (状態,エラーの大文字=異常)
||/ 名前                                        バージョン                 アーキテクチャ             説明
+++-===========================================-==========================-==========================-============================================================================================
un  librtmp-dev                                 <なし>                     <なし>                     (説明 (description) がありません)
ii  librtmp1:amd64                              2.4+20151223.gitfa8646d.1- amd64                      toolkit for RTMP streams (shared library)
ii  ros-melodic-openrtm-aist                    1.1.2-3bionic.20200303.062 amd64                      This package represents OpenRTM-aist that's built within ROS eco system.
ii  ros-melodic-openrtm-aist-python             1.1.0-0bionic.20190531.231 amd64                      Python binding of OpenRTM-AIST (see openrtm_aist for further information).
```

USE_ROBOTHARDWAREな`rtmlaunch hrpsys_choreonoid_tutorials chidori_choreonoid.launch`だと以下．表記は違うが似たようなものな気がする．
```
[rtmlaunch] Could not start rtmlaunch.py, Caught exception ( CORBA.MARSHAL(omniORB.MARSHAL_InvalidEnumValue, CORBA.COMPLETED_YES) )
[rtmlaunch] .. Could not connect to NameServer at  localhost:15005 , Caught exception ( CORBA.MARSHAL(omniORB.MARSHAL_InvalidEnumValue, CORBA.COMPLETED_YES) )
[rtmlaunch] .. Please make sure that you have NameServer running at localhost:15005/`
[rtmlaunch] .. You can check with `rtls localhost:15005/`
[rtmlaunch_hrpsys_ros_bridge-12] process has finished cleanly
log file: /home/leus/.ros/log/92f211f0-69cd-11ea-b8b5-7c7a914eb678/rtmlaunch_hrpsys_ros_bridge-12*.log
[... 1879 lines cut ...]
  step_height = 0.065[m], step_time = 1.2[s], toe_angle = 0[deg], heel_angle = 0[deg]
[9] footstep
  name = [rleg]
  pos = [ 0.942024,  -0.102352,  0.0459988]
  rot = [           1,  -7.9088e-06,  0.000299106] [ 7.90296e-06,            1,  1.95067e-05] [-0.000299106, -1.95044e-05,            1]
  step_height = 0.065[m], step_time = 1.2[s], toe_angle = 0[deg], heel_angle = 0[deg]
[10] footstep
  name = [lleg]
  pos = [ 0.942023,  0.0976483,  0.0459949]
  rot = [           1,  -7.9088e-06,  0.000299106] [ 7.90296e-06,            1,  1.95067e-05] [-0.000299106, -1.95044e-05,            1]
  step_height = 0[m], step_time = 1.2[s], toe_angle = 0[deg], heel_angle = 0[deg]
[el] [22.652000] velocity limit over LARM_JOINT4(29), qvel=-4, lvlimit =-3.99982, uvlimit =3.99982, servo_state = ON, q(limited) = -1.39797
[el] [22.708000] velocity limit over LARM_JOINT6(31), qvel=4, lvlimit =-3.99982, uvlimit =3.99982, servo_state = ON, q(limited) = 0.38095
[abc] Too large IK error in larm (vel_p) = [-0.00027651 -0.000444587 -5.12794e-05][m], count = 0
[abc] Too large IK error in larm (vel_r) = [3.3375e-05 -1.82938e-05 0.000186805][rad], count = 0
[co] 44/154 pair: LLEG_JOINT2/LARM_JOINT7(14), distance = -0.000996439
[co] 38/154 pair: LARM_JOINT6/WAIST(10), distance = -0.000989708
[co] 44/154 pair: LLEG_JOINT2/LARM_JOINT7(14), distance = -0.00323839
[el] [22.840000] error limit over LARM_JOINT2(27), qRef=0.430463, qCurrent=0.247272 , Error=0.183191 > 0.18 (limit), servo_state = ON, q(limited) = 0.427272
[el] [22.846000] velocity limit over LARM_JOINT2(27), qvel=5.31557, lvlimit =-3.99982, uvlimit =3.99982, servo_state = ON, q(limited) = 0.427831

[abc] Too large IK error in larm (vel_p) = [-0.00164102 -0.00404957 -0.00245678][m], count = 100
[abc] Too large IK error in larm (vel_r) = [0.00293359 -0.000323249 -0.000189957][rad], count = 100

```

ishiguroJSK, 2020-04-08:
解決した気がします.
https://discourse.choreonoid.org/t/rtshell-marshal-invalidenumvalue/97/13
https://choreonoid.org/ja/manuals/latest/openrtm/install.html#id6
OpenRTM1.1.2とChoreonoidの既知のバグで，ヘッダをすげ替えて対処できるとのこと．
すげ替えて再ビルドしてgoPosだのRvizだのざっと確認できました．

あとでREADME更新しておきます

YoheiKakiuchi, 2020-04-08:
WRSの側ではshared memory経由でchoreonoidとつなぐ感じにしていて、そっちだとchoreonoid側のRTMを気にしなくても良いのと、多分少し速いはず。
あと、最新のchoreonoidはopenrtmを分離している。
https://discourse.choreonoid.org/t/openrtm/363

ishiguroJSK, 2020-04-08:
最近は，実機もgazeboもchoreonoidも全部SHMでアクセスすれば，依存も保守も楽になっていいんじゃないかと思い始めました．そもそも実機がSHMですし．

pazeshun, 2020-04-12:
@ishiguroJSK 
> 解決した気がします.
> https://discourse.choreonoid.org/t/rtshell-marshal-invalidenumvalue/97/13
> https://choreonoid.org/ja/manuals/latest/openrtm/install.html#id6
> OpenRTM1.1.2とChoreonoidの既知のバグで，ヘッダをすげ替えて対処できるとのこと．

これは、OpenRTM1.2では治っているのでしょうか。
もしOpenRTM1.2をテストして動きそうなら、そちらをreleaseしてもらうよう、
https://github.com/tork-a/openrtm_aist-release
でお願いするべきかと思います。

ishiguroJSK, 2020-04-13:
> もしOpenRTM1.2をテストして動きそうなら

これをやろうとして立ち止まってしまったんだけど，思ったより難しい気がする

AIST側のレポジトリにしかないv1.2.0をtork-a側のレポジトリにgit pullしないとROSパッケージとしてOpenRTMv1.2.0を使えないと思いそれっぽくgit pullしてみたけどなんか違う気がする，
```
$ git pull aist tags/v1.2.0 --allow-unrelated-histories
From https://github.com/OpenRTM/OpenRTM-aist
 * tag                 v1.2.0     -> FETCH_HEAD
Auto-merging win32/OpenRTM-aist/utils/rtc-template/Makefile.am
CONFLICT (add/add): Merge conflict in win32/OpenRTM-aist/utils/rtc-template/Makefile.am
Auto-merging win32/OpenRTM-aist/utils/Makefile.am
CONFLICT (add/add): Merge conflict in win32/OpenRTM-aist/utils/Makefile.am
Auto-merging win32/OpenRTM-aist/rtm_config.vsprops
CONFLICT (add/add): Merge conflict in win32/OpenRTM-aist/rtm_config.vsprops
Auto-merging win32/OpenRTM-aist/rtm_config.props
CONFLICT (add/add): Merge conflict in win32/OpenRTM-aist/rtm_config.props
（略）
```

ros-melodic-openrtm_aistじゃなくて，openrtm_aistのdebパッケージ拾ってきてdpkg -iしたけども，
(今aptレポジトリから落とせないらしいhttp://www1.meijo-u.ac.jp/~kohara/cms/openrtm-aist-1-2-0_install_temp)

https://www.openrtm.org/redmine/issues/3688のdataport.timestamp_policy とか
1.2.0で追加された/usr/include/openrtm-1.2/rtm/Timestamp.hのせいか，
```
    virtual ReturnCode operator()(ConnectorInfo& info, DataType& data)
    {
      if (info.properties["timestamp_policy"] != m_tstype)
        {
          return NO_CHANGE;
        }
      coil::TimeValue tm(coil::gettimeofday());
      data.tm.sec  = tm.sec();
      data.tm.nsec = tm.usec() * 1000;
      return DATA_CHANGED;
    }
```
色んな所でタイムスタンプに関するエラーが出る．
```
make[2]: *** [rtc/VirtualCamera/CMakeFiles/VirtualCameraComp.dir/VirtualCamera.cpp.o] Error 1
make[2]: *** 未完了のジョブを待っています....
In file included from /usr/include/openrtm-1.2/rtm/InPort.h:39:0,
                 from /usr/include/openrtm-1.2/rtm/DataInPort.h:23,
                 from /home/leus/catkin_ws/master/src/hrpsys/rtc/Simulator/Simulator.h:18,
                 from /home/leus/catkin_ws/master/src/hrpsys/rtc/Simulator/Simulator.cpp:12:
/usr/include/openrtm-1.2/rtm/Timestamp.h: In instantiation of ‘RTC::Timestamp<DataType>::ReturnCode RTC::Timestamp<DataType>::operator()(RTC::ConnectorInfo&, DataType&) [with DataType = OpenHRP::SceneState; RTC::Timestamp<DataType>::ReturnCode = RTC::ConnectorListenerStatus::Enum]’:
/home/leus/catkin_ws/master/src/hrpsys/rtc/Simulator/Simulator.cpp:347:2:   required from here
/usr/include/openrtm-1.2/rtm/Timestamp.h:40:12: error: ‘struct OpenHRP::SceneState’ has no member named ‘tm’; did you mean ‘time’?
       data.tm.sec  = tm.sec();
       ~~~~~^~
       time
/usr/include/openrtm-1.2/rtm/Timestamp.h:41:12: error: ‘struct OpenHRP::SceneState’ has no member named ‘tm’; did you mean ‘time’?
       data.tm.nsec = tm.usec() * 1000;
```

関連するRTCとかをビルド対象から外せばhrpsys自体はビルド完走するけど，
hrpsys_ros_bridgeでなんか出る
```
Errors     << hrpsys_ros_bridge:make /home/leus/catkin_ws/master/logs/hrpsys_ros_bridge/build.make.000.log                                                                                            
/usr/bin/ld: /tmp/ccctgGQg.o: relocation R_X86_64_PC32 against symbol `_ZN7OpenHRP25VirtualForceSensorService4_nilEv' can not be used when making a shared object。 -fPIC を付けて再コンパイルしてください。
/usr/bin/ld: 最終リンクに失敗しました: 不正な値です
collect2: error: ld returned 1 exit status
make[2]: *** [/home/leus/catkin_ws/master/devel/.private/hrpsys_ros_bridge/lib/libVirtualForceSensorServiceStub.so] Error 1
make[1]: *** [CMakeFiles/RTMBUILD_hrpsys_ros_bridge_VirtualForceSensorService_genrpc.dir/all] Error 2
/usr/bin/ld: /tmp/ccu6mJLn.o: relocation make[1]: *** 未完了のジョブを待っています....
R_X86_64_PC32 against symbol `_ZN7OpenHRP19TorqueFilterService4_nilEv' can not be used when making a shared object。 -fPIC を付けて再コンパイルしてください。
/usr/bin/ld: 最終リンクに失敗しました: 不正な値です
collect2: error: ld returned 1 exit status
make[2]: *** [/home/leus/catkin_ws/master/devel/.private/hrpsys_ros_bridge/lib/libTorqueFilterServiceStub.so] Error 1
make[1]: *** [CMakeFiles/RTMBUILD_hrpsys_ros_bridge_TorqueFilterService_genrpc.dir/all] Error 2
/usr/bin/ld: /tmp/ccgATVmg.o: relocation R_X86_64_PC32 against symbol `_ZN7OpenHRP16WavPlayerService4_nilEv' can not be used when making a shared object。 -fPIC を付けて再コンパイルしてください。
/usr/bin/ld: 最終リンクに失敗しました: 不正な値です
collect2: error: ld returned 1 exit status
make[2]: *** [/home/leus/catkin_ws/master/devel/.private/hrpsys_ros_bridge/lib/libWavPlayerServiceStub.so] Error 1
make[1]: *** [CMakeFiles/RTMBUILD_hrpsys_ros_bridge_WavPlayerService_genrpc.dir/all] Error 2
/usr/bin/ld: /tmp/ccFgFQFg.o: relocation R_X86_64_PC32 against symbol `_ZrSjR9cdrStream' can not be used when making a shared object。 -fPIC を付けて再コンパイルしてください。
/usr/bin/ld: 最終リンクに失敗しました: 不正な値です
collect2: error: ld returned 1 exit status
make[2]: *** [/home/leus/catkin_ws/master/devel/.private/hrpsys_ros_bridge/lib/libpointcloudStub.so] Error 1
make[1]: *** [CMakeFiles/RTMBUILD_hrpsys_ros_bridge_pointcloud_genrpc.dir/all] Error 2
make: *** [all] Error 2
```
わからない

そもそもros-melodic-openrtm-aistをv1.2.0に上げて手元で試す正しい方法ってあるんでしょうか？
tork-aではそれに等しいことをしているはず？

pazeshun, 2020-04-14:
1.1.2と1.2.1の差分を明示的に指定してpatchを作り、release/melodic/openrtm_aistに当ててみても、同じようなところでこけますね。
```
$ git clone https://github.com/OpenRTM/OpenRTM-aist.git
Cloning into 'OpenRTM-aist'...
remote: Enumerating objects: 1, done.
remote: Counting objects: 100% (1/1), done.
remote: Total 47491 (delta 0), reused 0 (delta 0), pack-reused 47490
Receiving objects: 100% (47491/47491), 11.89 MiB | 5.98 MiB/s, done.
Resolving deltas: 100% (35772/35772), done.
$ cd OpenRTM-aist/
$ git diff svn/RELEASE_1_1_2 v1.2.1 > ~/openrtm.patch
$ cd ~/ros/jsk_demo_ws/src/rtm-ros-robotics/
$ git clone https://github.com/tork-a/openrtm_aist-release.git openrtm_aist
Cloning into 'openrtm_aist'...
remote: Enumerating objects: 6587, done.
remote: Total 6587 (delta 0), reused 0 (delta 0), pack-reused 6587
Receiving objects: 100% (6587/6587), 3.27 MiB | 2.89 MiB/s, done.
Resolving deltas: 100% (2493/2493), done.
$ cd openrtm_aist/
$ git checkout release/melodic/openrtm_aist
Branch 'release/melodic/openrtm_aist' set up to track remote branch 'release/melodic/openrtm_aist' from 'origin'.
Switched to a new branch 'release/melodic/openrtm_aist'
$ patch -p1 < ~/openrtm.patch 
patching file COPYRIGHT
patching file INSTALL.jp
patching file Makefile.am
patching file README
patching file README.jp
patching file README.jp.md
patching file README.md
patching file autogen.sh
patching file build/Makefile.am
patching file build/autogen
patching file build/cmakeconfgen.py
patching file build/codegen.py
patching file build/ezt.py
patching file build/makedeffile.py
patching file build/makevcproj.py
patching file build/makewrapper.py
patching file build/makewxs.py
patching file build/pickupprojs.py
patching file build/pkg_install_debian.sh
patching file build/pkg_install_fedora.sh
patching file build/pkg_install_raspbian.sh
patching file build/pkg_install_ubuntu.sh
patching file build/setuptest.py
patching file build/slntool.py
patching file build/uuid.py
patching file build/vcprojtool.py
patching file build/vcxprojtool.py
patching file build/vsprops2cmake.py
patching file build/yat.py
patching file configure.ac
patching file docs/DevelopersGuide/Tools.tex
patching file docs/DevelopersGuide/Tutorial.tex
patching file docs/Makefile.am
patching file docs/doxygen_classref.conf.in
patching file docs/doxygen_idlref.conf
patching file docs/doxygen_idlref.conf.in
patching file etc/component.conf
patching file etc/rtc.conf.sample
patching file examples/AutoTest/Makefile.am
patching file examples/Composite/Makefile.am
patching file examples/ConfigSample/Makefile.am
patching file examples/ExtTrigger/ConnectorComp.cpp
patching file examples/ExtTrigger/Makefile.am
patching file examples/ExtTrigger/run.sh
patching file examples/Makefile.am
patching file examples/SeqIO/Makefile.am
patching file examples/SeqIO/SeqIn.h
[... 521 lines cut ...]
patching file win32/OpenRTM-aist/installer/WixUI_uk-ua.wxl
patching file win32/OpenRTM-aist/installer/WixUI_zh-tw.wxl
patching file win32/OpenRTM-aist/installer/autowix.cmd.in
Reversed (or previously applied) patch detected!  Assume -R? [n] n
Apply anyway? [n] n
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file win32/OpenRTM-aist/installer/autowix.cmd.in.rej
patching file win32/OpenRTM-aist/installer/autowix_vc10.cmd
patching file win32/OpenRTM-aist/installer/crean.cmd
patching file win32/OpenRTM-aist/installer/langs.txt
patching file win32/OpenRTM-aist/installer/omniORB/omniORB_inc.wxs.in
patching file win32/OpenRTM-aist/installer/omniORB/omniwxs.py
patching file win32/OpenRTM-aist/prepare_openrtm.sh
patching file win32/OpenRTM-aist/prepare_openrtm_inst.sh
patching file win32/OpenRTM-aist/rtm/idl/Makefile.am
Hunk #1 succeeded at 4 with fuzz 2.
patching file win32/OpenRTM-aist/rtm/idl/device_interfaces/Makefile.am
patching file win32/OpenRTM-aist/rtm/idl/libRTCSkelvc8.vcproj.in
patching file win32/OpenRTM-aist/rtm_config.props
patching file win32/OpenRTM-aist/rtm_config.vsprops
patching file win32/OpenRTM-aist/rtm_config_ssl.props
patching file win32/OpenRTM-aist/rtm_config_ssl.vsprops
patching file win32/OpenRTM-aist/rtm_config_vc9.vsprops
patching file win32/OpenRTM-aist/utils/rtc-template/Makefile.am
```

pazeshun, 2020-04-14:
@ishiguroJSK 
openrtm_aist-releaseのrelease/melodic/openrtm_aistにOutPort.hのパッチを当てたものを作成しました。
https://github.com/pazeshun/openrtm_aist-release/tree/patch-to-1.1.2-melodic
/optに加えている変更を戻してもらって、上のものをworkspaceに入れてビルドして試してもらえませんでしょうか。

pazeshun, 2020-04-14:
> https://github.com/pazeshun/openrtm_aist-release/tree/patch-to-1.1.2-melodic

なぜかは全くわからないのですが、これを使ってworkspaceを作り直すと、
https://github.com/start-jsk/rtmros_hironx/issues/538
が出なくなっています。経過観察中です。

ishiguroJSK, 2020-04-14:
中身のdiffはmutex guard関係だったので，あり得る話だとは思う

pazeshun, 2020-04-14:
忘れないうちにPRにしておきました: https://github.com/tork-a/openrtm_aist-release/pull/7

k-okada, 2020-04-14:
@pazeshun 
release repository の使い方は
```
$ git diff
diff --git a/tracks.yaml b/tracks.yaml
index 061e027..9fc4443 100644
--- a/tracks.yaml
+++ b/tracks.yaml
@@ -127,15 +127,15 @@ tracks:
     - git-bloom-generate -y rosrpm --prefix release/:{ros_distro} :{ros_distro} -i
       :{release_inc}
     devel_branch: null
-    last_release: 22628ce
+    last_release: v1.2.0
     last_version: 1.1.2
     latest_release: 22628ce
     name: upstream
     patches: melodic
     release_inc: '7'
     release_repo_url: null
-    release_tag: 22628ce
+    release_tag: v1.2.0
     ros_distro: melodic
     vcs_type: git
     vcs_uri: https://github.com/OpenRTM/OpenRTM-aist.git
-    version: 1.1.2
+    version: 1.2.0

```
みたいなものを作って仮にコミットして
```
$ git-bloom-release melodic
途中でパッチを充てるところでコケるから
(bloom)git am --abort
(bloom)exit 0
$ git checkout debian/ros-melodic-openrtm-aist_1.2.0-1_bionic
$ dpkg-buildpackage -rfakeroot -uc -b
```
でdebを作るものです．
手元で作った1.2.0 は https://drive.google.com/file/d/1csoaVbc1YB5MgGHvd-mDVJ_PHeAaCkRY/view?usp=sharing
においておきます．

@ishiguroJSK 
 OpenRTM 1.2.0 が使えるか，という話だと思うけど，例えばhrpsys-base で以下の当たりを見ると.1.2.0 でやっている人はいそうだから質問してみては？
https://github.com/fkanehiro/hrpsys-base/issues/1284
https://github.com/fkanehiro/hrpsys-base/pull/1076

また，コンパイルを通すだけならそんなに難しくないはず，と思ってもう少し粘ってみると良いとおもいます．
```
/usr/bin/ld: /tmp/ccFgFQFg.o: relocation R_X86_64_PC32 against symbol `_ZrSjR9cdrStream' can not be used when making a shared object。 -fPIC を付けて再コンパイルしてください。
/usr/bin/ld: 最終リンクに失敗しました: 不正な値です
collect2: error: ld returned 1 exit status
make[2]: *** [/home/leus/catkin_ws/master/devel/.private/hrpsys_ros_bridge/lib/libpointcloudStub.so] Error 1
make[1]: *** [CMakeFiles/RTMBUILD_hrpsys_ros_bridge_pointcloud_genrpc.dir/all] Error 2
make: *** [all] Error 2
わからない
```
-fPIC を付けて下さい，教えてくれているけど，試してみたかな．

ishiguroJSK, 2020-04-14:
1.2.0の導入の仕方が正しいのかも不安だったので，進むのをやめてしまいましたが，もう少し進めてみます．

実際はhrpsys自体は一部を除いてコンパイルできていて，fPICのエラーはhrpsys_ros_bridgeです

ただhrpsys自体も1.2.0対応済みかというと・・・
https://github.com/fkanehiro/hrpsys-base/issues/1278#issuecomment-510686721

ishiguroJSK, 2020-04-20:
@k-okada 
> -fPIC を付けて下さい，教えてくれているけど，試してみたかな．

(個人的にno-pieなどといろいろ迷走していましたが)普通に-fPICつけたらコンパイルできました．
18.04からgccのPICやPIE周りのデフォルトが変わったことが原因と思われます．
https://stackoverflow.com/questions/50213089/linking-error-with-gcc-g-7-3-0-on-ubuntu-18-04

ここから実行時の問題ですが，

hrpsysまでのレイヤーでもほとんどのRTCが起動しない
1.2.0からデータポートに使えるデータ型に仕様変更があり，
https://github.com/OpenRTM/OpenRTM-aist/issues/847#issuecomment-616244079
コンパイルエラーが出た箇所のidlとcppを修正したつもりでも，
それ以上にほぼすべてのRTCが起動しない
```
~/catkin_ws/master/src/hrpsys (master) $ roslaunch launch/samplerobot.launch
... logging to /home/leus/.ros/log/df818d72-82c6-11ea-9e12-7c7a914eb678/roslaunch-W540-6882.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://W540:40683/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.5

NODES
  /
    hrpsys (hrpsys/hrpsys-simulator)
    modelloader (openhrp3/openhrp-model-loader)
    start_omninames (hrpsys/start_omninames.sh)

auto-starting new master
process[master]: started with pid [6892]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to df818d72-82c6-11ea-9e12-7c7a914eb678
process[rosout-1]: started with pid [6912]
started core service [/rosout]
process[start_omninames-2]: started with pid [6919]
omniNames: (0) 2020-04-20 14:22:07.472321: Data file: '/tmp/omniNames-logdir-6919/omninames-W540.dat'.
omniNames: (0) 2020-04-20 14:22:07.472371: Starting omniNames for the first time.
omniNames: (0) 2020-04-20 14:22:07.472490: Wrote initial data file '/tmp/omniNames-logdir-6919/omninames-W540.dat'.
omniNames: (0) 2020-04-20 14:22:07.472532: Read data file '/tmp/omniNames-logdir-6919/omninames-W540.dat' successfully.
omniNames: (0) 2020-04-20 14:22:07.472564: Root context is IOR:010000002b00000049444c3a6f6d672e6f72672f436f734e616d696e672f4e616d696e67436f6e746578744578743a312e300000010000000000000070000000010102000d0000003139322e3136382e31312e3500009d3a0b0000004e616d6553657276696365000300000000000000080000000100000000545441010000001c0000000100000001000100010000000100010509010100010000000901010003545441080000007f319d5e01001b09
omniNames: (0) 2020-04-20 14:22:07.472599: Checkpointing Phase 1: Prepare.
omniNames: (0) 2020-04-20 14:22:07.472649: Checkpointing Phase 2: Commit.
omniNames: (0) 2020-04-20 14:22:07.472692: Checkpointing completed.
process[modelloader-3]: started with pid [6923]
ready
process[hrpsys-4]: started with pid [6928]
#### file: ./rtc6928.log
##### file #####
HGcontroller0: onInitialize()
- port
  - port_type: DataInPort
- dataport
  - data_type: IDL:RTC/TimedDoubleSeq:1.0
  - subscription_type: Any
  - dataflow_type: push,pull
  - interface_type: corba_cdr,direct,shared_memory
- port
（略）
```
で
```
rtls localhost:15005/
SampleRobot(Robot)0.rtc  floor(Robot)0.rtc  HGcontroller0.rtc  ModelLoader  W540.host_cxt/
```
ほとんどのRTCが死んでいる.
rtc****.logを見ると
```
pr 20 05:02:03.129 INFO: manager: 1.2.0
Apr 20 05:02:03.129 INFO: manager: Copyright (C) 2003-2017
Apr 20 05:02:03.129 INFO: manager:   Noriaki Ando
Apr 20 05:02:03.129 INFO: manager:   Intelligent Systems Research Institute, AIST
Apr 20 05:02:03.129 INFO: manager: Manager starting.
Apr 20 05:02:03.129 INFO: manager: Starting local logging.
Apr 20 05:02:03.139 WARNING: NamingOnCorba: No endpoint for the CORBA naming service (localhost) was found.
Apr 20 05:02:03.139 INFO: NamingManager: NameServer connection succeeded: corba/localhost:15005
Apr 20 05:02:03.141 INFO: ManagerServant: Named manager reference (INS) was successfully created.
Apr 20 05:02:03.141 INFO: ManagerServant: Master manager servant was successfully created.
Apr 20 05:02:03.143 INFO: LocalServiceAdmin: All the local services are enabled.
Apr 20 05:02:03.148 INFO: manager: Component type conf file: /home/leus/catkin_ws/master/devel/share/hrpsys/samples/Sam\
pleRobot/SampleRobot.500.conf loaded.
Apr 20 05:02:03.148 INFO: ec_base: setRate(1000.000000) done
Apr 20 05:02:03.148 INFO: ec_base: setRate(1000000.000000) done
Apr 20 05:02:03.148 INFO: HGcontroller0: 1 execution context was created.
Apr 20 05:02:03.149 INFO: HGcontroller0: Initial active configuration set is default-set.
Apr 20 05:02:03.251 INFO: ec_base: setRate(1000.000000) done
Apr 20 05:02:03.251 INFO: ec_base: setRate(1000000.000000) done
Apr 20 05:02:03.251 INFO: GLbodyRTC0: 1 execution context was created.
Apr 20 05:02:03.251 INFO: GLbodyRTC0: Initial active configuration set is default-set.
Apr 20 05:02:03.255 ERROR: GLbodyRTC0: addOutPort() failed.
Apr 20 05:02:03.255 ERROR: GLbodyRTC0: addOutPort() failed.
Apr 20 05:02:03.255 ERROR: RobotHardwareService: appending provider interface failed
Apr 20 05:02:03.256 ERROR: GLbodyRTC0: addOutPort() failed.
Apr 20 05:02:03.256 ERROR: GLbodyRTC0: addOutPort() failed.
Apr 20 05:02:03.256 ERROR: RobotHardwareService: appending provider interface failed
Apr 20 05:02:03.257 INFO: ec_base: setRate(1000.000000) done
Apr 20 05:02:03.257 INFO: ec_base: setRate(1000000.000000) done
Apr 20 05:02:03.257 INFO: GLbodyRTC1: 1 execution context was created.
Apr 20 05:02:03.257 INFO: GLbodyRTC1: Initial active configuration set is default-set.
Apr 20 05:02:03.266 ERROR: GLbodyRTC1: addOutPort() failed.
Apr 20 05:02:03.266 ERROR: GLbodyRTC1: addOutPort() failed.
Apr 20 05:02:03.266 ERROR: RobotHardwareService: appending provider interface failed
Apr 20 05:02:03.267 ERROR: GLbodyRTC1: addOutPort() failed.
```
addOutPortがエラーを吐いているあたりいかにもではあるが，
HGcontrollerが生きていてSequencePlayerが落ちているのもよく分からない
hrpsys_ros_bridgeは18.04になってからfPICつけないとidlの**Stub.soがコンパイルできなかったのに，hrpsysが何も変更していないのも分からない．
そもそもOpenRTM-aist-pythonは1.2.0にしなくていいのか？と思い始めている