start-jsk/rtmros_choreonoid #331: eusモデルをchoreonoidに出現させる方法はありますか

Naoki-Hiraoka, 2020-05-19:
https://github.com/euslisp/EusLisp/tree/master/models にあるeusモデルをchoreonoidに出現させる方法はありますか?

gazeboにおける[eusurdf](https://github.com/jsk-ros-pkg/jsk_model_tools/tree/master/eusurdf)のようにして，
eusモデルをwrlに，sceneをyamlに変換するとよいのだと思うのですが，これらを行うことが可能なコードはどこかにありますでしょうか．

Naoki-Hiraoka, 2020-05-19:
https://github.com/fkanehiro/simtrans
を使用するとurdfからwrlが生成できそうなので，
[eusurdf](https://github.com/jsk-ros-pkg/jsk_model_tools/tree/master/eusurdf)で生成されたhttps://github.com/euslisp/EusLisp/tree/master/models のurdfをwrlに変換するのが早いでしょうか

YoheiKakiuchi, 2020-05-19:
> https://github.com/fkanehiro/simtransを使用するとurdfからwrlが生成できそうなので，
> eusurdfで生成されたhttps://github.com/euslisp/EusLisp/tree/master/models のurdfをwrlに変換するのが早いでしょうか

現状ではそれが一番コード書く量が少ないだろうと思います

Naoki-Hiraoka, 2020-05-19:
ありがとうございます．

closed 2020-05-19 by Naoki-Hiraoka

YoheiKakiuchi, 2020-05-19:
モデルはwrlじゃなくてbody(https://choreonoid.org/ja/manuals/latest/handling-models/bodymodel.html)になるかな？

Naoki-Hiraoka, 2020-05-19:
```
simtrans -i room73b2-foldable-desk/model.urdf -o /tmp/file.wrl
```
として，
.yamlファイルに
```
obj2:                                                                                                                                                                            
  name: 'OBJ'                                                                                                                                                                    
  file: '/tmp/file.wrl'                                                                                                                                                          
  translation: [1, 0, 0]                                                                                                                                                         
  rotation: [[1, 0, 0], [0, 1, 0], [0, 0, 1]]
```
と書いてロードしたところ，
物体がchoreonoid上に出現しましたが，
マウスでドラッグしても動かない,ロボットがcollision検出せずにすり抜けるなど，うまく動いていません．
[image]

bodyファイルとして必要なパラメータが足りていないのか，設定を見落としているのか，確認します．

YoheiKakiuchi, 2020-05-19:
僕の環境だとドラッグしても動かないは動かない。
これは、choreonoidのドラッグは矢印の大きさと物体の質量の積がかかる力になっていて、
小さい質量(tableの質量は0.1になっている)だと力が小さすぎて動かないのだと思う。

コリジョンはどうやって確かめたかな？
ロボットの真上の出現させたりしたら、ロボットとtableはコリジョンしてるように見えた。

YoheiKakiuchi, 2020-05-19:
https://discourse.choreonoid.org/t/importing-urdf-to-chorenoid/368
SDFPluginはURDFも読み込めると書いてあるようのでSDFPluginをビルドしてあれば直接読み込めるのではないだろうか。

Naoki-Hiraoka, 2020-05-19:
生成されたwrlに，
```
joints [                                                                  
    USE nil_link                                                             
   ] 
segments [                                                                
    USE nil_link                                                             
   ] 
```
を追加すると，マウスでドラッグするとわずかに動き,ロボットがgo-posでぶつかるとcollision検出するようになりました.

simtransのサンプルに載っている，
```
$ rosrun xacro xacro.py `rospack find pr2_description`/robots/pr2.urdf.xacro > /tmp/pr2.urdf
$ simtrans -i /tmp/pr2.urdf -o /tmp/pr2.wrl
```
で生成したPR2はそのままロードしても，
マウスでドラッグすると動き,ロボットがgo-posでぶつかるとcollision検出するので，
関節が無い物体だと何か問題があるのかもしれません．

SDFPluginのご紹介ありがとうございます．
BUILD_SDF_PLUGINオプションを付けてchoreonnoidをビルドし，urdfをロードすると，
真っ白な机が出現し，すぐに床をすり抜けて地下深くに落ちていきました．
experimentalということかもしれません．

Naoki-Hiraoka, 2020-05-19:
不思議なことに，今改めて
```
simtrans -i room73b2-foldable-desk/model.urdf -o /tmp/file.wrl
```
で生成した机をそのまま読み込むと，問題なくシミュレーションできました．
自分の環境が悪くて，choreonoidをビルドし直したことで直ったのかもしれません．

YoheiKakiuchi, 2020-05-19:
ソース見た感じも、jointsとsegmentsはchoreonoidでは見ていないように思っていて、それが正しいのじゃないかと思います。

YoheiKakiuchi, 2020-05-19:
あと、aptで入るmodel.urdfはmeshのところが、```model://``` になっていて、gazeboでは読めるけど、simtransだとエラーにならなかった？

Naoki-Hiraoka, 2020-05-19:
`roscd eusurdf/models`してから実行することでごまかしていました．

`export GAZEBO_MODEL_PATH=`rospack find eusurdf`/models:$GAZEBO_MODEL_PATH`
とすると，どこで実行してもエラーにならなくなりました．eusurdfをコンパイルしても自動的にはGAZEBO_MODEL_PATHに追加されないのですね．

しかし，worldファイルの場合は拡張子でエラーになり，
worldファイルの拡張子をsdfに書き換えても，
`model://`のエラーが上記いずれの方法でも解消されず，今のところ変換できていません．

YoheiKakiuchi, 2020-05-19:
```
diff --git a/simtrans/sdfhelper.cpp b/simtrans/sdfhelper.cpp
index ea76f18..ada24a0 100644
--- a/simtrans/sdfhelper.cpp
+++ b/simtrans/sdfhelper.cpp
@@ -16,7 +16,11 @@ filter(PyObject *self, PyObject *args)
     char* homePath = getenv("HOME");
     std::string home = homePath;
     sdf::addURIPath("model://", home + "/.gazebo/models");
-    
+
+    char* gzPath = getenv("GAZEBO_MODEL_PATH");
+    std::string gzPathstr = gzPath;
+    sdf::addURIPath("model://", gzPathstr);
+
     if (!sdf::readFile(filename, sdf))
       return NULL;
     
```

simtransをちょっと修正すると、```model://```を読むようになったが、まだ、pythonのどこかでコケてるね。

Naoki-Hiraoka, 2020-05-20:
https://github.com/start-jsk/rtmros_choreonoid/pull/332 と https://github.com/Naoki-Hiraoka/eusurdfwrl によってroom73b2をchoreonoidに出現させることができました．
```
rtmlaunch hrpsys_choreonoid_tutorials samplerobot_choreonoid.launch ENVIRONMENT_YAML:=`rospack find eusurdfwrl`/worlds/room73b2.yaml
```
[image]

worldファイルをwrlに変更するのはやめて，worldファイルからadd_objects.pyでロード可能なyamlファイルを生成するようにしました．
環境の物体の関節が多いとシミュレーションが遅くなるので，物体を空間に固定するのに加えて，物体の関節も原則固定しました．自分の環境ではほぼ1倍速で動いています．必要な物体だけ動かす形になりそうです．

- choreonoidは関節角度上下限がシミュレーションされないので，引き出しを動くようにすると，力を加えると引き出しが無限に飛んでいく．
- eusurdfの物体は質量パラメータが適当なので，[simtrans](https://github.com/fkanehiro/simtrans)の`-e`オプションで物体のバウンディングボックスから質量パラメータを計算しているが，あまり正確ではない
- 部屋によっては原点に壁があるので，cnoidを編集してロボットの初期位置をずらす必要がある．
- ドアの左右の扉の間の隙間が小さいため干渉し，ドアが開かない．

YoheiKakiuchi, 2020-05-30:
@Naoki-Hiraoka HRP2のデモで運んでいてメッシュがおかしくなった（穴の空いた）モデルの名前はなんてやつかな？

Naoki-Hiraoka, 2020-05-31:
以下の2つです．

下の写真はeusurdfの`.dae`をmeshlabで表示したものですが，黒くなっている部分にロボットが接触すると引っかかって離れなくなるようでした．
また，干渉計算の高速化のためにこれらのメッシュをmeshlabのQuadric Edge Collapse Decimationで単純化すると，黒くなっている部分のメッシュが欠けたりボロボロになりました．

room73b2-gifuplastic-900-cart
[image]

maxvalu-supermarket-basket
[image]

YoheiKakiuchi, 2020-06-03:
カートのハンドルはオリジナルのモデルの時点でそもそも面が反対でした。
以下のPRで修正されるはずです。
https://github.com/euslisp/EusLisp/pull/435
https://github.com/jsk-ros-pkg/euslib/pull/332

バスケットは、このブランチを使って(libcgal-devをインストールしcatkin buildする)、
https://github.com/YoheiKakiuchi/jsk_model_tools/tree/eus_cgal

eusgazebでロードすれば、ただしそうに変換されています。
```
eusgazebo (master)$ git diff
diff --git a/eusgazebo/euslisp/eusgazebo-util.l b/eusgazebo/euslisp/eusgazebo-util.l
index 45e245b..da57f97 100644
--- a/eusgazebo/euslisp/eusgazebo-util.l
+++ b/eusgazebo/euslisp/eusgazebo-util.l
@@ -1,5 +1,6 @@
 ;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;
 ;; urdf & gazebo util
+(require :constrained-delaunay "package://eus_assimp/euslisp/constrained-delaunay.l")
 
 ;; convert model
 (defun irteus2urdf-for-gazebo
```
[image]