start-jsk/rtmros_choreonoid #333: JaxonをChoreonoidのサンプルとして使用させていただいてもよろしいでしょうか？

s-nakaoka, 2020-07-23:
株式会社コレオノイドの中岡です。
すみません、本件Issuesに書くのは適切ではないかもしれませんが、ここが一番関係者の方々に伝わるかと思いまして、こちらに書かせていただくことをご了承ください。

表題にありますように、このリポジトリに含まれるJaxonのモデルを、Choreonoidのサンプルモデルとして使用させていただきたく思っています。これについて、このモデルのライセンス（ Creative Commons Attribution-ShareAlike 4.0 International License）上は問題ないかと思うのですが、念の為確認したく思っています。

これまでもChoreonoidではSR1というサンプルモデルは用意していたのですが、そちらはご存知のように非常に簡素なモデルでして、本格的な二足歩行ヒューマノイドロボットのモデルがサンプルとして利用できるとよいと思った次第です。言うまでもないですが、Jaxonは素晴らしいロボットで、DRC等での実績もあり、「本格的な二足歩行ヒューマノイドロボットのサンプルモデル」として申し分ないです。

もちろんこれまでもこちらのリポジトリから利用はできたと思うのですが、今回以下のような整備を行って、Choreonoidでより扱いやすいモデルにできればと思っております。

- 現在Choreonoidの標準モデルファイル形式であるBodyファイル形式のファイルとする
- よりサンプルとして扱いやすなるよう、メッシュを軽量化できないか試みる
- 足裏など環境と頻繁に接触する箇所に干渉チェック用のプリミティブモデルを付与する
- 歩行などのサンプルプロジェクトも用意する（シンプルコントローラですぐに動かせるものにする）

最終的にはChoreooid本体にサンプルとしてバンドルすることを考えております。その場合は、オリジナルのJaxonモデルのディレクトリに含まれるlicense.txtを対応するディレクトリに含めるようにし、オリジナルのクレジットとともに、当方で改変した旨の追記も入れておきます。

以上についてご見解をお知らせいただければ幸いです。
よろしくお願いいたします。

kindsenior, 2020-08-07:
ご返事が遅くなり申し訳ございません．東京大学の小島です

ChoreonoidのサンプルモデルとしてJAXONを検討なさっているとのことで大変光栄に存じます．
ライセンスにあります通り，ご希望の用途でのロボットモデルの使用は問題ございません．

また，関係者一同の希望としてもし可能でしたら

- ロボットモデルのファイル名やモデル名はJAXON_JVRCではなくJAXON_REDまたはJAXON2とすること
- モデルに修正を加える際にモデルの配色等が大きく変更されないよう留意すること

の2点に留意していただけると幸いです
以上，宜しくお願い致します

s-nakaoka, 2020-08-09:
小島さん

ご回答いただきありがとうございます。

> ライセンスにあります通り，ご希望の用途でのロボットモデルの使用は問題ございません．

ありがとうございます！大変有り難いです。

>  - ロボットモデルのファイル名やモデル名はJAXON_JVRCではなくJAXON_REDまたはJAXON2とすること
> - モデルに修正を加える際にモデルの配色等が大きく変更されないよう留意すること

承知いたしました。こちらについて留意の上進めます。

JAXONモデルをChoreonoidで活用できるかと思うとワクワクします。
今後作業の方進めてまいりますので、よろしくお願いします。

(本件Issueになりますので、私の方でクローズさせていただきます。）

closed 2020-08-09 by s-nakaoka