start-jsk/rtmros_choreonoid #335: autobalancerからchoreonoid内のリンク位置などの情報にアクセス出来ないでしょうか

HidemiANDO, 2020-09-28:
お忙しいところ大変失礼いたします。
私、筑波大学大学院制御ロボティクス研究室の安藤と申します。

以前hrpsys-baseのインストールに手間取っていたところ、こちらのパッケージをお教え頂き、なんとかやりたかったシミュレーションを行うことが出来ました。その件は誠にありがとうございました。

実は自分の研究に関連することで大変申し訳ないのですが、プログラムの実装において突き当たった問題について相談をさせていただきたく存じます。
現在RTCのautobalancerを改造して他の制御手法を実装するということを行っているのですが、その際に制御性能を検証するためchoreonoidシミュレーション内の重心位置（あるいは腰リンク位置）の真値を取得出来ないかと考えています。例えばロボットがy軸方向に転倒した際にはシミュレーション画面を反映してy軸方向の値が変動する…と言ったデータを取得したいのですが、そのようなことは可能でしょうか。
あるいは、choreonoidと何かしらの方法で通信を行ってchoreonoid内の値を取得するしか無いでしょうか。

厚かましいお願いではございますが、もし可能であればご教授いただけると幸いです。

ishiguroJSK, 2020-09-28:
体内制御プログラムの原則としてロボット体内から観測できない物理量にはアクセスできないようになっています．
（AutoBalancerは実機制御でも使うのでChoreonoidとは完全に切り離した実装になっています）

僕はやったことないので詳しくないですが，Choreonoidのプラグインとして実装すれば取得できたはずです．
https://choreonoid.org/ja/manuals/latest/plugin-development/ode-plugin.html#bodylink
そのプラグインからOpenRTMのポート繋ぐか，共有メモリなのか，実装大変そうですが…
CC：@YoheiKakiuchi @kindsenior @YutaKojio 

重心位置となると怪しいですが，オブジェクトの絶対位置をログファイルに書き出すだけなら標準機能でもできるはずです
https://choreonoid.org/ja/manuals/1.5/simulation/execution-and-playback.html#simulation-result-recording

YoheiKakiuchi, 2020-09-28:
https://github.com/start-jsk/rtmros_choreonoid/blob/master/hrpsys_choreonoid_tutorials/config/SensorReaderRTC_JAXON_RED.PD.conf#L18

こういう感じで、リンクのABS_TRANSFORM (Choreonoidでの座標)がchoreonoidのRobotName(Robot).rtc のRTCポートから出せます。
このポートをhrpsysのlog.rtcに繋げれば、hrpsysと同期したログを取れるはずです。
重心はどこかで計算するしかないです。リンクの位置姿勢が分かっていれば計算は簡単なはずです。

HidemiANDO, 2020-09-28:
＠ishiguroJSK  
＠YoheiKakiuchi
お忙しい中ご丁寧にお教えいただき、誠にありがとうございます。また、体内制御プログラムにつきまして勉強不足で申し訳ありません。承知いたしました。
また、方法までお教えいただきありがとうございます。今後はお教えいただいた方法で実装してみようと思います。
今回は貴重なお時間の中ご返信頂き、誠にありがとうございました。

closed 2020-09-28 by HidemiANDO