start-jsk/rtmros_choreonoid #339: #325 の影響でソースのhrpsysでもROBOT_IOB_VERSIONが3になる？

kindsenior, 2021-01-06:
まだ，検証中で自分の環境だけかも知れませんが，
#325 の影響でhrpsys-baseのソースを使っている場合でもROBOT_IOB_VERSIONが3でビルドされる気がします

de3edce36c8219f93cf64839afb28cc18612b450 の
https://github.com/start-jsk/rtmros_choreonoid/blob/e1b66948ce8ac12c20e44c3f8016e1106cb69f15/hrpsys_choreonoid/iob/CMakeLists.txt#L1-L6
の変更では，hrpsys_SOURCE_DIRが設定されているかどうかで，ソースとaptのhrpsysのどちらかを判定していると理解しています

で，僕の環境だとhrpsysがソースであってもhrpsys_SOURCE_DIRは設定されていなくてROBOT_IOB_VERSION=3でビルドされています

確認事項としては
- 基本的にhrpsysがソースの場合はバージョン4でビルドされるべきだというので合意されるか
- その場合どうやってSOURCEかどうかを判定すべきか

YoheiKakiuchi, 2021-01-07:
> 基本的にhrpsysがソースの場合はバージョン4でビルドされるべきだというので合意されるか

これは、あまり問題がなさそう。 どうしてもver3が使いたいパターンてあるのかな。

> その場合どうやってSOURCEかどうかを判定すべきか

hrpsysのバージョンで判定できないかな。

または、hrpsysをリリースするという方法がある。リリースはいいタイミングな気はする。

Naoki-Hiraoka, 2021-01-07:
自分のミスです。すみません。

> 基本的にhrpsysがソースの場合はバージョン4でビルドされるべきだというので合意されるか

hrpsysがソースの場合はIOB_VERSION4でビルドしたRobotHardware.soとIOB_VERSION2等でビルドしたRobotHardware.soを両方用意して、シミュレーション時にロボットごとにロードする.soを選択できるようにしておくのが良いのではないかと思っています。しかし、IOB_VERSION2ロボットユーザーの自分は、現状のIOB_VERSION4でビルドされるという仕様で今は困っていないので、必要になったらPRを出そうと思います。

> その場合どうやってSOURCEかどうかを判定すべきか

現状、hrpsysのバーションはaptとsourceで同じになっているので、hrpsys_PREFIXが`/opt/ros`から始まるかどうかで判定するしかないのではないかと思います。

バージョン管理についてよく分かっていないのですが、hrpsys-baseをリリースするかどうかはともかく、今のhttps://github.com/fkanehiro/hrpsys-base のpackage.xmlとCMakeLists.txtのバージョンを315.16に変えて、aptと開発版のバージョンが異なるようにしておくべきではないかと思いました。

sourceの場合。
```
~/hiraoka_ws/share_ws/devel/lib/pkgconfig$ cat hrpsys-base.pc 
prefix=/home/hiraoka/hiraoka_ws/share_ws/devel
exec_prefix=${prefix}/bin
libdir=${prefix}/lib
idldir=${prefix}/share/hrpsys/idl
includedir=${prefix}/include
link_shared_files=
link_static_files=
link_depend_dirs=
link_depend_files=
link_depend_options=
cflag_defs=
cflag_options=

Name: hrpsys-base
Description: Basic RT components and utilities to control robots using OpenRTM
Requires: openhrp3.1
Version: 315.15.0
Libs: ${link_depend_dirs} -L${libdir} ${link_shared_files} ${link_depend_options} ${link_depend_files} 
Libs.private: ${link_static_files}
Cflags: ${cflag_defs} ${cflag_options} -I${includedir}
```

aptの場合
```
/opt/ros/melodic/lib/pkgconfig$ cat hrpsys-base.pc 
prefix=/opt/ros/melodic
exec_prefix=${prefix}/bin
libdir=${prefix}/lib
idldir=${prefix}/share/hrpsys/idl
includedir=${prefix}/include
link_shared_files=
link_static_files=
link_depend_dirs=
link_depend_files=
link_depend_options=
cflag_defs=
cflag_options=

Name: hrpsys-base
Description: Basic RT components and utilities to control robots using OpenRTM
Requires: openhrp3.1
Version: 315.15.0
Libs: ${link_depend_dirs} -L${libdir} ${link_shared_files} ${link_depend_options} ${link_depend_files} 
Libs.private: ${link_static_files}
Cflags: ${cflag_defs} ${cflag_options} -I${includedir}
```

Naoki-Hiraoka, 2021-05-23:
https://github.com/start-jsk/rtmros_choreonoid/pull/349 の中で直しました。
https://github.com/start-jsk/rtmros_choreonoid/pull/349/files#diff-b3020cb3b91a035dec17620e50b893403cbc9e2e11e7996c3ff58836b6bed87b
hrpsys_PREFIXが`/opt/ros/`から始まるかどうかで判定します

Naoki-Hiraoka, 2021-06-09:
mergeされたhttps://github.com/start-jsk/rtmros_choreonoid/pull/352 で直したつもりです．

closed 2021-06-10 by kindsenior