start-jsk/rtmros_choreonoid #342: WSL で動かす方法

k-okada, 2021-03-31:
choreonoid を wsl で動かす方法はどうしたらよいでしょうか？

@kindsenior

k-okada, 2021-03-31:
wsl 1 -> wsl 2 に移動するにはどうしたらよいでしょうか？

YoheiKakiuchi, 2021-03-31:
ちょっと求められてるのと違う気もしますが、docker for windowsはどうでしょうか。
https://docs.docker.jp/docker-for-windows/install.html

以下から、docker-compose.yaml をダウンロードして、
https://gist.github.com/YoheiKakiuchi/11757132d13ce5433bdbb392c76dac60

以下で開始します。（１回めはイメージダウンロードするのでそれなりに時間がかかる）
```
docker-compose -f docker-compose.yaml up
```

ブラウザから接続できるxサーバーを使っているので、chromeなどで ``` localhost:3000 ``` に接続すると、画面が見れます。
linuxとwindowsで動作確認できてます。多分macでも大丈夫なのではないかと。

このxサーバーを使って、P50(linux)でシミュレーションが2.5倍くらい遅い、
P50でネイティブのxserverのnvidiaで、シミュレーションが1.37倍くらい遅い。
非力なwindowsだともっと遅くなりそう。

[image]

kindsenior, 2021-04-02:
遅くなりました

基本的に普通にWSL2をインストールすれば，いいので調べて出てくる内容と同じかと思います

1. VcXsrvのインストール（chocolatyがない場合は手動でインストール）
```
choco install vcxsrv -y
```

2. VcXsrvの設定と自動起動化（[このZIP](https://github.com/start-jsk/rtmros_choreonoid/files/6250153/config.zip)のconfig.xlaunchを%conf_file%とする）
```
set conf_file=%USERPROFILE%\Documents\config.xlaunch
set command_str=C:\PROGRA~1\VcXsrv\xlaunch.exe -run %USERPROFILE%\Documents\config.xlaunch
reg add HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\Run /v VcXsrv /t REG_SZ /d "%command_str%" /f
```

3. WSLのインストールのための設定
```
dism /online /Enable-Feature /FeatureName:Microsoft-Hyper-V-All /NoRestart
rem Enable-WindowsOptionalFeature -Online -FeatureName Microsoft-Hyper-V -All
dism /online /Enable-Feature /FeatureName:Microsoft-Windows-Subsystem-Linux /NoRestart
```

4. WSL2のインストールのための設定
```
dism /online /Enable-feature /Featurename:VirtualMachinePlatform /all /norestart

set update_msi=%USERPROFILE%\Downloads\wsl_update_x64.msi
wget https://wslstorestorage.blob.core.windows.net/wslblob/wsl_update_x64.msi %update_msi%
powershell -command "start-process powershell -wait -verb runas -argumentlist {-command msiexec /i %update_msi%}"
```

5. PCを再起動する

6. 変数を設定する
```
set ubuntu_ver_dot=18.04
set wsl_name=<ubuntu-name>
set wsl_user=<user-name>
set wsl_dir=c:\wsl\
```

7. WSL2のインスタンスを作成する
```
set /P ubuntu_ver_dot="What release of Ubuntu do you install? (like 18.04): "
set ubuntu_ver_dot=%ubuntu_ver_dot:~0,5%
set ubuntu_ver=%ubuntu_ver_dot:~0,2%%ubuntu_ver_dot:~3,2%
set org_ubuntu_name=Ubuntu-%ubuntu_ver_dot%

set appfile=%USERPROFILE%\Downloads\wsl-ubuntu-%ubuntu_ver%.appx
if not exist %app_file% wget https://aka.ms/wsl-ubuntu-%ubuntu_ver% -O %app_file%

PowerShell -command "Add-AppxPackage %app_file%"

ubuntu%ubuntu_ver% run exit
wsl --set-version %org_ubuntu_name% 2
wsl --set-default-version 2

set tar_name=%USERPROFILE%\Downloads\%wsl_name%.tar

wsl --export %org_ubuntu_name% %tar_name%
wsl --unregister %org_ubuntu_name%
wsl --import %wsl_name% c:\wsl\%wsl_name% %tar_name%

powershell -command "Get-ItemProperty Registry::HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\Lxss\*\ DistributionName | Where-Object -Property DistributionName -eq %wsl_name% | Set-ItemProperty -Name DefaultUid -Value ((wsl -d %wsl_name% -u %wsl_user% -e id -u) | Out-String)"
```

8. WSLないで通常のUbuntuの環境構築をする

k-okada, 2021-04-03:
ありがとう．ちなみに普通にWSLでUbuntuを入れたときは以下の画像のように「WSLにチェック」→「再起動」→「ストアでインストール」だったんだけど，
普通にWSL2をインストールも同じようにできないのかな？

それともストアでインストールした後に，1-10をやるのかな？それとも，すとーあでインストールの代わりが3-7のコマンドなのかな．

[image]

kindsenior, 2021-04-03:
ストアでのインストール手順は7の`set appfile=%USERPROFILE%\Downloads\wsl-ubuntu-%ubuntu_ver%.appx
`から`ubuntu%ubuntu_ver% run exit`までに該当します

アンインストールする場合は削除したいWSLのインスタンスを`wsl --list`で確認して`wsl --unregister <instance-name>`で削除します．
他の手順はWSLを使えるようにする設定なのでそのままで問題有りません
（exportとimportでインスタンスの保存ファイルを移動している場合はそのファイルも削除した方がいいです．上の例でいうと`c:\wsl\%wsl_name%`です）

kindsenior, 2021-04-18:
すみません

> 8. WSL内で以下の設定をする（必須ではない）

に必須な部分があったので一部修正しました（ご存知かもですが，その設定がないとcatkin buildが動きません）

k-okada, 2021-04-19:
wslでemacs & すると日本語が入力できない，と言われたんだけど，どうしている？emacs -nw だと出来るらしい。

kindsenior, 2021-06-29:
すみません，見逃してました

僕はCUI派なので-nwしか使ってませんでしたが，ctrl-\だと日本語入力できる気がします

半角/全角キーとかを押すと<zenkaku-hankaku> is undefinedみたいなエラーが出ることを指してますか？

kindsenior, 2021-06-29:
すみません，見逃してました

僕はCUI派なので-nwしか使ってませんでしたが，ctrl-\だと日本語入力できる気がします

半角/全角キーとかを押すと is undefinedみたいなエラーが出ることを指してますか？入力メソッドに何を使うかというので変わりそうですが，fcitxの場合以下でできそうです

    install input method you want to use
    sudo apt install fcitx-skk
    fcitx # launch inpout method tool
    
    sudo apt install x11-xserver-utils # disable repeated hits of henkan in windows

以下を.bashrc等に追加（profileの方が的確かも）

    # settings for fcitx
    IM=fcitx
    export XMODIFIERS="@im=$IM"
    export GTK_IM_MODULE=$IM
    export QT_IM_MODULE=$IM
    export NO_AT_BRIDGE=1
    
    xset -r 49 # disable repeated hits of henkan in windows